realsenseai/librealsense #9772: Hardcoded path in generated cmake config files

ahoarau, 2021-09-15:
With any version of librealsense, when we compile the `install` target, we get `<CMAKE_INSTALL_PREFIX>/lib/cmake/realsense2/realsense2Targets.cmake` that contains hardcoded paths like this : 

```
# Create imported target realsense2::usb
add_library(realsense2::usb INTERFACE IMPORTED)

set_target_properties(realsense2::usb PROPERTIES
  INTERFACE_LINK_LIBRARIES "C:/Users/<USER>/librealsense-2.48.0/build-release/libusb_install/lib/usb.lib"
)

# Create imported target realsense2::fw
add_library(realsense2::fw STATIC IMPORTED)

set_target_properties(realsense2::fw PROPERTIES
  INTERFACE_LINK_LIBRARIES "C:/Users/<USER>/librealsense-2.48.0/build-release/common/fw/fw.res"
)
```

Portability is then more difficult.

MartyG-RealSense, 2021-09-15:
Hi @ahoarau   I wonder whether this is a situation where building librealsense as a **static link library** would be helpful, as described in https://github.com/IntelRealSense/librealsense/issues/4709

ahoarau, 2021-09-15:
I was in deed a situation where I tried with a static lib building, but those hardcoded path are written in any case.

MartyG-RealSense, 2021-09-15:
The only available information reference that I could find about **INTERFACE_LINK_LIBRARIES** that relates to librealsense is in https://github.com/IntelRealSense/librealsense/issues/7947 where the **fw.res** line that you quoted above is discussed.

Is your situation that your static link library build is working but you would prefer for it to be easier in regard to portability, please?

ahoarau, 2021-09-15:
Linking against the shared library works, even though it has this hardcoded path.
Linked against the static library does not work, and the reason being this hardcoded path.

In any case, having hardcoded path in generated cmakes prevents from sharing binaries.

A relative path w.r.t the installation path might do the trick.

MartyG-RealSense, 2021-09-16:
I'm pleased to hear that https://github.com/IntelRealSense/librealsense/issues/7947 was a useful reference for you.  As you suggested, I will close this case.  Thanks again!

closed 2021-09-16 by MartyG-RealSense