realsenseai/librealsense #9773: MATLAB file from SDK2.0/matlab/+realsense

nchu-cha, 2021-09-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC |
| SDK Version                            | v2.49.0 | 
| Language                            | MATALB | 
| Segment			| others | 

### Issue Description
hello,

I'm trying to understand all the MATALB function. I had tried `align,m` and `camera_info.m`    , them I get some error

1. In the `align.m`, comes an error that shows `not enough input arguments` in line 6 `narginchk(1,1);`. Should I revise the `(1,1)` range?
`function this = align(align_to)`
            `narginchk(1, 1);`
            `validateattributes(align_to, {'realsense.stream', 'numeric'}, {'scalar', 'nonnegative', 'real', 'integer', '<=', int64(realsense.stream.count)});`
            `out = realsense.librealsense_mex('rs2::align', 'new', int64(align_to));`
            `this = this@realsense.filter(out);`
        `end`

2. In `canera_info.m` , I get  an error about `Cannot call the constructor of 'realsense.camera_info' outside of its enumeration block.` 

3. I want to let my depth resolution up to 1024*768, and let my RGB resolution up to 1920*1080, hoe can I change the resolution from using MATLAB function?

4.  I have seen other people get the intrinsic and extrinsic parameters before aligning the images, can anyone teach me how to get these parameters?

Does anyone know hoe to solve it?

Thank you

MartyG-RealSense, 2021-09-16:
Hi @nchu-cha   Thanks very much for your questions.

1 and 2.  The files that you mention are part of the MATLAB wrapper's operating files and are not meant to be run as example programs.  The scripts in the wrapper's file list that are examples that can be tried have the word **example** in their name.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab

These include:

advanced_mode_example.m
capture_example.m
depth_example.m
depth_view_example.m
pointcloud_example.m
rosbag_example.m

If you wish to perform alignment in the MATLAB wrapper, the script created by a RealSense user in https://github.com/IntelRealSense/librealsense/issues/3735 provides an example of doing so.

3 and 4.  https://github.com/IntelRealSense/librealsense/issues/3735 also demonstrates defining custom stream configurations with **cfg** instructions and obtaining intrinsics when alignment is being used.

nchu-cha, 2021-09-16:
thanks for your apply, I had tried the measure in #3735, but I got some questions.

1. When I key `cfg = realsense.config();` , it couldn't connect with L515 it will show `config with no properties`. Then I change to used `context` and it can connect the L515 camera. Can anyone tell me what's the different between `context` and `config` ?

2. After confirming the connection, I tried to used `cfg.enable_stream(realsense.stream.depth,1280,720,realsense.format.z16);` in the #3735, but it shows `Dot indexing is not supported for variables of this type.` , is that mean L515 camera can't use matlab to change resolution?
[image]

3. Because I need to change my depth resolution to 1920_1080 and RGB resolution to 1920_1080, I tried `depth_img = permute(reshape(depth_data',[3,depth_color.get_width(),depth_color.get_height()]),[3 2 1]);` , let `depth_color.get_width(1920)`, `depth_color.get_height(1080)` , it shows `too many input argument`, I think I need to convert the image resolution first. (When I used the Viewer the depth resolution only up to 1024_768. Can I use MATLAB set the depth resolution to 1920_1080?)

4. I found that there are many kind of "Constructor" and "Destructor" in the realsense matlab file, how can I know which function I need to use?

thank you

MartyG-RealSense, 2021-09-16:
1.  My understanding is that **cfg = realsense.config();** is the appropriate instruction for a MATLAB wrapper script, so I'm not sure why you encountered problems with it and why **context** worked for you instead.

2 and 3.  The L515 does not support 1280x720 depth (the reference script was designed for a RealSense 400 Series camera).  L515 supports a maximum **1024x768** resolution for depth.  RGB supports 1920x1080 though.

4.  I could not find any useful reference information about use of constructors and destructors in the MATLAB wrapper.  As long as you stick to investigating the **_example** programs mentioned earlier and not go into the wrapper's operational files, constructors and destructors is something that you should not need to be concerned about.

nchu-cha, 2021-09-17:
thanks for your apply, I finally can change my resolution for the depth image successfully.

I need the depth information, when using the Viewer it will have a RAW file, how can I get the RAW file form using MATALB?

thank you

MartyG-RealSense, 2021-09-17:
A .raw file would not work with the **rosbag_example** program.  It is expecting a **bag** format file, which is like a video recording of camera data instead of a static image. 

In the Viewer you can create a bag file by enabling streams in the RealSense Viewer and then left-clicking on the circular **Record** option beneath the camera's name in the Viewer's options side-panel.  The Viewer will then record into the bag all streams that were enabled at the time that the Record option was activated.

nchu-cha, 2021-09-17:
Can I used MATALB code to save a bag file? Because I want to do all the stuff on the MATLAB

MartyG-RealSense, 2021-09-17:
https://github.com/IntelRealSense/librealsense/issues/6057 discusses creating a script in the MATLAB wrapper to record a bag file.

nchu-cha, 2021-09-18:
I used the method in #6057 like the below to have a bag file
`filepath = fullfile('D:\matlab_code\bag_file','testrecord.bag');`
`cfg.enable_record_to_file(filepath);`

Then I want to read the information in the bag file, but it can't open the file, can anyone help me to solve the problem?
`bag = rosbag("D:\matlab_code\bag_file\testrecord.bag")` --> I key the command like this, and get the error
[image]

this is my MATLAB code：

clc; clear all; close all;

filepath = fullfile('D:\matlab_code\bag_file','testrecord.bag');
% set resolution
cfg = realsense.config();
cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);
% cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);
cfg.enable_record_to_file(filepath);

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();
% Start streaming on an arbitrary camera with default settings
profile = pipe.start(cfg);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
for i = 1:5
    fs = pipe.wait_for_frames();
end
% Stop streaming
pipe.stop();
% Select depth frame
depth = fs.get_depth_frame();
% Colorize depth frame
color = colorizer.colorize(depth);
% Get actual data and convert into a format imshow can use
% (Color data arrives as [R, G, B, R, G, B, ...] vector)
data = color.get_data();
img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
% Display image
imshow(img);
title(sprintf("Colorized depth frame from %s", name));
bag = rosbag("D:\matlab_code\bag_file\testrecord.bag");

MartyG-RealSense, 2021-09-18:
Reading a bag file in the MATLAB wrapper involves a different process to recording one with enable_record_to_file.  It is done with **cfg.enable_device_from**, as demonstrated in the **rosbag-example.m** script.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/rosbag_example.m#L6

After stopping the pipeline with pipe.stop(), you should be able to use the cfg.enable_device_from instruction to change the cfg configuration from the one that the script started with.

nchu-cha, 2021-09-19:
Is that mean if I need a bag file I need to used `cfg.enable_record_to_file()` to get a bag file, and after the `pipe.stop()` I can use `cfg.enable_device_fron_file()` to open my bag file?

I add the  `cfg.enable_device_from()` instruction after `pipe.stop()` , but my MATALB crash.

MartyG-RealSense, 2021-09-19:
Yes, use **cfg.enable_record_to_file()** to create the bag and **cfg.enable_device_from_file()** to play it back.

To obtain a complete, readable bag, the pipeline needs to be stopped so that the bag is **indexed** upon closure of the recording process.

The SDK's C++ reference about record and playback recommends updating the **device** after closing recording to signify that a new pipeline is being used where playback will take place instead of record.

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback#record-and-playback

```
pipe->start(); // Resume streaming with default configuration
device = pipe->get_active_profile().get_device();
```

In regard to getting the distance, https://github.com/IntelRealSense/librealsense/issues/6612#issuecomment-645229665 provides references for doing this in the MATLAB wrapper using **depth.get_distance**

nchu-cha, 2021-09-19:
I add the instruction like below, but the MATLAB crash and shows the error in the picture.
I had looked the #6057, but I still don't know how to solve the problem?
Can you help me?

this is my code：

filepath = fullfile('D:\matlab_code\bag_file','20120919_test3.bag');
% set resolution
cfg = realsense.config();
cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);
% cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);
cfg.enable_record_to_file(filepath);

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();
% Start streaming on an arbitrary camera with default settings
profile = pipe.start(cfg);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
for i = 1:5
    fs = pipe.wait_for_frames();
end

```
% Stop streaming
pipe.stop();
cfg.enable_device_from_file("D:\matlab_code\bag_file\20120919_test3.bag");
```

% Select depth frame
depth = fs.get_depth_frame();
% Colorize depth frame
color = colorizer.colorize(depth);
% Get actual data and convert into a format imshow can use
% (Color data arrives as [R, G, B, R, G, B, ...] vector)
data = color.get_data();
img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
% Display image
imshow(img);
title(sprintf("Colorized depth frame from %s", name));

[image]

MartyG-RealSense, 2021-09-19:
1.  The SDK's C++ record and playback example suggests that immediately after pipe stop, the **pointer** is reset for a new pipeline before starting playback.

`pipe->stop(); // Stop the pipeline that holds the file and the recorder`
`pipe = std::make_shared<rs2::pipeline>(); //Reset the shared pointer with a new pipeline`

My MATLAB wrapper programming knowledge is limited though, so I do not know what the wrapper equivalent of `pipe = std::make_shared<rs2::pipeline>();` would be, unfortunately.

2.  If you are using **depth_example.m** as a basis to capture and render the depth frame with a live device, you are aiming to display the depth stream (and the RGB stream too) on-screen and record it to a bag file at the same time, similar to the RealSense Viewer's record function?

There was a recent discussion that talked about doing simultaneous streaming and recording using **multithreading**.  I pointed the RealSense user in that case to the C++ multithreading script at https://github.com/IntelRealSense/librealsense/issues/6865  

Multithreading is not a simple subject though and I do not have knowledge about how it could be used in a MATLAB wrapper script.

nchu-cha, 2021-09-19:
The information I currently needed are : 1. RGB image, 2. Depth image, 3. Depth distance (in Viewer is a RAW file), I need these three information when I capture the photo from theL515 camera.

I need A bag file is because the RGB and DEPTH image can align together, and the methods I found all said that the parameters in the bag file are required, so I tried to get a bag file and read the parameters in it.

Now I can capture the depth image and get the bag file, so I start to capture the RGB image want to see the RGB image and the depth image. Is there an example for capture the RGB image?

MartyG-RealSense, 2021-09-19:
All stream types that have been defined with cfg instructions should be captured in the bag file if they are enabled at the time that recording begins.

I researched your situation further and found a MATLAB script contributed by a RealSense user in https://github.com/IntelRealSense/librealsense/issues/6763#issuecomment-657217234 that aligns depth and color in real-time and then generates a point cloud.  Adapting this script may avoid the need to record a bag file if you only needed a bag for the purpose of performing alignment.

nchu-cha, 2021-09-20:
I can capture the RGB image and DEPTH image at the same time. Now I need whole DEPTH image's distance information, so I tried the `get_distance(this, x, y)` instruction to have the distance information, but it didn't work, and had an error：
[image]

after get the depth frame I try to get the depth distance, is that a correct instruction if I want to get the distance information?

this is my code：

clc; clear all; close all;

tic

filepath = fullfile('D:\matlab_code\bag_file','20120919_test3.bag');

cfg = realsense.config();
% set resolution for RGB & DEPTH
cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);
cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);
cfg.enable_record_to_file(filepath);  % 之後要改檔名
% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Make Colorizer object to prettify depth output
colorizer = realsense.colorizer();
% Start streaming on an arbitrary camera with default settings
profile = pipe.start(cfg);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
for i = 1:5
    fs = pipe.wait_for_frames();
end
% Stop streaming
pipe.stop();

% Select color & depth frame
RGB_frame = fs.get_color_frame();
depth_frame = fs.get_depth_frame();

```
for y = 1:1024
   for x = 1:768
       depth_dis = depth_frame.get_distance(x,y);  % ******
   end
end
```

% Colorize depth frame
depth_color = colorizer.colorize(depth_frame);
% Get actual data and convert into a format imshow can use
% (Color data arrives as [R, G, B, R, G, B, ...] vector)
% depth_frame(深度影像上色資訊)
depth_data = depth_color.get_data();
depth_img = permute(reshape(depth_data',[3,depth_color.get_width(),depth_color.get_height()]),[3 2 1]);
% color_frame
RGB_data = RGB_frame.get_data();
RGB_img = permute(reshape(RGB_data',[3,RGB_frame.get_width(),RGB_frame.get_height()]),[3 2 1]);

% Display image
figure;imshow(depth_img);title(sprintf("Colorized depth frame from %s", name));
figure;imshow(RGB_img);title(sprintf("Colorized RGB frame from %s", name));

toc

MartyG-RealSense, 2021-09-20:
What happens if you remove the % part of the line?

**depth_dis = depth_frame.get_distance(x,y);**

MartyG-RealSense, 2021-09-20:
If you cannot make get_distance work, the real-world distance in meters can alternatively be achieved by multiplying the 16-bit **uint16_t** raw depth pixel value by the depth unit scale of the RealSense camera model being used (which is **0.001** meters by default on 400 Series cameras).  For example, 160000 (uint16_t depth pixel value) x 0.001 (depth unit scale value) = 16 (meters real-world distance).

 If the SDK's Align processing block (**align_to**) is used then the SDK automatically adjusts the aligned image if depth and RGB are different resolutions.

nchu-cha, 2021-09-22:
Sorry for I can't clearly describe my problem.

Is that mean I can use MATLAB to get the RAW file? I used **get_distance** to get the depth distance is because I didn't found the method can get the raw file from the MATALB. If MATLAB can directly generate RAW file I think I know how to change the unit from unit16 to meter.

Can MATLAB directly generate the RAW file?

thank you

MartyG-RealSense, 2021-09-22:
https://github.com/IntelRealSense/librealsense/issues/1485 discusses in detail how to write depth data in binary format, including how to approach the problem of doing so in the RealSense MATLAB wrapper.  For example, one RealSense user in that discussion in https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380487211 wrote a script for saving the data as a .txt file that MATLAB could read.

If you want to go even more directly than the MATLAB wrapper, the function **imwrite** in MATLAB itself offers the possibility of saving data in binary raw format, though it may not be the same as the .raw files that the RealSense SDK generates.

https://uk.mathworks.com/help/matlab/ref/imwrite.html

[image]

In regard to the script with a **point cloud player**, aside from having to change the stream resolutions to be compatible with L515 I cannot see a reason why the script would not work on an L515.

nchu-cha, 2021-09-23:
Sincerely thank you for answering me patiently.
I successfully can align the RGB and DETPH.
Below is my program, if anyone needs it can refer to it.

- Using get distance to have the whole frame distance information：
(My depth resolution is 1920_1080. When using `get_distance`, it records the depth distance from zero, so if I want to that the distance fill in the **dis_matrix** I need to that the for loop start from zero, and that the coordinates plus one.)
```
% get the distance
dis_matrix = zeros(1080,1920);
for y = 0:(depthHeight-1)
   for x = 0:(depthWidth-1)
       depth_dis = depth_frame.get_distance(x,y);
       dis_matrix(y+1,x+1) = depth_dis;
   end
end
```

- align RGB and DEPTH frame
(Adding `% *****` after the program is the program that needs to be added if alignment is required)

```
cfg = realsense.config();
% set resolution for RGB & DEPTH
cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);
cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);
pipe = realsense.pipeline();
colorizer = realsense.colorizer();
pcl_obj = realsense.pointcloud();
profile = pipe.start(cfg);
alignto = realsense.stream.color;  % *****
alignedFs = realsense.align(alignto); % *****
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
for i = 1:5
    fs = pipe.wait_for_frames();
end
% Stop streaming
pipe.stop();
% Select color & depth frame
RGB_frame = fs.get_color_frame();
% depth_frame = fs.get_color_frame();  % original depth frame
aligned_frames = alignedFs.process(fs);  % align image *****
depth_frame = aligned_frames.get_depth_frame();  % after align Depth image *****
% get depth image parameters
depthSensor = dev.first('depth_sensor');  % *****
depthScale = depthSensor.get_depth_scale();  % *****
depthWidth = depth_frame.get_width();  % *****
depthHeight = depth_frame.get_height();  % *****
depth_color = colorizer.colorize(depth_frame);
depth_data = depth_color.get_data();
depth_img = permute(reshape(depth_data',[3,depth_color.get_width(),depth_color.get_height()]),[3 2 1]);
figure;imshow(depth_img);title(sprintf("Colorized depth frame from %s", name));
```

MartyG-RealSense, 2021-09-23:
Thanks so much for sharing your MATLAB wrapper script with the RealSense community, @nchu-cha     :)

nchu-cha, 2021-09-23:
Can I have a view before I capture the frame? Because I want to make sure that the image which I capture is I want. I'm thinking that can I us the **capture_example** revise to the capture function?

thank you

MartyG-RealSense, 2021-09-23:
You could have the program wait for a keypress before initiating capture, like in the C++ script at https://github.com/IntelRealSense/librealsense/issues/8007#issuecomment-745518890

MartyG-RealSense, 2021-09-24:
1.  It looks as though most of the properties are related to the positional layout and display on screen of the graphical elements such as the text and ring axes of the accel and gyro display, the display of the buttons and where the different steam panels are located on the screen.

**RsPipeLine** would be the definition for the pipeline, and **RsColorizer** would refer to the use of a **colorizer** component to color-shade the depth data based on the depth values.

2.  It looks as though the x, y and z in the **set_motion_data** block of code is calculating xyz (Yaw, Pitch and Roll) values for the IMU's gyro and acceleration values in the x, y and z directions for its accelerometer and then displaying those values on-screen at the top of the application.

I do not know what 'r' represents though on the gyro and accel displays in the **capture-example.m** program though.  It is probably equivalent to 'n' in the gyro and accel readout in the Viewer.

[image]

3.  I would assume that **draw_motion_data** refers to rendering the gyro and accel data on-screen, yes, especially as the draw instructions in the code are labelled as AccelTextAX and GyroTextAX.
  
4.  I'm not sure but it looks like it may be code that decides what to do if a particular camera stream type is not detected as being supported (for example, if the color stream or the accel and gyro streams are not accessible).

5.  Again, not sure.  I would speculate that after checking whether a particular stream type is supported, it updates the screen with **drawnow** and then releases frames with **delete(RsFrameSet)** to free up memory capacity.

nchu-cha, 2021-09-24:
Is the function `StartUpFunc(app)`  in the `capture_example` is that mean when I click the **Start** button than it will do all the instruction in the function?
[image]

I want to create a capture button, so I imitate the stop button：
```
function onCaptureButton(app,~)
    if strcmp(app.tmr.Running, 'on')
       stop(app.tmr);
       app.RsPipeLine.stop();
    end
end
```

Than I create a Capture function like the below, but when I click the capture button nothing happen, it didn't shows the frame.
I don't know where the problem is, can anyone help me?
```
function CaptureUpFunc(app)
    %--time--%
    now_time  = datestr(now,30); %
    time_str = [now_time(5:8),'_',now_time(10:15)]; %
    filepath = fullfile('D:\matlab_code\dataset\',time_str); %
    
    app.cfg = realsense.config();
    app.cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);
    app.cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);
    app.pipe = realsense.pipeline();
    app.colorizer = realsense.colorizer();
    app.profile = app.pipe.start(app.cfg);
    app.dev = app.profile.get_device();
    name = app.dev.get_info(realsense.camera_info.name);
    for i = 1:5
        app.fs = app.pipe.wait_for_frames();
    end
    app.pipe.stop();
    app.RGB_frame = app.fs.get_color_frame();
    app.depth_frame = app.fs.get_depth_frame();  % *****
    app.depth_color = app.colorizer.colorize(app.depth_frame);
    app.depth_data = app.depth_color.get_data();
    app.depth_img = permute(reshape(app.depth_data',[3,app.depth_color.get_width(),app.depth_color.get_height()]),[3 2 1]);
    figure;imshow(app.depth_img);title(sprintf("Colorized depth frame from %s", name));
    imwrite(app.depth_img,[filepath,'_DEPTH_img.png']);
end
```

thank you

MartyG-RealSense, 2021-09-24:
A function is a list of instructions that is jumped to when the function name is called from elsewhere in the script and then the instructions are carried out in the order that they are listed in the function.

For example, if you created a Start button and set that start button to jump to a function named **Start()** that contained the start-up code, then that code would be run when the start button is clicked on.

Have you defined the new Capture button in the **properties (Access = public)** list?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/capture_example.m#L4

Have you also defined a list of settings for CaptureButton like the ones for the Start and Stop buttons in the link below?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/capture_example.m#L294-L298

MartyG-RealSense, 2021-09-24:
That is what your **onCaptureButton()** function looks like to me.  If the button is pressed then the onCaptureButton is jumped to and the stop code currently within that function should cause the pipeline to be stopped.  But because of the **If** logic statement that the stop code is within, the stop will only activate if the If condition's terms are met.  In other words, the stop would only activate if the state **app.tmr.Running**is currently On.  If it is Off then the If condition is not met and the stop code would not be able to be triggered.  So nothing would happen when the onCaptureButton button is pressed.

nchu-cha, 2021-09-24:
I change the **onCaptureButton()** function to the below：
I think if I want to capture the frame the pipeline need to keep going.
```
function onCaptureButton(app,~)
      if strcmp(app.tmr.Running, 'off')
         app.RsPipeLine.start();
         start(app.tmr);
      end
  end
```
And change my **CaptureUpFunc(app)** function too, but it can't save the frame that I want.
```
function CaptureUpFunc(app)
    %--time--%
    now_time  = datestr(now,30); %
    time_str = [now_time(5:8),'_',now_time(10:15)]; %
    filepath = fullfile('D:\matlab_code\dataset\',time_str); %
    
    cfg = realsense.config();  %
    cfg.enable_stream(realsense.stream.color,1920,1080,realsense.format.rgb8,30);  %
    cfg.enable_stream(realsense.stream.depth,1024,768,realsense.format.z16,30);  %
    app.RsPipeLine.start(cfg);  %
    app.RsPipeLine = realsense.pipeline();
    app.RsColorizer = realsense.colorizer();
    RsFrameSet = app.RsPipeLine.wait_for_frames();
    
    if app.SupportDepth ~=0
       df = RsFrameSet.get_depth_frame();
       dc = app.RsColorizer.colorize(df);
       dd = dc.get_data();
       d = permute(reshape(dd',[3,dc.get_width(),dc.get_height()]),[3 2 1]);
       app.hD = imshow(d,'Parent',app.DepthAX);
       imwrite(d,[filepath,'_DEPTH_img.png']);  %
    else
       d = zeros(480,640,3,'uint8'); d(:)=240;
       app.hD = imshow(d,'Parent',app.DepthAX);
       hold(app.DepthAX,'on');
       text(app.DepthAX,140,220,'The <Depth> property is not supported','Color','k','FontSize',16);
       hold(app.DepthAX,'off');
    end
end
```

I had found that if I write the `imwrite()` in the **StartUpFunc(app)** function it will save the frame before I press the start button, but I want to save the frame after I press the button. Do you know how to solve the problem?

MartyG-RealSense, 2021-09-24:
Perhaps you could define a bool condition such as 'pause' that can be true or false, and set it to false by default in your script's start function.  Then you could enclose wait_for_frames within an 'If pause = false' condition, so that new frames can only arrive if pause = false.

When the capture button is pressed, the onCaptureButton() function's code could set pause to 'true'.  As soon as it is set to true, wait_for_frames should no longer be able to be accessed.  The capture could then be saved and pause set to 'false' again so that wait_for_frames can resume streaming frames.

nchu-cha, 2021-09-25:
I'm not really sure how to write the code you mean.
Is like the code below?
```
function onCaptureButton(app,~)
    if strcmp(app.tmr.Running, 'off')
        app.RsPipeLine.start();
        start(app.tmr);
    else  % strcmp(app.tmr.Running, 'on')
        stop(app.tmr);
        app.RsPipeLine.stop();
    end
end
```
I this code when I press the capture the frame will stop pipeline but it can't save the frame, than I press again the view in the app will keep going.

I missed a place that need to add the **CaptureUpFunc** in the bottom part of the code, so I add the `runCaptureupFcn(app,@CaptureUpFunc);` in, but it shows that it **couldn't recognized function `runCaptureupFcn`**. I don't know why it can recognized the **StartUpFunc** function, but it can't recognized the Capture function.
[image]

```
methods (Access = public)

    function app = CHA_capture_test
     createComponents(app)
     registerApp(app, app.UIFigure)
     runStartupFcn(app,@StartUpFunc);
     runCaptureupFcn(app,@CaptureUpFunc);  % ****************************
     if nargout == 0
      clear app
     end            
    end

    function delete(app)
     delete(app.UIFigure)
    end
end
```

MartyG-RealSense, 2021-09-25:
Is the `@CaptureUpFunc `definitely outside of the other functions (for example, not accidentally nested within `@StartUpFunc`).  You could make sure of this by finding the first line of a function and then pasting the code for `@CaptureUpFunc` directly above that line.

[image]

nchu-cha, 2021-09-26:
Can I open the Viewer and the MATLAB at the same time for preview the frame of the L515 camera?

I open the viewer for checking the RGB frame but shows **no frame received**, what's wrong with the viewer? the depth frame is working normally.
[image]

nchu-cha, 2021-09-26:
I changed another way to solve the problem of the preview screen, I used **Realsense Viewer** to make sure the frame is I want, then I turn off the Viewer and run the MATALB code of the capture function. But I found that if I open the RGB camera first, to will shows **no frame receive** but if I turn on the depth sensor first then turn on the RGB camera it will be normal. do you know why the L515 camera will act like that? Because before I can turn on the RGB camera first, and it still can get the frame.
[image]

MartyG-RealSense, 2021-09-26:
I have heard of this behaviour once before, with Depth first and RGB second working but not the other way around.  I do not have an explanation for it though.

nchu-cha, 2021-09-27:
Thank you for your patient reply, I will close the issue

closed 2021-09-27 by nchu-cha

MartyG-RealSense, 2021-09-27:
Thanks very much @nchu-cha for the update and for your patience too!