realsenseai/librealsense #9774: Cannot enable two depth stream at once with L515

GragaBot, 2021-09-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {Python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm trying to collect depth images with both L515 at the same time. When I only enable two color streams, everything works fine. So there is no problem connecting both sensor to the pi. I can also enable one depth stream and two color streams. However, I cannot enable both infrared stream or depth stream at the same time. I would get the error : "Frame didn't arrive within 5000". I tried to increase it to 20000, but still couldn't get it to work. I also tried running the box-dimensioner demo program with 2 L515 but couldn't get it to work.

MartyG-RealSense, 2021-09-16:
Hi @GragaBot  May I first ask you to test whether the two L515 cameras can work normally if they are pointed in opposite directions away from each other when depth or infrared is being used on both cameras?

[image]

GragaBot, 2021-09-16:
Thank you for the response. Yes, I tested on pc realsense viewer, where I can stream both L515 depth images at the same time, with the three different orientations setup discussed in the white paper. I can see the interference as described in the white paper as well. But when I try to run the python scripts in raspberry pi 4 (8G), the second depth stream never arrives, no matter what orientation I set them to.

MartyG-RealSense, 2021-09-16:
Usually, only one RealSense camera is used with a Raspberry Pi board.  For example, Intel's ethernet networking white-paper uses multiple cameras but places one camera per Pi 4.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

Conceivably, if you can achieve two depth streams on PC from two cameras but not on Pi 4 then the problem may be related to the Pi 4 supplying insufficient power to support two depth streams.  The ethernet paper suggests providing power to the Pi via a **PoE Hat** for power other ethernet, or provide an external power source via Raspberry Pi USB-C connector.

https://www.raspberrypi.org/products/poe-hat/

closed 2021-09-28 by MartyG-RealSense