realsenseai/librealsense #9775: 'libcurl/libcurl_install/lib/libcurl.a', needed by 'tools/realsense-viewer/realsense-viewer', missing and no known rule to make it

KyungminJungARA, 2021-09-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i} | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2. 49.0}                          | 
| Language                            |  {C++}                          | 
| Segment			|  {others}                          | 

### Issue Description
<I am getting the following error while building the files
```[build] ninja: error: 'libcurl/libcurl_install/lib/libcurl.a', needed by 'tools/realsense-viewer/realsense-viewer', missing and no known rule to make it```
If I build the libcurl separately, it's no longer a problem, but I want to make it such that `librealsense` can build `libcurl` library together.>

MartyG-RealSense, 2021-09-16:
Hi @KyungminJungARA  I have not encountered this particular error with librealsense.  A useful starting point in investigating your case may be to look at the link below, which discusses what may cause the **missing and no known rule to make it** message with Ninja in general and what can be done to fix it.

https://stackoverflow.com/questions/54866067/cmake-and-ninja-missing-and-no-known-rule-to-make-it

Nir-Az, 2021-09-19:
Hi @KyungminJungARA ,
It could be no network when building LRS from source.
As a work around try CMake configure with the flag -DCHECK_FOR_UPDATES=false

MartyG-RealSense, 2021-09-25:
Hi @KyungminJungARA  Do you require further assistance with this case, please?  Thanks!

archaeopteryx-org, 2021-10-01:
Hi @MartyG-RealSense
Not-The-OP here, but I was just experiencing the same error message.
(I'm on Ubuntu 20.04 however.)
The workaround provided by Nir-Az worked for me, FWIW.

MartyG-RealSense, 2021-10-01:
Hi @archaeopteryx-org  Thanks very much for the helpful confirmation that the workaround worked for you in your particular case!

closed 2021-10-06 by MartyG-RealSense