realsenseai/librealsense #9776: Runtime error

zijian98, 2021-09-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | V2.28.1 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  Linux 4.4.38+ aarch64  | 
| Platform                                 | DJI Manifold 2G (Jetson TX2)  |
| SDK Version                            | SDK 2.0 (v2.49.0)                     | 
| Language                            |  C/C++ | 
| Segment			|  Drone | 

### Issue Description
Hi, 

I am trying to get D435i RGB to work but whenever I opened realsense-viewer, i encountered this error:

 `Invalid Value in rs2_get_option(options:0x7f5c02f520, option:Emitter On Off):
hwmon command 0x7d failed. Error type: No data to return (-21).`

The rest of the modules like stereo and motion are working fine. Also, when i tried to run `rs-hello-realsense`, I encountered this error: 

`terminate called after throwing an instance of 'std::runtime_error'
  what():  6�z�
Aborted (core dumped)
`

Does anyone knows why and how do I fix this? The device is connected through USB 3.
Thanks

MartyG-RealSense, 2021-09-16:
Hi @zijian98  Could you first confirm for me the firmware version of your D435i camera please, as **V2.28.1** is not a RealSense firmware.  You can find the currently installed firmware version by launching the RealSense Viewer and left-clicking on the **Info** option near the top of the options side-panel.

[image]

If you are using RealSense SDK version **2.49.0** then the recommended firmware version to be installed in the camera is **5.12.15.50**

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

zijian98, 2021-09-17:
Hi @MartyG-RealSense, firmware version is already 5.12.15.50 and below is a screenshot of the errors before realsense viewer crash:
[image]
[image]

MartyG-RealSense, 2021-09-17:
I see that you have Depth, RGB and the Motion Module all enabled at the same time.  Doing so is known to cause problems in the RealSense Viewer.  When the problem does occur, it is the enabling of the Motion Module that causes the issue.  It does not matter which order the three streams are enabled in, as enablement of the Motion Module is always the trigger.

Do you still experience problems if you only enable two streams, such as Depth and RGB?

MartyG-RealSense, 2021-09-29:
Case closed due to no further comments received.

liuyuhao1011, 2023-11-22:
> Case closed due to no further comments received.

yes still have this error while Depth and RGB

liuyuhao1011, 2023-11-22:
[image]

liuyuhao1011, 2023-11-22:
[image]

MartyG-RealSense, 2023-11-22:
I note in a Viewer image provided at https://github.com/IntelRealSense/librealsense/issues/9776#issuecomment-921400090 that there is a red **Multiple realsense udev-rules were found!** error pop-up in the top corner.  

[image]

This error means that more than one **udev** device handling rule may be installed on your computer, which can cause conflicts between the rules.  This can occur if librealsense has been installed from both packages and source code on the same computer.  Usually in this situation there are two rules instead of the usual one, and deleting one of the rules immediately fixes the error.

As explained by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547 the rule that should be deleted can be found at **/etc/udev/rules.d/99-realsense-libusb.rules**

liuyuhao1011, 2023-11-22:
sry but i use win11,not linux

MartyG-RealSense, 2023-11-22:
@liuyuhao1011  The quoted udev image was from @zijian98 who created this case.  It looks as though you are a different person who is commenting on an old issue from September 2021.  Apologies for the confusion.

I see that you are using SDK version **2.35.2**, which is now very old.  You are using the modern camera firmware driver **5.15.0.2** with this SDK version, which will cause errors.  The firmware that should be used with 2.35.2 is version **5.12.7.100** so you will need to downgrade your firmware to this older version.

If you do not wish to change your firmware then you should update your Windows SDK to version **2.54.1** or download version 2.54.1 of the RealSense Viewer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

liuyuhao1011, 2023-11-22:
@MartyG-RealSense really thank u for the help

MartyG-RealSense, 2023-11-28:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks very much for the update!

closed 2023-11-28 by MartyG-RealSense