realsenseai/librealsense #9777: Android C++ context.query_devices().size() return 0

Didi3333, 2021-09-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Operating System & Version |   {Android 11  | 
| Platform                                 | Android  |
| SDK Version                            |  { legacy / 2.49.0 }                          | 
| Language                            |  {QT / C++ }                          | 
| Segment			|  {Context }                          | 

### Issue Description
Hi All,
I'm developing an Android Application with C++ dependencies.
I'm able to obtain from my side the permission for USB access and obtain the device Name and its file Descriptor.

At this step i share thiese data do my own .so which include realsense headers.

Unfortunatly when i permform from my library :

rs2::context ctx;
ctx.query_devices().size();

It always return 0 as size.

Looking for in the github i identify that Java Wrapper call into DeviceWatcher file :

private static native void nAddUsbDevice(String str, int i); = > nAddUsbDevice(usbDesc.name, usbDesc.descriptor);

Function located in usbhost/enumerator-usbhost.cpp

I think this is what i have 0 as size

My question is how to do this step from C++ side.

Regards

MartyG-RealSense, 2021-09-16:
Hi @Didi3333  It would indeed seem to be an access permisions issue, as described in https://github.com/IntelRealSense/librealsense/issues/4742

May I first confirm that you are using the current **unrooted** version of the RealSense Android wrapper in the link below, as there was a different wrapper before that current one was introduced that could only be used on rooted Android devices .

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

The wrapper documentation link for unrooted devices below states that the **AAR** file acquires camera access permissions and so a RealSense Android app needs to include **librealsense.aar** in its dependencies.  Have you got this include in your project, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/android.md#intel-realsense-sdk-20-for-android-os

MartyG-RealSense, 2021-09-23:
Hi @Didi3333  Do you require further assistance with this case, please?  Thanks!

Didi3333, 2021-09-23:
Hi @MartyG-RealSense , no further assistance needed. As you mentioned, on an unrooted Android it's necessary to use the aar library which allows to create the Context allowing camera access from NDK.

Regards

MartyG-RealSense, 2021-09-23:
Thanks very much @Didi3333 for the update - it's excellent to hear that a solution has been achieved.  I will close the case as you do not require further assistance.  Thanks again!

closed 2021-09-23 by MartyG-RealSense

alpc72, 2021-10-06:
@Didi3333 Hi, I encountered the same problem, can you tell me how to solve it?