realsenseai/librealsense #9778: Naming a File After Recording pyrealsense2

codeandstuf, 2021-09-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy  }                          | 
| Language                            |  {Python}                          | 
| Segment			|  {others }                          | 

### Issue Description
<I have a script that starts a pipeline, waits for an object to enter a specified spot (by checking if the depth changes) then starts recording. My goal is to have this setup loop and for each new recording file to be named by a timestamp. My issue is that I can't seem to name the recording file once the pipeline has already started. I have tried stopping the pipeline and starting a new one to record, but this takes too long. If I can't find a solution I will make another script to rename already saved files to the time they were created, but I am hoping there is a better way. 

THIS IS HOW I SET UP THE PIPELINE
```
# Create a context object. This object owns the handles to all connected realsense devices
pipeline = rs.pipeline()
#creates a datetime object for file saving
now = datetime.now() # current date and time

dist_to_center_previous = 0
config = rs.config()
#depth resolution is a max of 1280 x 720
#max depth
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
cowID = 'cow1Test2.bag'
config.enable_record_to_file(cowID)
cow = False
#pipeline.start(config)
#starts the pipeline
pipeline_record_profile = pipeline.start(config)
device_record = pipeline_record_profile.get_device()
device_recorder = device_record.as_recorder()
#pause the recording until a cow is detected
rs.recorder.pause(device_recorder)
```
AFTER A DISTANCE CHANGE IS DETECTED I RUN THIS
```
#records the depth picture of a cow at the scale
#@param depth_to_cow is the depth to the cow that was first detected
def record(depth_to_cow):
    #resume the recording
    rs.recorder.resume(device_recorder)
    #set cow to false to allow a while loop
    cow = False
    while not cow:  
        print("Cow Detected")
        #waits for frames
        pipeline.wait_for_frames().keep()
        #finds distance to center
        dist_to_center = setDist()
        #checks if cow left
        cow = checkForCow(depth_to_cow, dist_to_center)
        time.sleep(1)
```

Any help would be appreciated.
Thanks in advance!>

MartyG-RealSense, 2021-09-17:
Hi @codeandstuf   If you are seeking to save separate bag files for every frame then the **save_single_frameset()** SDK instruction may be suitable if you need to avoid stopping and restarting the pipeline.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.save_single_frameset.html
https://github.com/IntelRealSense/librealsense/issues/2588#issuecomment-453757755

A RealSense user created a full Python script that uses save_single_frameset

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

**************

Alternatively, the C++ SDK tool **rs-convert** which extracts frames from a bag and uses timestamps to name the files, takes the approach of retrieving the timestamp with **get_timestamp()** and storing it as a string.

https://github.com/IntelRealSense/librealsense/blob/master/tools/convert/converters/converter-bin.hpp#L86-%23L125

closed 2021-09-21 by codeandstuf

MartyG-RealSense, 2021-09-22:
Great to hear that you made positive progress, @codeandstuf - thanks very much for the update!

A good starting point for creating a multicam Python application may be the **multiple_realsense_cameras.py** project that a RealSense user created.

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras