realsenseai/librealsense #9779: Cannot stream depth, color, infrared in one window using pyrealsense2

Chepo2393, 2021-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |  Ubuntu 20.4.3 | 
| Kernel Version (Linux Only)    |  5.11.0-34-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.49.0 | 
| Language                            |  python      | 
| Segment			|  others  | 

### Issue Description
I'm trying to gather the depth, color, infrared data using D435 but when I put everything together with hconcat, vconcat or vstack, hstack generates an index error in the arrays. have any idea how i could do it, this is my code.

```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configuracion de las camaras para el stream
pipeline = rs.pipeline()
config = rs.config()
#configuracion de la camara de progundidad
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
#configuracion de la camara RGB
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 15)
#configuracion de los dos infrarojos
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)

align_to = rs.stream.color
align = rs.align(align_to)

# Inicia el stream con las configuraciones hechas
pipeline.start(config)

# Ingnora el primer segundo para que se inicialize la camara
for i in range(30):
    frames = pipeline.wait_for_frames()

try:
    while True:

        # Solicita los primeros frames de las camaras
        frames = pipeline.wait_for_frames()
        #frames de profundidad
        aligned_frames = align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()
        #frames RGB
        color_frame = frames.get_color_frame()
        #frames IR
        infrared_frame = frames.first(rs.stream.infrared)
        if not depth_frame or not color_frame:
            continue

        # Convierte las imagenes en un array numpy
        Infra_image = np.asanyarray(infrared_frame.get_data())
        depth_image = np.asanyarray(depth_frame.get_data())     
        color_image = np.asanyarray(color_frame.get_data())
        
        # Se aplica un mapa de color a la imagen de profundida las imagenes deben ser convertidas de 8bits a pixeles
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Aplila las camaras en una sola ventana
        images1 = cv2.hconcat([color_image, depth_colormap])
        images2 = cv2.hconcat([color_image, Infra_image])
        final = cv2.vconcat([images1, images2])
        # Muestra las camaras en una ventana
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', final)
        #cv2.imshow('Imagen RGB', color_image)
        #cv2.imshow('Imagen de Profundidad', depth_colormap)
        #cv2.imshow('IR image', Infra_image)
        
         # Al presionar ESC termina el programa
        c = cv2.waitKey(1) % 0x100
        if c == 27:
            break
finally:
    #Detiene el stream
    pipeline.stop() 
    cv2.destroyAllWindows()
```

This is the error:
```
Traceback (most recent call last):
  File "prueba.py", line 50, in <module>
    images2 = cv2.hconcat([color_image, Infra_image])
cv2.error: OpenCV(4.5.1) /tmp/pip-req-build-jr1ur_cf/opencv/modules/core/src/matrix_operations.cpp:66: error: (-215:Assertion failed) src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type() in function 'hconcat'

```

MartyG-RealSense, 2021-09-16:
Hi @Chepo2393  The case https://github.com/IntelRealSense/librealsense/issues/5093 demonstrates the approach that a RealSense user took to aligning depth, infrared and color after advice from a RealSense team member.

Chepo2393, 2021-09-16:
I already tried to align the flows but I realized that the error is when I do the combination of IR with RGB or depth

```
align_to = rs.stream.infrared
align = rs.align(align_to)
...
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()
        infrared_frame = frames.first(rs.stream.infrared)
...
images2 = cv2.hconcat([color_image, Infra_image])
```

MartyG-RealSense, 2021-09-17:
An alternative approach that another RealSense user took in https://github.com/IntelRealSense/librealsense/issues/5963#issuecomment-638403524 was to align depth to RGB and then align left infrared to depth, so that they ended up with the left infrared aligned to depth and color.

MartyG-RealSense, 2021-09-29:
Case closed due to no further comments received.

closed 2021-09-29 by MartyG-RealSense

pfcouto, 2022-09-27:
HI @MartyG-RealSense, I would like to discuss this issue again. The realsense-viewer software uses the infrared data to help the depth sensors calculating the depth. One way to see that is to enable the stereo in the software and if you put your finger in front of the infrared camera you can see that we lose details and the capability of calculating the depth gets worse as it is visible in the stream.

I would like to use this the infrared camera to help the depth cameras too, but in python. How can I do it? Thanks!

MartyG-RealSense, 2022-09-27:
Hi @pfcouto  On RealSense 400 Series stereo cameras there are a pair of left and right infrared sensors (hence the name 'stereo') that are used to construct the depth image.  Putting a finger over either of the infrared sensors will cause the depth image to be obscured.

You do not actually need to enable the infrared streams to obtain a depth image though.  The depth image is not constructed by the Infrared and Infrared 2 streams in the Viewer, but from **raw** left and right IR frames inside the camera hardware at the point of capture.  The assembled depth frame is then sent along the USB cable to the computer.  The depth frame can therefore be obtained even if both Infrared streams are not enabled.

*******

Most RealSense models (not all of them) are equipped with a separate **projector** component that casts a semi-random infrared dot pattern onto the surfaces of objects in the scene.  The camera can use these dots as a 'texture source' to analyze for depth detail.  Increasing the **Laser Power** option's value makes the dots more visible to the camera and may increase the amount of depth detail on the image, whilst reducing Laser Power reduces the visibility of the dots and can cause the amount of detail on the depth image to be reduced.

400 Series cameras can also use ambient light in a scene to aid analysis of surfaces instead of using the projector's dot pattern if the scene is well lit.

pfcouto, 2022-09-27:
Yes. That's what I am refering to. I am using a d400 series camera - d435i and it has the IR Projector, can't I use this feature in python code to improve the depth quality?

Also, using this picture as an example, with the following line of code (`depth_frame.get_distance(109, 280)`) what am I calculating, Deth or Range? In my case if I have two objects on the same plan, but one is right in the center of the stream and the other one is further way to the side the distance calculated to both will be the same or really close, I believe this is because I am getting the depth and not the range (real distance to object, not to the plane it is on) as I would like.

How can Iget the Range instead of the Depth? Thanks a lot!!

[image]

MartyG-RealSense, 2022-09-27:
Yes, you can enable the projector with Python code by setting the SDK instruction **rs.option.emitter_enabled** to '1'.    Below is an example of such code.

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 1)
```

The above image that you quoted is from page 17 of the current edition of the data sheet document for the 400 Series. 

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

You are correct, the depth value is representative of the depth and not the range.  Text accompanying the image in the data sheet states: "The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range as illustrated". 

The depth value is equivalent to measured real-world distance in meters, with a small margin of error compared to a physical real-world tape measure from the camera lens to the observed object.

If you generate a pointcloud then you can obtain real-world 3D XYZ world coordinates from its **vertices**, like in the SDK's Python example program **opencv_pointcloud_viewer.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

pfcouto, 2022-09-27:
Ok, I will try it out. Currently I am using this piece of code. This does not use the IR Project automatically right? If I use a code similar to the one you exemplified I should get a better result, or not?

[image]

The image below represents my problem. I have two water bottles, they are in the same plane (or pretty close at least) so the depth is the same in both. However what I want is to get the range to both. In reality to the small water bottle it is accurate - 1.78m (or close) - as it is in the center of the capture, but that is not the case for the bigger one, in reality the range is 1,97m. How can I calculate this range values? As you can see I am also calculating the angles - horizontal and vertical - using the a fov, but I don't know the correct Fov to use, so maybe I can try to calculate the range with this? Right now I am using the Fov of the Depth_VGA but should I be using another one? Depth_VGA, Depth_HD or Color_Camera? Thanks a lot!!!

[image]

[image]

[image]

MartyG-RealSense, 2022-09-27:
The RealSense Viewer applies a range of configuration presets by default when launched, whilst for self-created scripts you need to manually program those settings into your scripting in order to replicate the images produced by the Viewer.  You should enable the IR Emitter in your script with the provided code in order to project a depth-enhancing dot pattern onto the scene

The official RealSense store page and the official information page for the D435i both quote the Depth HD FOV of 87 x 58 degrees so I would recommend using the Depth HD FOV.

On the D435i camera model the color FOV is smaller than the depth FOV, so you may get more accurate results if you align depth to color so that the depth FOV is automatically resized to match the color FOV size.  Because the depth and color sensors are at different physical positions on the front of the camera, the color view will be horizontally offset from the depth sensor's view unless the two streams are aligned.

pfcouto, 2022-09-27:
Isn't the alignment done by the line `frames = self.align.process(frames)` shown in the first print? So you think that I should use the Depth_HD FOV instead? On the process to align, am I aligning to de Color or Depth? If it is to the color maybe using the color FOV would be wiser.

What about the range? If there is no way to get the range using the SDK can you provide a way to calculate it, like using the angles of the point using the center point as measure, because the central point is the normal of the camera or some other way? Thanks.

MartyG-RealSense, 2022-09-27:
Could you provide a link to your previous answer that contains the line **frames = self.align.process(frames)** please?

If you have that line in your code then you are already using alignment in your project, yes.

Yes, I would recommend using Depth FOV HD (87 x 58 degrees).

Most people perform depth to color alignment.  It is very rare that color to depth alignment is used.

The main difference between them is that on D435i (where the color FOV is smaller than the depth FOV), the outer edges of the aligned image may be cut off.  This is because the smaller color FOV means that there is not color coordinates at the edges of the image to match up to depth coordinates.

If color to depth alignment is used on D435i then the smaller color FOV means that the color image will be stretched to the size of the larger depth FOV, so the aligned image will fill the screen and not have information cut off at the edges.

*************

It sounds as though the information at https://github.com/IntelRealSense/librealsense/issues/9297 about finding an object's height (distance from ground) using the angle might be a helpful reference for your range question.

pfcouto, 2022-09-27:
OK, so in my code I have `self.align = rs.align(rs.stream.color)` that means I am aligning the color to depth right? If that's the case then I should be using the Depth FOV and not the Color FOV. Or is ir the opposite and I am aligning the depth to the color and then I should use the Color FOV? Sorry for all the trouble and thank you!

MartyG-RealSense, 2022-09-27:
It's no trouble at all  :)

rs.stream.color in the brackets means that it is aligning depth to color, as whatever is in the brackets is the target stream that is being aligned to.  To align color to depth, you change the brackets to rs.stream.depth

Whilst the majority of RealSense users align depth to color, which one you use will depend on which works best for your particular project.

pfcouto, 2022-09-27:
OK, then the correct FOV to use should be the Color one, correct? I might be getting somewhere with the range using the angles so overcoming this problem of the FOVs would be awesome. Thanks!

MartyG-RealSense, 2022-09-27:
rs.stream.color is the FOV that most people align to.  Try both though and see which one provides the most accurate results, since it is easy to test both modes just by changing the bracket contents.  

I will be pleased to continue this discussion with you tomorrow if you have further questions.  Good luck!