realsenseai/librealsense #9780: D435e is not extendable to RS2_EXTENSION_AUTO_CALIBRATED_DEVICE?

iprinsep, 2021-09-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | (FRAMOS D435e) | 
| Firmware Version                   | (1.11.0.0) | 
| Operating System & Version |   {Ubuntu 20.4)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                           |  { legacy / 2.<?>.<?> }                          | 
| Language                            |   {C++ }                          | 
| Segment			        |  {Other}                          | 

### Issue Description
HI

I am trying to health check the calibration of my cameras but my code is failing with a null ptr during the construction of auto_calibrated_device in rs_devices.hpp at line 315:

class auto_calibrated_device : public calibrated_device
    {
    public:
        auto_calibrated_device(device d)
            : calibrated_device(d)
        {
            rs2_error* e = nullptr;
            if (rs2_is_device_extendable_to(_dev.get(), RS2_EXTENSION_AUTO_CALIBRATED_DEVICE, &e) == 0 && !e)
            {
                _dev.reset();
            }
            error::handle(e);
        }

The rs2_is_device_extendable to query causes my object to be reset.  Can someone tell me why my D435e doesn't support this functionality, or whether there is an error in what I am trying to please?

My code for this task is pasted below, 

Many thanks.  Ian.

#include <librealsense2/rs.hpp>
#include <librealsense2/hpp/rs_device.hpp>
#include <iostream>
#include "date/tz.h"
#include <ctime>

// Main program
int main(int argc, char *argv[])
{
    rs2::pipeline pipe;
    rs2::config cfg;
    auto callback = [&](const rs2::frame &frame)
    {};

    rs2::pipeline_profile profile = pipe.start(cfg, callback);
    
    rs2::device dev = profile.get_device();
    rs2::auto_calibrated_device cal = rs2::auto_calibrated_device(dev);

    const char* cam_version = dev.get_info(RS2_CAMERA_INFO_DEVICE_VERSION);
    std::string crtTime = date::format("%d-%m-%Y %H-%M-%S", std::chrono::system_clock::now());
    std::string camMsg = crtTime.substr(0, crtTime.size()-6)+ "  Device version of camera connected: " + cam_version;
    std::cout << camMsg << std::endl;

    float* health;

    std::cout << "Running get_calibration_table" << std::endl;
    std::vector<uint8_t> CalTableBefore = cal.get_calibration_table();

    std::cout << "Running run_on_chip_calibration" << std::endl;
    std::vector<uint8_t> CalTableAfter = cal.run_on_chip_calibration(NULL, health, 5000);

    pipe.stop();

}

MartyG-RealSense, 2021-09-17:
Hi @iprinsep  Intel do not provide technical support for the FRAMOS industrial versions of RealSense cameras, as they are not Intel products and support for them is provided by FRAMOS.  You can send FRAMOS your technical support question by visiting the link below and clicking the **Contact Us** button at the bottom of the page.

https://www.framos.com/en/resources/support

[image]

iprinsep, 2021-09-20:
Hi MartyG
I have since received a response from FRAMOS support confirming they do not support on chip calibration, hence my issues above.  I will recalibrate using targeted calibration as per their procedure.
Many thanks
Ian

closed 2021-09-20 by iprinsep

MartyG-RealSense, 2021-09-20:
It's great to hear that you received support for your issue, @iprinsep - thanks very much for the update!