realsenseai/librealsense #9783: Python API pause bag for debugging

DiXcipuli, 2021-09-20:
Hi! I can not find a good example / documentation on how to pause and resume a bag file in python.
I am using vscode debugger, and I would like to check frame by frame what is happening (I am using keypoint detection DL over the stream) . For now, my pipeline is based on the 'load bag' example:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

I have only seen this topic in c++, which differs significantly from my pipeline: https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.49.0 | 
| Operating System & Version |  Ubuntu 18.04 (L4T Jetson Xavier)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Jetson Xavier |
| Language                            |  python                        | 
| Segment			|  Robot                         |

MartyG-RealSense, 2021-09-20:
Hi @DiXcipuli  In https://github.com/IntelRealSense/librealsense/issues/4660#issue-482102697 there is an example of a Python script that uses **playback.pause()** and **playback.resume()** to pause and resume bag playback.

DiXcipuli, 2021-09-20:
Thanks @MartyG-RealSense , indeed it is a good start! However, as I may pause the stream for a while, while debugging, I have the runtime error: **RuntimeError: Frame didn't arrive within 5000** . Does it mean I cant excess this duration?
I set: **playback = profil.get_device().as_playback()**  and  **playback.set_real_time(False)**

closed 2021-09-20 by DiXcipuli