realsenseai/librealsense #9784: D435 on robot arm

xczzz, 2021-09-20:
I used D435 to measure the practical 3D coordinate of a fixed point in the real world according to a pixel of a rgb image. I used the function "rs2_deproject_pixel_to_point" to get the 3D coordinate. I calibrated the RGB camera and robot. Then I turned the 3D coordinate into the basic coordinate system of robot arm. But when the robot arm moved, it seemed that the 3D coordinate in the real world was moving too. It was confused me. How can I get the real coordinate in the robot system?

MartyG-RealSense, 2021-09-20:
Hi @xczzz  It sounds as though your project has some similarities to https://github.com/IntelRealSense/librealsense/issues/9749 where the camera is mounted in a fixed position and captures an RGB image of the scene.  The 2D RGB coordinate is converted to a 3D world coordinate using **rs2_deproject_pixel_to_point**.  This information is then used to move a carriage component to a certain real-world position, whilst the camera remains stationary.

I recommend reading that case and then returning here with any new questions that arise from it.

xczzz, 2021-09-20:
@MartyG-RealSense Thank you very much. It seems that this issue is helpful to me. I will try as soon as I get time.

MartyG-RealSense, 2021-09-20:
You are very welcome, @xczzz - please do update here when you have results of your test.  Good luck!

xczzz, 2021-09-21:
Hi @MartyG-RealSense I have a question about the coordinates system of the camera. I find a pixel in RGB image and using "rs2_project_color_pixel_to_depth_pixel" to turn this pixel into depth pixel. Then I use "rs2_deproject_pixel_to_point" to get a 3D point. I wonder which coordinate system this point is relative to. RGB sensor or depth sensor? (It seems that the two systems are different.) If the point is relative to the depth coordinate system, how can I turn it into RGB system?

MartyG-RealSense, 2021-09-21:
The depth world-origin point (0,0,0) of the coordinate system of the 400 Series cameras is the center-line of the left infrared sensor.  When facing the camera from the front, this is the infrared sensor that is on the right side of the camera.

[image]

I would think that if you needed a 2D (xy) RGB coordinate then you could store the coordinates of a particular RGB pixel before you use rs2_project_color_pixel_to_depth_pixel to convert the RGB pixel into a depth pixel.  

Otherwise, after having used rs2_deproject_pixel_to_point to transform a 2D depth pixel (obtained by rs2_project_color_pixel_to_depth_pixel) into a 3D depth point then you could subsequently use **rs2_project_point_to_pixel** to convert a 3D xyz depth point back to a 2D xy pixel coordinate.

xczzz, 2021-09-21:
@MartyG-RealSense I think there is a misunderstanding between us. I want to get the 3D point relative to RGB sensor coordinate system finally. I have noticed the function "rs2_transform_point_to_point". Can I use it  to change the 3D point relative to depth coordinate system into 3D point relative to RGB coordinate system?

MartyG-RealSense, 2021-09-21:
**rs2_transform_point_to_point** transforms the position of coordinates, such as taking multiple individual point clouds and positioning and rotating them so that they all occupy the same space as a single large point cloud (a process known as an 'affine transform').

I believe that what you are referring to is a rather complex subject that will involve finding the **extrinsics** of the position between depth and RGB sensors.  https://github.com/IntelRealSense/librealsense/issues/2948 may be a useful reference.

xczzz, 2021-09-21:
@MartyG-RealSense Actually I just need a transformation between RGB 3D space and depth 3D space to help me turn the 3D point in RGB coordinate system into depth coordinate system. According to the description of rs2_transform_point_to_point, it is “Knowing the extrinsic parameters between two streams allows you to transform points from one coordinate space to another, which can be done by calling rs2_transform_point_to_point(...)” which is from documentation. It seems that it's different from what you said. It confused me.

MartyG-RealSense, 2021-09-21:
RGB coordinates should be in the depth system if they have been mapped to depth coordinates (either by converting a single RGB pixel to depth pixel with rs2_project_color_pixel_to_depth_pixel OR performing depth to color alignment on the entire image). And then, as a final step, using rs2_deproject_pixel_to_point to transform 2D pixels into 3D depth points in the depth coordinate system.

xczzz, 2021-09-21:
@MartyG-RealSense You mean after  converting a single RGB pixel to depth pixel with rs2_project_color_pixel_to_depth_pixel, the 3D point calculated by  rs2_deproject_pixel_to_point is relative to RGB system because the two systems are the same?

MartyG-RealSense, 2021-09-21:
Using rs2_project_color_pixel_to_depth_pixel on a single specific pixel to convert it from an RGB pixel to a depth pixel is like using **align_to** to align RGB and depth for the entire image.  This is why rs2_project_color_pixel_to_depth_pixel is much more processing-efficient than align_to if you only need to map color to depth for one pixel instead of all the pixels in the image.

There is one coordinate system, with the center of the left infrared imager being the origin point of that coordinate system.  This is described in the SDK's **Projection** documentation.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

xczzz, 2021-09-22:
@MartyG-RealSense I refered to the issue but there is still the problem. I wonder where the camera coordinate system is? And Where are the depth sensor coordinate origin and color sensor coordinate origin respectively?

xczzz, 2021-09-23:
@MartyG-RealSense It is alright. Yesterday I found a new problem. I found that when I turned the center pixel coordinate (320, 240) in a 640×480 RGB image into 3D coordinate in RGB space, the x and y in 3D coordinate are both not zero. In my opinion, I think x and y should  be both zero because the pixel is in the center where they are from.  Especially the bias of x is big. The same situation happened in depth space.

MartyG-RealSense, 2021-09-23:
Z-depth measurements have an error factor called **RMS Error** in them.  Depth measurement error starts at around zero at the camera lenses and increases linearly with distance from the camera.  The further away that an observed object or surface is, the greater the error factor.  

On the D435 / D435i models, this error becomes particularly noticable for coordinates that are at a distance from the lens of 3 meters or more.  The newer D455 model has 2x the accuracy over distance, meaning that it has the same accuracy at 6 meters distance that a D435 / D435i has at 3 meters.

You can read more about RMS error in point **5.  Understand theoretical limit** of the section of Intel's camera tuning paper linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

In regard to xy not being zero at the center (320,240) of a 640x480 image, another RealSense user in https://github.com/IntelRealSense/librealsense/issues/7100 experienced this and questioned it.  A RealSense team member provides them with advice.

xczzz, 2021-09-24:
Hi @MartyG-RealSense  I refered to the issue #7100 but there is still the problem. I did not wrap the x and y like this issue. I think the input is right in my code. Can you please give me some other advices?

xczzz, 2021-09-24:
Here is my code. @MartyG-RealSense 
`
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include <cmath>
#include <fstream>
#include <iostream>
#include <string>

#include <librealsense2/rs.hpp>
#include <librealsense2/rsutil.h>
#include <ros/ros.h>

#include "../examples/example.hpp"
#include "locator/BlackPoint3D.h"
using namespace std;
#include <opencv2/aruco.hpp>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/imgproc/imgproc.hpp>
using namespace cv;

int main(int argc, char** argv) {
  ros::init(argc, argv, "aruco_to_point_and_talk");
  ros::NodeHandle nh;
  ros::Publisher pub = nh.advertise<locator::BlackPoint3D>("point_coordinate", 1000);
  rs2::colorizer c;
  rs2::pipeline pipe;
  rs2::config cfg;

  cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);
  cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);

  rs2::pipeline_profile profile = pipe.start(cfg);
  auto depth_sensor = profile.get_device().first<rs2::depth_sensor>();

  static float depth_scale = depth_sensor.get_depth_scale();
  static struct rs2_intrinsics depth_intrin;
  static struct rs2_intrinsics color_intrin;
  static struct rs2_extrinsics depth_extrin_to_color;
  static struct rs2_extrinsics color_extrin_to_depth;

  const rs2::stream_profile color_profile =profile.get_stream(RS2_STREAM_COLOR);
  const rs2::stream_profile depth_profile =profile.get_stream(RS2_STREAM_DEPTH);
  depth_intrin = depth_profile.as<rs2::video_stream_profile>().get_intrinsics();
  color_intrin = color_profile.as<rs2::video_stream_profile>().get_intrinsics();
  depth_extrin_to_color =depth_profile.as<rs2::video_stream_profile>().get_extrinsics_to(color_profile);
  color_extrin_to_depth =color_profile.as<rs2::video_stream_profile>().get_extrinsics_to(depth_profile);

  ros::Rate rate(30);

  while (ros::ok() && waitKey(1))
  {
    locator::BlackPoint3D BP;

    rs2::frameset frameset = pipe.wait_for_frames();

    auto depth = frameset.get_depth_frame();
    auto color = frameset.get_color_frame();
    auto colorized_depth = c.colorize(depth);

    const int depth_w = depth.as<rs2::video_frame>().get_width();
    const int depth_h = depth.as<rs2::video_frame>().get_height();
    const int color_w = color.as<rs2::video_frame>().get_width();
    const int color_h = color.as<rs2::video_frame>().get_height();
    const int colorized_depth_w = colorized_depth.as<rs2::video_frame>().get_width();
    const int colorized_depth_h = colorized_depth.as<rs2::video_frame>().get_height();

    Mat depth_image(Size(depth_w, depth_h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
    Mat color_image(Size(color_w, color_h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
    Mat colorized_depth_image(Size(colorized_depth_w, colorized_depth_h), CV_8UC3, (void*)colorized_depth.get_data(), Mat::AUTO_STEP);

    Ptr<aruco::Dictionary> dictionary = getPredefinedDictionary(aruco::DICT_ARUCO_ORIGINAL);

    Ptr<aruco::DetectorParameters> parameters = aruco::DetectorParameters::create();

    vector<vector<Point2f>> markerCorners, rejectedCandidates;

    vector<int> markerIds;

    detectMarkers(color_image, dictionary, markerCorners, markerIds, parameters, rejectedCandidates);

    if (markerIds.size() > 0) {
      aruco::drawDetectedMarkers(color_image, markerCorners, markerIds);
      for (int i = 0; i < markerIds.size(); i++) {

        cout << markerIds[i] << endl;

	float cx = color_w/2.0;
	float cy = color_h/2.0;

        Point po;
        po.x = int(cx);
        po.y = int(cy);
        cout<<"coor in color image: "<<"x= "<<po.x<<" y= "<<po.y<<endl;
        circle(color_image, po, 2, Scalar(0, 0, 255), -1); 

        float upixel[2];
        float upoint[3];
        upixel[0] = static_cast<float>(cx);
        upixel[1] = static_cast<float>(cy);

        float depth_pixel[2];

        rs2_project_color_pixel_to_depth_pixel(depth_pixel,
        reinterpret_cast<const uint16_t*>(frameset.get_data()),
        depth_scale,
        0.1,3.0,
        &depth_intrin,
        &color_intrin,
        &color_extrin_to_depth,&depth_extrin_to_color,upixel);

        Point pd;
        pd.x = int(depth_pixel[0]);
        pd.y = int(depth_pixel[1]);
        circle(colorized_depth_image, pd, 3, Scalar(0, 0, 0), -1);
        cout<<"coor in depth image: "<<"x = "<<pd.x<<" y = "<<pd.y<<endl;

        auto dist_to_object = depth.get_distance(depth_pixel[0], depth_pixel[1]);

        rs2_deproject_pixel_to_point(upoint, &depth_intrin, depth_pixel, dist_to_object);
        cout << "Point in depth: " << " x = " << upoint[0] << " y = " << upoint[1] << " z = " << upoint[2]<<endl;

        float to_point[3];
        rs2_transform_point_to_point(to_point, &depth_extrin_to_color, upoint);
        cout << "Point from depth to RGB: " << " x = " << to_point[0] << " y = " << to_point[1] << " z = " << to_point[2]<<endl;

        float color_pixel[2];
        rs2_project_point_to_pixel(color_pixel, &color_intrin, to_point);
        cout << "back to color pixel: " << " x = " << color_pixel[0] << " y = " << color_pixel[1] <<endl;

        cout<<endl;

        BP.cpoint.push_back(upoint[0]);
        BP.cpoint.push_back(upoint[1]);
        BP.cpoint.push_back(upoint[2]);

        rate.sleep();
      }
    } else {
      BP.cpoint.push_back(0.0);
      BP.cpoint.push_back(0.0);
      BP.cpoint.push_back(0.0);

      ROS_INFO_STREAM("NONE coordinate : " << " x = " << BP.cpoint[0] << " y = " << BP.cpoint[1] << " z = " << BP.cpoint[2]);
    }
    pub.publish(BP);

    rate.sleep();

    imshow("color_Image", color_image);
    imshow("colorized_depth", colorized_depth_image);
  }
}

`

MartyG-RealSense, 2021-09-24:
How near to the center of the image does the large drift in X occur?  There is a known issue with use of the **get_distance** instruction where measurements are relatively correct at the center of the image and become less accurate when moving towards the outer areas of the image.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/9088#issuecomment-847583702

xczzz, 2021-09-25:
@MartyG-RealSense Hello again. I made  several tests. When the pixel is in the center of the color image, the x is about 3cm difference to the ground truth in the  real world. The y is about 1cm and the z is about 2cm. The camera is within 1m distant to the object.

MartyG-RealSense, 2021-09-25:
You mentioned in your opening message that you calibrated the RGB.  Did you use the RealSense SDK's official **Dynamic Calibration** tool to do this?

xczzz, 2021-09-25:
The calibration is to define the transformation matrix betwewn RGB camera and robot arm. I did not use Dynamic Calibration.

xczzz, 2021-09-25:
Besides, I found that the the depth value calculated by "get_distance" is not accurate. There is an error about 1~2 cm. Should I use Dynamic Calibration first?

MartyG-RealSense, 2021-09-25:
RealSense cameras usually have very good calibration when new after purchase because they are calibrated in the factory.  They may need re-calibrating after a time though, especially if they have received a physical shock such as a hard knock, drop on the floor or have experienced severe vibration (e.g when mounted on a drone or on a vehicle travelling on rough ground such as agricultural land).  Could your camera have experienced anything like this?  If so then it would definitely be worth performing a Dynamic Calibration.

xczzz, 2021-09-25:
I am not sure. Maybe the camera happened something during the express delivery. I think it is a good attempt to calibration the camera again. I will try it. Thank you again. When I get some progress or I find new problems, I will update my issue.

xczzz, 2021-09-29:
Hi @MartyG-RealSense I re-calibrated the intrinsic parameters using a circle grid board. I found that the distortion coefficients of my re-calibration was different from the camera info. The coefficients of camera info are all zero while the coefficients of my re-calibration are not zero. They are 0.023 0.918 0.0015 0.003 -3.636. I wonder if they will impact the using of camera. And how can I update the coefficients in camera info?

xczzz, 2021-09-29:
Ok, I got it. @MartyG-RealSense Besides, there is a problem about fx and fy of intrinsic parameters. I got the fx and fy via my re-calibration. They are about 599. But the fx and fy are about 606 according to the camera info. (By the way, the ppx and ppy are nearly the same.) Will this problem impact the accuracy when detecting via visual method? If it impacts, how can I update them?

MartyG-RealSense, 2021-09-29:
ppx and ppy describe the pixel coordinates of the principal point (the center of projection), whilst fx and fy (which describe the focal length of the image as a multiple of pixel width and height) are allowed to be different but are commonly close.  This information is quoted from the Intrisics section of the SDK's Projection documentation.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#section-intrinsic-camera-parameters

The SDK's **rs-enumerate-devices -c** calibration mode demonstrates that ppx / ppy and fx / fy can change at different resolutions.  The image below is taken from a D435i camera model.

[image]

xczzz, 2021-09-29:
Ok, according to my understanding, I do not need to update the intrinsic parameters with my re-calibration value because the difference is really small. Is my understanding right?

MartyG-RealSense, 2021-09-29:
That is my understanding that you should not need to update the intrinsics.  It is worth bearing in mind that the **Dynamic Calibration** tool does not calibrate intrinsics and only calibrates extrinsics, because it is extrinsics that have the most impact on the image.

MartyG-RealSense, 2021-10-05:
Hi @xczzz  Do you require further assistance with this case, please?  Thanks!

xczzz, 2021-10-05:
@MartyG-RealSense No, you can help me to close the issue. Thank you again.

MartyG-RealSense, 2021-10-05:
Thanks very much @xczzz for the update.  As you suggested, I will close the case.  Thanks again!

closed 2021-10-05 by MartyG-RealSense