realsenseai/librealsense #9785: Intel RealSense L515 RGB-camera: No frames received

wjsprockel, 2021-09-21:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0172540|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003a89c65f4|
|**Firmware Update Id**|00000003a89c65f4|

Please provide a description of the probl
[image]
em

Dear All,
RGB-camera: No frames received
And I don't have a clue. Any help/suggestions what may be the problem?
Thanks in advance, Willy

MartyG-RealSense, 2021-09-21:
Hi @wjsprockel   Would it be possible to update your RealSense SDK version to a recent one such as **2.49.0** please, as 2.36.0 is over a year old now at the time of writing this.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

If you download and run the **Intel.RealSense.SDK-WIN10** automated installer file in the **Assets** list at the bottom of each SDK version's information listing then the recent SDK version will automatically overwrite the earlier 2.36.0 version on your computer so that you have an up to date version.

[image]
 
If you are able to update the SDK version, does doing so make a difference to your **No Frames Received** RGB problem?

wjsprockel, 2021-09-21:
Hi Marty,
Many thanks for your prompt reply. I will run the update tomorrow.
And I will come back to you with the outcome.
Have a nice evening, Willy

wjsprockel, 2021-09-23:
Hi Marty,
Updating both RealSense SDK and firmware had positve result. It's working like a charm.
I use the L515 to explain and demonstrate the differences and possibilities of the Lidar depth sensor and camera RGB/infrared sensor in training automotive technicians on ADAS. For about  a year physical training sessions were not possible due to Covic measures.
Please see attached screencap. Many thanks.
Best regards, Willy
[image]

MartyG-RealSense, 2021-09-23:
You're very welcome!  It's great to hear that you were successful.  Thanks for the update and the details of how you're using your camera   :)

MartyG-RealSense, 2021-09-29:
Hi @wjsprockel  Do you require further assistance with this case, please?  Thanks!

wjsprockel, 2021-09-29:
Hi Marty, everything is OK now. Thanks for your reminder.
I will close this issue. 
Have a nice day, Willy

closed 2021-09-29 by wjsprockel