realsenseai/librealsense #9786: Camera doesn't work more than once and "No Frames Received!" on Viewer Ubuntu 18.04

vigkrish123, 2021-09-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-84-generic )                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 

### Issue Description
Hi, I am using the above configuration for my RealSense D435i Camera for Video input purpose (such as OpenCV). The camera gives the video output the first time but after closing the application and opening the next time, it does not work. Similarly, RealSense viewer works well at first time and doesn't work the second time. Hence there is a pattern. 

Solutions tried:
1. Updated SDK
2. Changed USB slots
3. Updated all other version (eg OpenCv etc.)

Only time it works again is if i restart the system and even after that it works once and the problem continues.

Looking for a solution as I have been working on this issue for days trying to find a fix.

MartyG-RealSense, 2021-09-22:
Hi @vigkrish123  There are some past cases of this issue or similar occurring, examples of which are https://github.com/IntelRealSense/librealsense/issues/1889 and https://github.com/IntelRealSense/librealsense/issues/4655

In the RealSense Viewer, is a complete shutdown and restart of the **computer** required to clear the problem, or can it be resolved if you perform a reset of the camera?  The camera can be reset by going to the **More** option near the top of the Viewer's options side-panel and selecting the **Hardware Reset** option from its drop-down menu.

[image]

vigkrish123, 2021-09-23:
Hi, thank you for the reply. The problem is even at first trial, it says "connection time out" and no frames received after a certain time period. I cannot reset the hardware as you mentioned as I get the connection time out error when i try to reset. 

[image]

MartyG-RealSense, 2021-09-23:
Are you using the official 1 meter USB cable that was supplied with the camera or a longer USB cable of your own choice, please?

vigkrish123, 2021-09-23:
I am using the official one which came along with the camera. I also tried another cable thinking maybe the official one was faulty. But both gave me the same result

MartyG-RealSense, 2021-09-23:
Thanks very much for the information.  Which method did you use to install the RealSense SDK on your computer?

vigkrish123, 2021-09-23:
I followed the link from the homepage 
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md

I followed the steps provided in this link

MartyG-RealSense, 2021-09-24:
This problem has the characteristics of a program that has been improperly shut down after the first successful run, as described in https://github.com/IntelRealSense/librealsense/issues/1086

Are you closing the RealSense Viewer by left-clicking on the orange 'X' in the top corner, please?

vigkrish123, 2021-09-24:
The problem arises even when the program is running as it stops receiving frames after a time period or due to connection time out. When that happens , I am using the 'X' in the top corner to exit the application.

MartyG-RealSense, 2021-09-24:
Are you enabling only the RGB stream in the Viewer and your OpenCV program, or are you enabling three stream types at the same time (RGB, Depth, Motion Module)

vigkrish123, 2021-09-24:
I am enabling only RGB Stream in OpenCV (i also mapped the correct camera input i.e. in my case RGB is '/dev/video4') and in the Viewer all streams work but in any case, after a time period it stops functioning and doesn't work more than once

MartyG-RealSense, 2021-09-24:
If you open the **Controls** section of the 'Stereo Module' options in the Viewer and observe the ASIC Temperature and Projector Temperature read-outs, what values do they display?  Are they higher than 40?

Also, is the casing of the camera hot to the touch if you touch it after the camera has been running for a few minutes after starting from a cold state?

[image]

vigkrish123, 2021-09-27:
Sorry for the late reply.

The temperature is well under 40. Also the camera is not hot at all. It runs only for a few seconds before it says there is a connection time out and "No frames Received" error occurs

[image]

MartyG-RealSense, 2021-09-27:
Does the time-out after a few seconds still occur if the **Emitter Enabled** drop-down menu is first changed from 'Laser' to 'Off' before enabling the depth stream?

[image]

vigkrish123, 2021-09-28:
Yes. The time out takes place even if I turn off the laser before enabling the stream. Also while using it via OpenCV, it still stops after a time period and wont work the next time until camera plugged out and plugged in again.

MartyG-RealSense, 2021-09-28:
So the depth stream does seem to be updating correctly in the first few seconds after stream start before it time-outs?  If it does, this suggests that the camera hardware is okay and there is a **communications** problem.

vigkrish123, 2021-09-28:
Seems so. The camera works perfectly when it works. But cant get it working for a longer period. Any idea how I can solve the issue? Coz while using it for OpenCV, am just taking the camera directly as an input so there's nothing much I can do with any software updates to the realsense viewer.

MartyG-RealSense, 2021-09-29:
Have you got any other cameras apart from the D435i attached to the computer that might be conflicting with the D435i?  I noticed earlier that your RGB was using /video4.  On a one-camera computer I would normally expect the /video detection to be /video0 for depth, video1 for infrared and /video2 for RGB.

vigkrish123, 2021-09-29:
Nope the D435i is the only webcam attached to the PC and it shows 6 devices attached. /video0, video1... video5. I had to do a trial and error to find that RGB was video4.

I am currently planning on trying the D435i on a Jetson Nano to see if the problem is occurring in all machines. But I still want to figure out why it wouldn't work in the PC.

MartyG-RealSense, 2021-09-30:
Building the SDK from source code with the **RSUSB backend** method instead of building from packages would help to eliminate the possibility that there is a conflict with your Linux version or kernel version, as RSUSB is dependent on neither and does not require kernel patching.

vigkrish123, 2021-09-30:
Could you tell me how I can do this? I am fairly new to Linux and I found only V4L backend. If there is a installation documentation on the RSUSB Backend, please do refer me that

MartyG-RealSense, 2021-09-30:
If you are planning on using Nano then an RSUSB Backend installation guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 should be helpful.

In regard to an RSUSB Backend installation on PC, installation guidance can be found at https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

vigkrish123, 2021-10-01:
Thanks a lot. Does this mean I need to change a lot in the code to accommodate the fact that I am using RSUSB Backend instead of V4L? My code i.e. which uses the camera with openCV looks like this:

if 'webcam' in input_path:
        
        video_reader = cv2.VideoCapture('/dev/video4')
        fps = video_reader.get(5)
        
       
        width  = video_reader.get(cv2.CAP_PROP_FRAME_WIDTH)   # float
        height = video_reader.get(cv2.CAP_PROP_FRAME_HEIGHT)  # float
        out = cv2.VideoWriter('output.mp4', cv2.VideoWriter_fourcc(*'mp4v'), 5, (int(width), int(height)))

Sorry for being very specific as I am new to this and want to learn how the changes have an impact and where.

MartyG-RealSense, 2021-10-02:
The difference between builds made with V4L and RSUSB backends is mostly with processes in the background, and the same code can be used for both backends.  The back-end differences may result in some behavioural differences for certain kinds of function when the program is running.  

For example, an RSUSB build may check less often (every 5 seconds by default) whether device settings have changed, meaning that when the program makes a request to alter a setting, the program may miss that request.  The time between checks can be reduced with a code edit though, with the trade-off that CPU percentage usage increases.  This phenomenon is discussed in https://github.com/IntelRealSense/librealsense/issues/6921

vigkrish123, 2021-10-07:
Thanks a lot. I shall try this one. I will close this issue and reopen it if the solution does not work or get further problems.

closed 2021-10-07 by vigkrish123