realsenseai/librealsense #9787: L515 no longer works

sjmduncan, 2021-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   Win 10  | 
| SDK Version                            |  2.44.0 or 2.49.0                          | 

One of the L515 cameras we have has simply stopped working, we have a few others with the same firmware that work fine on the same computers.

With RealSense viewer 2.49.0 the camera is not listed as a source when it is plugged in, with the following log messages:

```
13:20:30 [Info]  - librealsense version: 2.49.0

13:20:30 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Framebuffer size changed to 176 x 1
13:20:30 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Window size changed to 176 x 1
13:20:30 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Scale Factor is now 1
13:20:31 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
13:20:31 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:308 - null pointer passed for argument "device"
13:20:31 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2621 - Couldn't refresh devices - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
13:20:31 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\win\win-helpers.cpp:550 - Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
13:20:31 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 0xfe)
13:20:34 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Framebuffer size changed to 3840 x 2066
13:20:34 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Window size changed to 3840 x 2066
```

With version 2.44.0 the camera is listed as a source, and the RGB camera and motion module work fine but the depth sensor does not start, with this error message:

```
Invalid Value in rs2_open_multiple(sensor:00000212BB6ECBB0,
profiles:00000212BB87AB28, count:2):

Failed to resolve the request:
	Format: Z16, width: 640, height: 480 	Format: Y8, width: 640, height: 480

Into:
	Formats:
	 Y8 	 Z16
```

and with these log messages.

```
13:35:22 [Info]  - librealsense version: 2.44.0

13:35:22 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2610 - Framebuffer size changed to 2168 x 1141
13:35:22 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2610 - Window size changed to 2168 x 1141
13:35:22 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2610 - Scale Factor is now 1
13:35:23 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:382 - Found 1 RealSense devices (mask 0xfe)
13:35:23 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\common\post-processing-filter.h:52 - Starting Face Detection : OpenVINO
13:35:23 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\tools\realsense-viewer\openvino-face-detection.cpp:109 - Loading CPU extensions...
13:35:23 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\openvino\rs-vino\object-detection.cpp:74 - Loading object detection model from: face-detection-adas-0001.xml
13:35:23 [Info]  - Intel RealSense L515 was selected as a default device
13:35:23 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\sw-update\http-downloader.cpp:143 - Download error from URL: https://librealsense.intel.com/Releases/rs_versions_db.json, error info: Couldn't resolve host name
13:35:23 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\openvino\rs-vino\age-gender-detection.cpp:68 - Loading age/gender model from: age-gender-recognition-retail-0013.xml
13:35:30 [Info]  - Starting streaming of Depth, Infrared...
13:35:30 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1416 - Request: Y8 Infrared, 
Resolved to: Y8 Infrared, 
13:35:30 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1416 - Request: Z16 Depth, 
Resolved to: Z16 Depth, Y8 Infrared, 
13:35:30 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - _reader->ReadSample(...) returned: HResult 0x80070018: "The program issued a command but the command length is incorrect."
13:35:30 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\l500\l500-depth.cpp:714 - Exception caught in l500_depth_sensor::open
```

MartyG-RealSense, 2021-09-23:
Hi @sjmduncan   As a starting point in investigating your case, could you first unplug the micro-sized end of the USB cable from the camera, turn it around to the reverse direction and re-insert it into the camera, please (USB Type-C cables are two-way insertion on the micro end).  Does this improve your results?

sjmduncan, 2021-09-23:
@MartyG-RealSense thanks for the quick reply.

Switching the orientation of the camera-end of the cable has no effect, nor does using a different cable, and we have only used the cables which were included with the cameras.

MartyG-RealSense, 2021-09-24:
Thanks very much @sjmduncan for the cable testing.  Next, could you test whether you can successfully run the **rs-capture** example program with your L515 in Windows using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

sjmduncan, 2021-09-26:
The program is non-responsive and then it just closes, with the following console output:

```
 27/09 10:19:37,036 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:37,070 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:37,104 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:37,267 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:37,305 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:37,468 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:37,624 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:37,780 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:37,936 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:42,776 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:42,928 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:42,964 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:43,116 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:43,153 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:43,300 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:43,452 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:43,604 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:43,784 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:48,608 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:48,760 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:48,795 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:48,943 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:48,979 ERROR [8780] (win-helpers.cpp:550) Could not find camera (vid 45e pid 2c4 uid 193e61f8) in windows device tree
 27/09 10:19:49,132 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:49,283 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:49,436 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:49,591 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:54,440 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
 27/09 10:19:54,592 ERROR [8780] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
RealSense error calling rs2_pipeline_start(pipe:000002B425143CF0):
    No device connected
````

MartyG-RealSense, 2021-09-27:
Is it possible to update the affected camera to firmware **1.5.8.1** in the RealSense Viewer?  Though your other cameras work on your computer with firmware 1.5.5.0, updating the firmware may clear any firmware related problems that are in this particular camera.

sjmduncan, 2021-09-30:
I updated the camera to 1.5.8.1 downloaded [from here](https://dev.intelrealsense.com/docs/firmware-releases-l500) using version 2.44 of the RS viewer since I can't interact with the camera at all using version 2.49. I still get the same error/log messages with both versions of the viewer and with `rs-capture`. The issue also persists if I re-install firmware version 1.5.5.0 (which is the recommended firmware for RS viewer version 2.44).

MartyG-RealSense, 2021-09-30:
And you have the same problems if you download and run the standalone **Intel.RealSense.Viewer.exe** version of the RealSense Viewer from the 2.49.0 'Assets' file list?  This file should be launched by double-leftclicking on it once downloaded and not by using the shortcut launch icon on the Windows desktop.

[image]

sjmduncan, 2021-10-01:
Yep, same issue with the standalone viewer (also the the standalone version of 2.44 is has the same issue as the one included with the SDK).

MartyG-RealSense, 2021-10-01:
I have highlighted your case to Intel to seek advice.  Thanks very much for your patience!

MartyG-RealSense, 2021-10-01:
Hi @sjmduncan  Intel have requested that you return the affected camera to the retailer that you purchased it from using that retailer's **Return Material Authorization (RMA)** procedure.

Could you also please email me the serial number of the affected camera at the address below:

martyx.grover@intel.com

sjmduncan, 2021-10-03:
Done, thanks for your help.

MartyG-RealSense, 2021-10-04:
I have received the serial number by email and passed it to Intel.  Thanks very much!

MartyG-RealSense, 2021-11-20:
Hi @sjmduncan  Do you have an update that you can provide about any progress in the process of returning your camera, please?  Thanks!

sjmduncan, 2021-11-25:
@MartyG-RealSense Yep it's been returned and we have received the replacement. Thanks.

MartyG-RealSense, 2021-11-26:
Thanks very much @sjmduncan for the update.  As you have received the replacement, I will close this case.  Thanks again!

closed 2021-11-26 by MartyG-RealSense