realsenseai/librealsense #9788: how to run python file on jetson nano

dohyeonYoon, 2021-09-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu 18.04.5 LTS  | 
| Kernel Version (Linux Only)    |  4.9.201 -tegra                                    | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            |                   2.49.0         | 
| Language                            | python                          | 
| Segment			|  others                          | 

### Issue Description
how to run python file on jetson nano? 

when I run the python file on miniforge prompt 
I encountered "no module named pyrealsense2" error 
how can I install pyrealsense2 module on jetson nano?

MartyG-RealSense, 2021-09-23:
Hi @dohyeonYoon  It looks as though miniforge is based on **Conda**.  This creates complications in regard to installing the Python wrapper, as the Python path on Conda may be different to the ones that are normally used with Python.

You may be able to find the correct Python path for Conda with the command **which python3**

dohyeonYoon, 2021-09-23:
https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation 

Looking at the above documentation, on my jetson ubuntu 18.04
I even verified that realsense-viewer works. But when I try to run my python file,
I get the message "no module name pyrealsense2". How can I install the pyrealsense2 module on my jetson?

MartyG-RealSense, 2021-09-23:
Let's verify next whether the pyrealsense2 wrapper has at least successfully installed on your Jetson.  If you navigate to the folder location **/librealsense/build/wrappers/python**, do you see there a set of files resembling the image below?

[image]

dohyeonYoon, 2021-09-24:
[image]
I only do this 3 step. Then I can play the realsense-viewer.

after your reply I download the SDK 2.0 source code zip file and unzip in downloads folder. 
but there are no "build" folder. 
[image]

[image]

what can i do next?

MartyG-RealSense, 2021-09-24:
It looks as though you installed using the Jetson instructions in the link below.

https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation

You confirmed that the RealSense Viewer is working for you, which suggests that the SDK installation process went okay.

There is no **build** folder in the source code folder by default though when it is first downloaded.  You have to create that folder yourself, and then when the RealSense Python wrapper is successfully built then the Python wrapper files should be in the sub-folder of build **/librealsense/build/wrappers/python**

Installing the SDK from Debian packages and then building the Python wrapper from source code is also complicated, as packages and source-code are two totally different installation methods.  There is also the additional complication of **miniforge** being used, since there are few references available for building the wrapper with Conda-related Python and none that I could find about miniforge specifically.  

Ultimately, the best chance of getting it working on Jetson may be to build both the librealsense SDK and the Python wrapper from source code at the same time using instructions for Jetson in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

dohyeonYoon, 2021-09-26:
according to #6964 (comment)'s direction 
In conclusion, the following file structure was obtained.

[image]

[image]

[image]

[image]

But still can't find the pyrealsense2 module. 
how can I do next?

MartyG-RealSense, 2021-09-26:
In your build > Python folder there are none of the files that would suggest that a build of the Python wrapper has completed successfully, such as the pyrealsense2.so files (unless miniforge has placed the files somewhere else).  So it is difficult to provide advice about it, though the fact that there are files there at all suggests that the build process has done something.

[image]

Did you use the Nano instructions in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-858226681 please?

Are you using Python 3.6, and if you are using a different Python 3 version then did you change the references to 3.6 in the installation guide to match the Python version number that you are using?

Did you edit and source the bashrc file?  I recently provided notes about editing and sourcing the file on Nano in points 5 and 6 of the instructions in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4405767306003/comments/4405978344083