realsenseai/librealsense #9791: Unable to configure stream on remote camera

colinmsnow, 2021-09-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-46-generic                                       | 
| Platform                                 | PC, Raspberry Pi 4 remote device  |
| SDK Version                            |  2.49.0 Local, 2.49.0 remote                         | 
| Language                            |  C++                      | 
| Segment			|  Robot                        | 

Both remote and local device were built from source from the latest sdk version.

### Issue Description
I have been following the tutorial: https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras and can view the depth and rgb images in realsense-viewer. However, I am unable to change the stream configuration or extract data from the stream outside of the viewer. With realsense-viewer open I am able to toggle both visual and depth streams, but when I click the triangle to the right of a stream the viewer crashes and prints the following:
```
 24/09 11:21:19,310 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Backlight Compensation control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Brightness control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Contrast control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Exposure control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Gamma control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Hue control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Saturation control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Sharpness control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add White Balance control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Enable Auto Exposure control for RGB Camera
 24/09 11:21:19,311 ERROR [140094442825472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 24/09 11:21:19,311 ERROR [140094534211328] (sensor.cpp:537) acquire_power failed: failed to set power state
 24/09 11:21:19,311 WARNING [140094534211328] (sensor.cpp:1126) Failed to add Enable Auto White Balance control for RGB Camera
RealSense error calling rs2_get_device_info(dev:0x562fa39ba0d0, info:Product Id):
    info Product Id not supported by the device!
```
[image]

I have seen in a few other issues such as #8812 that these options seem to be supported for network devices. Is this true? or is what I am seeing normal behavior? From what I can understand it seems like it is trying to connect over usb to get the settings which immediately fails because there is no usb device. I'm just not sure why it tries to do that.

This problem also seems to extend to applications outside the viewer as I am able to write a custom app that connects to the camera and reads information about it such as serial # etc. but when I try to enable a stream with some configuration, ex: `cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);` I am not able to read data from it or change the stream settings.

Other things I have tried:
1. I started with the provided image for raspberry pi and have also tried updating the sdk on the Pi to the latest version (2.49.0). This change caused the default frame size to change from 480p to 720p but otherwise seemed to have no effect (I still can not change the frame size).
2. Running the built-in gl-net example which fails to get any depth data. I am able to run the normal gl example when I plug the camera directly into my computer but something about the network link is breaking it.
3. Testing with the hello-realsense example and trying to instantiate a remote device. It finds the device and connects successfully, but never gets any depth data.
4. Using the prebuilt realsense-viewer package instead of the one I built. They both work exactly the same.
5. Using several different Raspberry pi's.
6. All the patches suggested [here](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md)
7. Using a different D435i camera.

Any help or advice would be greatly appreciated.
Thanks so much,
Colin

MartyG-RealSense, 2021-09-24:
Hi @colinmsnow  As a starting point in investigating your case, I will address some of the points raised in your questions above.

1.  Under the **Multi-Streaming Model** rules of the RealSense SDK, once a specific stream type on a specific camera has been accessed by an application then the application 'claims' that particular camera stream.  Until that stream is disabled and the claim on it released, another RealSense application cannot access that stream.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

For example, if the Viewer enabled depth first then another RealSense program such as rs-capture would not be able to access the depth stream.  It also works in the other direction - if rs-capture enabled the depth stream first then the Viewer would not be able to subsequently enable it until rs-capture was closed and the claim on the depth stream released.  The same would be true for your custom app.  If the Viewer started a stream first then the custom app could not access it, and if your app claimed it first then the Viewer could not access it,

2.  The problem of using cfg with a network connection has been experienced before in https://github.com/IntelRealSense/librealsense/issues/8423 and the RealSense user in that case shared their C++ solution in https://github.com/IntelRealSense/librealsense/issues/8423#issuecomment-795557815

3.  When a camera is set as a network camera then the resolution and FPS modes that it can access over a network connection are currently limited.  So if a program tried to stream depth with the D435i's default depth resolution of 848x480 because it does not contain cfg custom stream definitions then it would likely not work on a network connection because rs-server does not currently support 848x480 depth.  The supported resolution and FPS modes are listed in the ethernet paper's **Software Limitations** section in the link below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

rs-hello-realsense does not use cfg code and would use the D435i model's default stream configuration of 848x480 depth and 1280x720 RGB.

colinmsnow, 2021-09-26:
Thanks so much for the info, but it seems as If I was a little unclear on some of the points.

1. I am not trying to stream to multiple points at once. I am running only the viewer or a custom script separately and wanted to mention both because I think they may both be caused by the same problem. If I open up the viewer and connect to a network device I can stream video or depth data just fine, I just can't change the resolution or framerate (defaults to 1280x720 30fps) because when I press the triangle to the left of a stream the app crashes, regardless of what else is running. At first I thought this was just a limitation with the viewer and network devices, but I have seen several other screenshots of settings menus on remote devices that seem to be working just fine. Is this something that has happened before?

2/3. Thanks for the link, I actually already looked at that and it sort of worked. When I try to set up a stream without a config it does not work, as is to be expected from the last thing you mentioned. However, when I set up a stream that I know should exist (such as the one in the example), it successfully connects and streams, but the depth data is missing z values and the color data has no rgb values. I included my code below which combines rs-hello-realsense with the solution code from #8423 if you want to take a look, but I suspect there is a problem beyond that given that the viewer is not working entirely correctly either.

<details>
  <summary>Adaptation of rs-hello-realsense for network device</summary>
  
```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>
#include <librealsense2-net/rs_net.hpp>

int main(int argc, char * argv[]) try

{

    std::string server_ip = argv[1];    
    std::cout << "Connecting to " + server_ip << std::endl;
    rs2::net_device dev(server_ip);

    rs2::context ctx; 
    dev.add_to(ctx);
    rs2::pipeline p(ctx);
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    auto profile = p.start(cfg);
    std::cout<<"started" << std::endl;

    while (true)
    {
        // Block program until frames arrive
        rs2::frameset frames = p.wait_for_frames();

        // Try to get a frame of a depth image
        rs2::depth_frame depth = frames.get_depth_frame();

        // Get the depth frame's dimensions
        auto width = depth.get_width();
        auto height = depth.get_height();

        std::cout << "width: " << width << std::endl;
        std::cout << "height: " << height << std::endl;

        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(width / 2, height / 2);

        // Print the distance
        std::cout << "The camera is facing an object " << dist_to_center << " meters away \r" << std::endl;
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

```
</details>

Let me know if you have any thoughts or ideas and thanks again for the help,
Colin

MartyG-RealSense, 2021-09-27:
Thanks very much @colinmsnow for the clarification.  When you refer to the left-hand triangle, do you mean the one beside Stereo Module that expands open its options so that resolution and FPS can be selected?  And that the problem is that because it defaults to 1280x720 at 30 FPS and you cannot change the stream settings, it crashes when enabling the stream because a Network Camera connection supports just 6 FPS at 1280x720?

[image]

In regard to the adaptation of rs-hello-realsense for a network connection, in the **CMakeLists.txt** file of the project have you added realsense2-net to target_link_libraries as recommended by the networking paper when adapting an existing program for networking?

[image]

MartyG-RealSense, 2021-09-28:
Have you tried the camera network IP suggested by the networking paper, which is **10.0.0.99**

colinmsnow, 2021-09-28:
I just tried setting the IP to that address and it made no difference. I also tried running the viewer with the same SDK version on a friend's computer and it acts exactly the same.

colinmsnow, 2021-09-28:
UPDATE: Downgrading the local SDK version to 2.47.0 fixes all the above problems. The camera connects and settings can be changed and the custom code works as well. The remote device is still running the latest version (2.49.0). I have not tried 2.48.0 but either that or the newest version seems to break network devices, at least for my application. I do not anticipate needing features from the latest release so I will probably just stay at a lower version for now, but this seems like something that should be resolved if it is indeed a more widespread problem. Let me know if you want me to run any more tests or if you find a solution.

MartyG-RealSense, 2021-09-28:
Great to hear that you achieved a solution!

At the time of writing this, there have been no other reports of this problem occuring.  So whilst you do not need to perform further tests, this case will be an excellent reference if it occurs again for another RealSense user who is using networking.  Thanks again!

closed 2021-10-04 by MartyG-RealSense