realsenseai/librealsense #9793: This error occurred when the software was opened.

yunswj, 2021-09-26:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-09-26:
Hi @yunswj   Do you still experience this error if you download the standalone **Intel.RealSense.Viewer. exe** version of the Viewer from the **Assets** file list on the RealSense SDK's 'Releases' page?

https://github.com/IntelRealSense/librealsense/releases

[image]

Intel.RealSense.Viewer. exe should be launched with a double-leftclick on the .exe file at the location on your computer that you downloaded it to, and not with the RealSense Viewer launch icon on the Windows desktop.

yunswj, 2021-09-26:
[image]

I did download the application from the website you gave. I can still open it normally a few days ago. When I used it today, I reported this error. The software works normally on other computers.

I did download the application from the website you gave. I can still open it normally a few days ago. When I used it today, I reported this error. The software works normally on other computers.

MartyG-RealSense, 2021-09-26:
Is your computer able to run the **rs-capture** example program?  If you have installed the full RealSense SDK (**Intel.RealSense.SDK-WIn10**) then you can run a pre-built version of rs-capture in Windows using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

Flemmingcan, 2021-09-26:
gave it  up a while ago :>

yunswj, 2021-09-27:
@Flemmingcan 

Have you encountered this problem too?

yunswj, 2021-09-27:
@MartyG-RealSense My camera is R200, can I use SDK2.0?  I haven't compiled and run the complete SDK yet. I will test it today.

MartyG-RealSense, 2021-09-27:
Hi @yunswj  RealSense SDK 2.0 is not compatible with the old R200 camera model.  SDK 2.0's 1.0 predecessor, now known as Legacy Librealsense, is compatible with it though.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

yunswj, 2021-09-28:
That is, I can use SDK2.0 to program my R200, right? For example, you can use the MATLAB example in SDK2.0.

MartyG-RealSense, 2021-09-28:
If you have an R200 camera then it will not work with SDK 2.0.  May I please confirm that you have an R200, and not another model such as SR300 or D435, by showing you an image of what an R200 looks like.

[image]

yunswj, 2021-09-28:
Can you give me an installation package about SDK1.0? The resources found are all 2.0.

MartyG-RealSense, 2021-09-28:
SDK 1.0, now known as Legacy Librealsense, is at https://github.com/IntelRealSense/librealsense/tree/v1.12.1

Its page of example programs are at https://github.com/IntelRealSense/librealsense/tree/v1.12.1/examples

closed 2021-10-10 by MartyG-RealSense