realsenseai/librealsense #9795: External HW sync

Artur078, 2021-09-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.15.50 | 
| Operating System & Version |   Linux Raspian Debian 10 (Buster)  | 
| Kernel Version (Linux Only)    | 4.19.97                                         | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  2.42}                          | 
| Language                            |  C++                          | 
| Segment			| others                          | 

### Issue Description
For a simple setup, i want to operate 2 cameras with external HW Trigger from a light barrier  to capture only one synchronized frame pair with the syncer class. 
When I start the sensor with low-level sensor-API (depth_sensor.start(sync)) it continuously captures framepairs with ext_HW_Trigger_option in slavemode 3 (without triggering), and does not capture anything in genlock mode 4, ( even when triggered). It should capture one depth-frame and one synchronized color-frame with the trigger-signal.
(My own alternative solution with triggering the GPIO is not satisfactory, because of the huge time delay.)
Thank you very much.

MartyG-RealSense, 2021-09-27:
Hi @Artur078  The syncer section of the SDK's **Frame Management** documentation in the link below states that when the syncer class is used, the SDK tries to use hardware sync if it is enabled but does not automatically enable it.

https://dev.intelrealsense.com/docs/frame-management#section-frame-syncer

What code are you using to set the Inter Cam Sync Mode for the slave cameras in your script, please?

It is also important to note that a camera that is set to mode '3' is a **Full Slave**, whilst a camera using mode 4 or above is a **Genlock Slave**.  

[image]

So whilst a genlocked camera will indefinitely wait for a trigger instead of capturing constantly, that may not be the case for mode 3.

Also in mode 4+, the Slave cameras should be set to an FPS that is a multiple of the external trigger's FPS.  So if the external trigger is set to 30 then the slaves can be set to 60.  Are you using a higher FPS with your slaves in mode 4+ as recommended by the table below?

[image]

The External Synchronization paper should be considered experimental.  Unlike the original hardware sync paper (which covers modes 1 and 2 and is considered by Intel to be a validated and mature system), modes 3 and above introduced in the External Synchronization paper are not considered to be a validated system.

Artur078, 2021-09-27:
@MartyG-RealSense Thank you for the quick response. Here's the principle code. 

rs2::options* depth_options = (rs2::options*)&depth_sensor;
depth_options->set_option(RS2_OPTION_INTER_CAM_SYNC_MODE ,4);

depth_sensor.open(depth_stream_profile);
color_sensor.start([](rs2::video_frame f){});
depth_sensor.start(sync);
rs2::frameset fs=sync.wait_for_frames();

Camera framerate is set to 30fps. The Trigger is an undefined single 1,8 V Pulse.
May be if the Trigger didn't release the camera, there is some electronic issue.

MartyG-RealSense, 2021-09-27:
A difficulty with this scenario is that because genlock was an experimental feature, there is little documentation about use of RGB in mode 4+, and none about capturing RGB and depth together as far as I am aware.  I also do not know of any plans to add to its documentation in future.

I'm increasingly wondering whether using hardware sync is necessary for this particular application.  For example, the link below has a C++ script that saves a color image file from each attached camera simultaneously.  Conceivably, the saving of a depth image file could be added to the code too.

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

Alternatively, if the light barrier is being triggered constantly then RGB and depth could be recorded at the same time into a **bag** file (which is like a video recording of camera data) and image frames extracted from the bag file after the recording is made.

Artur078, 2021-09-28:
Yes, with using this solution (#2219) the sync class is not needed: But I still have the problem of knowing if the trigger really released a depth frame because the camera is continuously collecting color-frames .

MartyG-RealSense, 2021-09-28:
Some people use an 'if' statement to check whether a frame is of a particular type, such as **if frame.Is_depth_frame** and **if frame.is_color_frame** and then take a particular action if that frame type is detected.  An example of this is in the C++ script at https://github.com/IntelRealSense/librealsense/issues/4018#issuecomment-517878422

Artur078, 2021-09-28:
OK, in my case it is a bit more complex, since it is an unsorted frameset, but thank for your help.

MartyG-RealSense, 2021-09-28:
The SDK has the ability for users to define their own custom frameset instead of using the standard frameset by using a **custom processing block**.  Would that be useful to you if you have an unsorted frameset?

https://github.com/IntelRealSense/librealsense/issues/5847

MartyG-RealSense, 2021-10-04:
Hi @Artur078  Do you require further assistance with this case, please?  Thanks!

Artur078, 2021-10-04:
Hi @MartyG-RealSense it seems that the trigger can not be enabled in mode 4. I tested it with the Viewer , a signal generator and Osciloscope for all 6 Cameras.

MartyG-RealSense, 2021-10-04:
If you use Mode 4 (genlock) then the cameras should be connected together with self-created **sync cabling**, as described in the link below.  Do you have sync cabling joining your six cameras together and connecting to the signal generator, please?

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-3-camera-connections

Artur078, 2021-10-04:
I found out that the sync cable connectors caused the problem. The overlaying plasic edges of the connector case prevented an electrical contact of the pins. I cutted them off. Now it works with the Viewer.

MartyG-RealSense, 2021-10-04:
That's excellent news, @Artur078   :)

MartyG-RealSense, 2021-10-10:
Thanks very much @Artur078 for the update!  As you do not require further assistance, I will close the case.  Thanks again!