realsenseai/librealsense #9796: About the Measure tag in realsense-viewer

zma69650, 2021-09-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455| 
| Firmware Version                   |05.12.15.50 | 
| Operating System & Version |  win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |    2.45  | 
| Language                            |  C++    | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to implement this kind of label. In fact, I am new to OpenGL, and I did not find the corresponding function in measuremen.cpp. How is this implemented?
[image]

MartyG-RealSense, 2021-09-27:
Hi @zma69650  Have you seen the RealSense SDK's **rs-measure** C++ example program that goes through the code step by step in a tutorial?

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

[image]

The RealSense SDK also provides a pre-built version of this example once the SDK is installed.  In Windows you can find it on your computer at the SDK folder location below and launch the file with a double-leftclick on **rs-measure.exe**

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**

[image]

zma69650, 2021-09-27:
thank you for your reply，I try to understand

closed 2021-09-27 by zma69650