realsenseai/librealsense #9797: SDK example hello-realsense doesn't run

simon88, 2021-09-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265  | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | Intel NUC  |
| SDK Version                            |  2.49.0                         | 
| Language                            | C++                         | 
| Segment			| SDK example                          | 

### Issue Description

I have downladed the SDK here [https://github.com/IntelRealSense/librealsense/releases](url) v2.49.0 on Win10 computer, i have visual studio community 2019 Version 16.11.2. WHen I try to Debug hello-realsense SDL example I have an exit code (0x1), no errors everything compile, and this is strange beceause for example ar-basic example works well.
So I've searched and the program stopped to run at this line 
      `auto width = depth.get_width();`
      `auto height = depth.get_height();`

MartyG-RealSense, 2021-09-27:
Hi @simon88   T265 does not have built in depth sensing.  There is an example program in the link below that emulates depth sensing using the fisheye sensors but it is in the Python language and there is not a C++ equivalent.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

simon88, 2021-09-27:
Ah nice thank's @MartyG-RealSense :)

closed 2021-09-27 by simon88