realsenseai/librealsense #9802: PLY vertices data should be on a single line per vertex

Alexxkrehmen, 2021-09-29:
### Issue Description

As shown in the PLY file format (http://gamma.cs.unc.edu/POWERPLANT/papers/ply.pdf) all the data for each vertex should be on a single line.
Adding a carriage return after each property makes the file unusable in some third party applications (see here : https://developer.blender.org/T88745)

The fix seems easy to do, in librealsense/include/librealsense2/hpp/rs_export.hpp, between line 240 to 260, remove all the carriage returns added after each property, and only put one at the end of the line, like this :

```
for (size_t i = 0; i <new_verts.size(); ++i)
{
    out << new_verts[i].x << " ";
    out << new_verts[i].y << " ";
    out << new_verts[i].z << " ";

    if (mesh && use_normals)
    {
        out << normals[i].x << " ";
        out << normals[i].y << " ";
        out << normals[i].z << " ";
    }

    if (use_texcoords)
    {
        out << unsigned(new_tex[i][0]) << " ";
        out << unsigned(new_tex[i][1]) << " ";
        out << unsigned(new_tex[i][2]) << " ";
    }
    out << "\n";
}
```

MartyG-RealSense, 2021-09-29:
Thanks very much @Alexxkrehmen for sharing your method!   If you would like it to be added officially to the RealSense SDK, I would recommend submitting the change yourself as a **Pull Request** so that it can be considered by the RealSense team for inclusion in the SDK.

https://github.com/IntelRealSense/librealsense/pulls

Alexxkrehmen, 2021-09-29:
Pull Request  posted :)

MartyG-RealSense, 2021-09-29:
Thanks so much - I have added the **Enhancement** label to this case as a reminder that it should be kept open whilst the Pull Request is active.  Thanks again!

Alexxkrehmen, 2021-10-03:
> Thanks so much - I have added the **Enhancement** label to this case as a reminder that it should be kept open whilst the Pull Request is active. Thanks again!

I've closed my Pull request since it wasn't right ... I Can't find how to create one properly, either it includes many other requests I don't want or I just simply cannot create it because I don't have the right to do it :(

Alexxkrehmen, 2021-10-03:
@MartyG-RealSense In fact I just can't create a new branch... That's why I'm stuck.

MartyG-RealSense, 2021-10-04:
Great, thanks very much @Alexxkrehmen   :)

ideasman42, 2022-04-23:
Any updated on this? The fix seems trivial and the issue has been reported twice in Blender:

- https://developer.blender.org/T92130
- https://developer.blender.org/T88745

Alexxkrehmen, 2022-04-23:
> Any updated on this? The fix seems trivial and the issue has been reported twice in Blender:
> 
>     * https://developer.blender.org/T92130
> 
>     * https://developer.blender.org/T88745

I tried to send the fix : https://github.com/IntelRealSense/librealsense/pull/9819/commits
but my commit is still pending... I don't know why...

MartyG-RealSense, 2022-04-23:
Thanks very much @Alexxkrehmen for your response to @ideasman42 - it appears that there is a build check failure on your pull at https://github.com/IntelRealSense/librealsense/pull/9819/

Normally a pull provides information about the specific aspects of the automated build check process that have failed but in this particular instance, clicking the **Details** option on this pull results in a broken link.

[image]

[image]

Alexxkrehmen, 2022-04-23:
Yes... So I strictly have no idea on how to fix this problem...

MartyG-RealSense, 2022-04-23:
You could perhaps create a new pull with the same fix and close the old one.

ideasman42, 2022-04-23:
Force pushing to your branch should re-run the tests too.

Alexxkrehmen, 2022-04-23:
Sorry, I thought I had made a mistake so I did the request twice >_<
This should be ok to review now :)

MartyG-RealSense, 2022-04-23:
Thanks very much, @Alexxkrehmen 

I am not involved in pull request processing, so I cannot unfortunately provide a time estimate for when a reviewer from the RealSense team may be assigned to your pull.

In the meantime, RealSense users should be able to apply your fix themselves by editing rs_export.hpp using your instructions at the top of this case.

MartyG-RealSense, 2026-10-25:
Case closed due to the Pull Request (PR) associated with it having been already closed.

closed 2026-10-25 by MartyG-RealSense