realsenseai/librealsense #9806: Request to push patches to main line Linux kernel sources

mlederhi, 2021-09-30:
Intel RealSense team, please consider getting your changes back into the mainline Linux kernel. It is really cumbersome if everybody has to do it on it's own and every time again when the upstream kernel sources change. It would be really a plus for a product if it works out of the box.

Sincerely,

Martin Lederhilger
DS Automotion GmbH
Austria

MartyG-RealSense, 2021-09-30:
Hi @mlederhi  At https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 there is a discussion of the advantages and disadvantages of using an RSUSB installation of librealsense (also known as libuvc backend) versus a patch-based installation.  A quote from that information regarding a patch-based installation that may be relevant to your question is as follows:

"All kernel patches should eventually get upstreamed into Linux or abandoned.  From the user's perspective the main obstacle in accepting the kernel patch model is the deployment scheme that requires certain engineering skills level.  We address this by by wrapping and redistributing the patches with DKMS Debian package, at least for Ubuntu LTS kernels".

Installations that use the RSUSB Backend method are not dependent on Linux versions or kernel versions and do not require patching.  As highlighted in https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604  though, RSUSB does have disadvantages that may make it unsuited for some projects.

MartyG-RealSense, 2021-10-08:
Hi @mlederhi  Do you require further assistance with this case, please?  Thanks!

mlederhi, 2021-10-08:
Hello, thanks for your reply. I'm currently busy with other stuff, but I have read in different posts, that patching should not be required with recent Kernels anymore.

I have Kernel version

> Linux version 5.10.27-gentoo-x86_64 (root@wslederhi2) (x86_64-pc-linux-gnu-gcc (Gentoo 10.3.0 p1) 10.3.0, GNU ld (Gentoo 2.35.2 p1) 2.35.2) #3 SMP PREEMPT Wed Sep 29 16:11:43 CEST 2021

but when I run

> rs-enumerate-devices

I receive

> No device detected. Is it plugged in?

I have installed the udev rules, and when plugging the camera in I can see in dmesg the following output

> [87948.358201] usb 2-4: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
> [87948.380164] usb 2-4: New USB device found, idVendor=8086, idProduct=0b5c, bcdDevice=50.c7
> [87948.380168] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [87948.380169] usb 2-4: Product: Intel(R) RealSense(TM) Depth Camera 455 
> [87948.380170] usb 2-4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455 
> [87948.380172] usb 2-4: SerialNumber: 117623060223
> [87948.406382] hid-generic 0003:8086:0B5C.0008: hiddev96,hidraw4: USB HID v1.10 Device [Intel(R) RealSense(TM) Depth Camera 455  Intel(R) RealSense(TM) Depth Camera 455 ] on usb-0000:03:00.0-4/input5
> [87948.416278] videodev: Linux video capture interface: v2.00
> [87948.421718] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
> [87948.422422] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
> [87948.435003] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb2/2-4/2-4:1.0/input/input24
> [87948.435682] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
> [87948.435685] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
> [87948.452283] usbcore: registered new interface driver uvcvideo
> [87948.452284] USB Video Class driver (1.1.1)

I can even watch the video streams for example with "Qt V4L2 test utility" (but only once for each /dev/videoX device, then I have to unplug and replug the camera again).

What Kernel options are actually required for using the cameras? Maybe I am missing something.