realsenseai/librealsense #9809: [UNITY] [Android] no depth and color stream

TheBricktop, 2021-09-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.12.15.50) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android. Samsung Galaxy S8 |
| SDK Version                            |  { 2.49}                          | 
| Language                            |  {unity }                          | 
| Segment			|  {Smartphone/VR/AR}                          | 

### Issue Description
While depth and color stream is being properly binded to the raw image texture and rendered on screen in the example scene for depth and color stream the same example on my samsung galaxy is not working.
The OS is prompting me with permission to record and recognizes the camera model, then starts the ir projector so its not the camera problem.
Considering that after update to latest firmware the RS Camera app on the Play Store has started to throw errors on camera initialisation and is not displaying any image apart from the camera info i suspect that the android support for realsense is having some problems with the last release of the firmware or the sdk.

MartyG-RealSense, 2021-09-30:
Hi @TheBricktop  Problems with using examples and the RS Camera app on the Samsung Galaxy S8 have previously been reported in https://github.com/IntelRealSense/librealsense/issues/4216

Within that discussion, Android wrapper developer **matkatz** suggests in https://github.com/IntelRealSense/librealsense/issues/4216#issuecomment-503239684 to disable all streams and then enable only depth at 640x480 and 30 FPS

TheBricktop, 2021-09-30:
@MartyG-RealSense  ok but why the UNITY build acts the same ? It sometimes start the color feed and sometimes not, it looks like some timeout problem for camera .
Ive confirmed that this is not the issue with cable or the camera, nor with the phone as other apps are streaming ok.
(after testing the rs camera app does start to render colored point cloud after disabling all streams and exiting the advance mode, guess it is working now )

MartyG-RealSense, 2021-09-30:
I see in https://github.com/IntelRealSense/librealsense/issues/6898#issuecomment-662589319 that you have already studied the details of the George Adamon project for using Android with RealSense and Unity.

Given that you were able to now use the camera with the RS Camera app, are you still experiencing problems with streaming the camera to Android using Unity?  If so, have you tested whether earlier SDK versions such as 2.45.0 have the same problem?

TheBricktop, 2021-10-01:
Yes, im building in unity 2020.3.8 and tried librealsense 2.45 to emulate what was working before.
For now ive just barely managed to run realsense and get color stream but i have to kill the process that used realsense before, disconnect, reconnect and then start the unity app again to get some results.
Right now binding to the rawimage texture is not working at all, only binding to the material works but only for color, i cant get the depth stream to show. Im using uniwersal render pipeline and unlit materials . 
While at it im not sure how does the coloriser works, if im putting it in the processing pipeline it does not change the depth stream color on the pc, its still displayed as red gradient on the material. Im sure thats why there is a custom shader that does that but why not just colorise it in the pipeline and then use the output to further use in shaders? 
I need depth to my ar app, and while everything works on the pc editor i cant make it run on the android device.

MartyG-RealSense, 2021-10-01:
I would speculate that the Unity wrapper was designed with the original **Built-In** render pipeline in mind rather than the URP and HDRP pipelines.  

I use HDRP rather than URP in my own projects, but I have noted that if you simply change the shader from its original setting such as Standard to a more modern setting such as Unlit then the texture does not render.  The material has to go through a conversion process.

Even after conversion I have had a small number of materials that remain unrenderable pink though.

The Unity documentation link below describes how to convert materials made for Built-In for a URP project.  You should make a backup copy of your Unity project folder before doing the conversion though.

https://docs.unity3d.com/Packages/com.unity.render-pipelines.universal@7.0/manual/upgrading-your-shaders.html

I note though that you have tried Built-In, without complete success.  and also, the graphics render correctly on a PC version of your Unity application though, but do not render correctly when used with Android, yes?

A simple solution that has been suggested in a past Unity forum case for unrenderable pink materials has been to reset the shaders.

http://answers.unity.com/answers/1381876/view.html

In regard to the colorizer's design in Unity, I am not involved in development activities and so do not have influence over that.

TheBricktop, 2021-10-05:
@MartyG-RealSense  I've managed to get the depth texture but using the rspipeline with colorizer and setting the rstexture render to rgb.
The other problem is that it is VERY SLOW even on the topmost android devices (tested on samsung s21 5g 8gb ram ). The speed of the frame processing highly depends on the depth texture (that is set to 640 x 480 in the stream settings). The RsPipeline consist of two nodes : RsAlign and RsColoriser set to the whitetoblack. 
Are these nodes hard to compute ? How to improve performace of it?

MartyG-RealSense, 2021-10-05:
Alignment is a processing-intensive activity in general, not only in Unity.  This is why RealSense applications that can make use of graphics acceleration functions (such as CUDA on computers with an Nvidia graphics GPU) benefit from lower CPU usage when performing point cloud generation, alignment or color conversion.  This is because the CPU work for those functions is offloaded onto the GPU.

In https://github.com/IntelRealSense/librealsense/issues/6076#issuecomment-601081541 the suggestion is made to use Unity's **compute shaders** to improve performance if you are able to offload work to the GPU, or alternatively to use Unity's **Burst Compiler** and **Jobs** systems if you have to use the CPU.

TheBricktop, 2021-10-05:
The native RS camera app still manages to get a lot more frames than unity wrapper. Is it because unity has a lot more overhead than native java app?

MartyG-RealSense, 2021-10-05:
Unity can have processing overhead when playing an application in edit mode that it does not have in the full **Build and Run** mode because a full run does not have the editor tools also running in the background.

It is also worth bearing in mind that in Unity the camera data is being visually rendered by converting and applying it to a Unity material.

TheBricktop, 2021-10-05:
From what i've read the actual overhead is pretty negligible when the actual texture is already stored in GPU, so probably the one bottleneck is pumping all these bits through the CPU but the rendering part is pretty fast on simple shaders,  also my application is run on device with full build without debug. 
The other thing is the alignment node, what does it actually do? Is there a feature tracking involved to properly match the depth and color stream? Or is it based on some mathematic transformation with known sensor parameters?