realsenseai/librealsense #9811: camera connected as usb 2.1

Albert950207, 2021-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435I } | 
| Firmware Version                   | (05.12.15.50) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 20.04)|                                   | 
| Platform                                 | PC  |
| SDK Version                            |  { v2.49.0}                          | 

### Issue Description
Camera connected as 2.1. I am sure my port is 3.0 and I tried it on other computer and it is still 2.0. I did pull it out and pull it into quickly but it still doesn't work. The cable is from with camera together so it could not be the reason. I also updated the firewall. It still doesn't work. Could anyone please help ? Thank you very much !

Albert950207, 2021-10-01:
Btw, I tried them on both Ubuntu 20.04 and win 10. Both are usb 2.1. I can received rgb and infrared pictures in Ubuntu 20.04 but I cant received motion module pictures. Once I opened it, I even cant receive rgb and infrared pictures. However, in the windows 10, I cant received one of pictures.

Albert950207, 2021-10-01:
anyone of pictures*

MartyG-RealSense, 2021-10-01:
Hi @Albert950207  Does the camera remain identified as USB 2.1 if you reset the camera in the RealSense Viewer?  This can be performed by going to the **More** option near the top of the side-panel and selecting the **Hardware Reset** option from its drop-down menu.

[image]

****************

If you are activating three stream types at the same time in the RealSense Viewer including the Motion Module and experiencing errors such as **No Frames Received** on one or more of the streams, this is a known issue.  It is triggered by enabling the Motion Module, and there is not a fix to correct it in the Viewer.  

It does not matter which order the three streams are enabled in, because enabling the Motion Module is the trigger for the problem.  If you use the Motion Module then I would therefore recommend only enabling two stream types (Depth and Motion Module, or RGB and Motion Module, etc).

When programming your own script there are two workarounds for this issue - creating two separate pipelines (with IMU on one and depth & RGB on the other) or by using **callbacks**.

The three-stream issue does not occur in the SDK's **rs-multicam** example program, which can be used to display depth, RGB and Motion Module for a single D435i camera.  On the Windows version of the SDK you can find a pre-built launchable .exe version of rs-multicam at **C: > Program files (x86) > Intel RealSense SDK 2.0 > Tools**.   On Ubuntu it should be in the **usr/local/bin** folder if the SDK tools and examples have been created by the SDK installation process.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

[image]

It is also worth bearing in mind that on a USB 2.1 connection the **Infrared 2** (right-side infrared) cannot be accessed, only **Infrared** (left-side infrared).  Infrared 2 requires a USB 3 connection.

Albert950207, 2021-10-01:
@MartyG-RealSense  Thank you very much for your replying. For usb 2.1 problem, I also did reset the camera(Hardware) in realsense viewer but it didnt work either. For the streams issue, now I see what happen on Unbuntu but on windows, I only enable one stream type(maybe rgb or maybe infrared) and I still cant get any frames. What leads to this ?

MartyG-RealSense, 2021-10-02:
Which method did you use to install the RealSense SDK on Windows, please?  Was it:

1.  With the **Intel.RealSense.SDK-WIN10** installer file in the 'Assets' list of the SDK 'Releases' page?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

2.  Or did you install the SDK on Windows from source code using CMake?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

[image]