realsenseai/librealsense #9813: get_timestamp returning a timestamp that is less than the previous frame timestamp

GrubbyHalo, 2021-10-01:
**Camera Model**          D430
**Firmware Version**      05.12.11.00
**OS**                             Raspbian 10
**Kernel**                       5.10.25-v7l
**Platform**                    Rasberry Pi 4
**SDK Version**             2.45.0

I have the following piece of code that runs during a frame callback

```
else if (f.get_profile().stream_type() == RS2_STREAM_DEPTH)
{

        char* d = (char*)vf.get_data();
        double fts=f.get_timestamp();
        if(fts < last_fts){
            DebugString("Current frame timestamp is smaller than last.. current fts= " +
                                            FloatToString(fts)+" last fts="+FloatToString(last_fts));
        }
        last_fts=fts;
}

```
And after about 30 seconds I start getting the message that the current frame timestamp is smaller than the last frame timestamp

Is it possible for frames to come out of order ?

MartyG-RealSense, 2021-10-01:
Hi @GrubbyHalo  Advice provided in https://github.com/IntelRealSense/librealsense/issues/7837#issuecomment-754970222 by a RealSense team member about 'old' frames being returned may be helpful.

GrubbyHalo, 2021-10-01:
Hi @MartyG-RealSense 

The issue referenced in [#7873](https://github.com/IntelRealSense/librealsense/issues/7837#issuecomment-754970222) seems to be slightly different. The problem there seems to be that the timestamp of the actual call to get_timestamp is being compared to the frame timestamp. Also that users snippet of code is calling ```wait_for_frames``` as opposed to a callback being used to retrieve and process the frame. 

In my code above, I am comparing the frame time stamp of subsequent frames and finding a discrepancy between successive frames. 

I was under the impression that if a frame is not processed by an application that it is dropped or does the backend  buffer frames and only send the very latest frame and then start sending older frames ?