realsenseai/librealsense #9814: bird eye view with L515's depth

hyebin-rdv, 2021-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515} | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-77-generic     | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Can I use bird eye view in a L515?
I plan to use 3d depth data, not 2d.

MartyG-RealSense, 2021-10-01:
Hi @MartyG-RealSense  You could certainly achieve a straight-down or tilted at an elevation view with an L515 lidar depth camera.  If you are planning to use the camera outdoors or in an indoor area with sunlight then an L515 would not be a suitable choice, and a 400 Series stereo depth camera such as the **D455** model may be more suitable.  

An L515's optimal depth sensing range is 4 meters or below, whilst a D455 has a longer range and has optimal depth sensing accuracy at 6 meters distance (compared to 3 meters on a D435 / D435i model).

hyebin-rdv, 2021-10-04:
[image]

Then, can I use a depth camera to figure out that the rails are all the same? 

I tried filtering with 2d, but the desired result didn't come out.

 I want to detect only rails using depth data.

MartyG-RealSense, 2021-10-04:
If you are testing with an L515 camera then it should be positioned at least 25 cm (0.25 meters) away from the rails in order for depth to be rendered, as the L515 has a minimum depth sensing distance of 25 cm.

As mentioned earlier, the L515 is not suited for outdoor use due to the potential for sunlight interference, though this may not be as significant a factor if the camera is pointing permanently downward towards the ground.

Applying the L515's **Short Range** configuration preset may improve the depth image in this particular scenario.  This preset can be applied in the RealSense Viewer or with script code if you are creating your own application.  If ROS is used then the preset should be settable as a dynamic reconfigure parameter.

If the camera is in a fixed position constantly, always observing the same set of rails, then you could perhaps capture a 3D point cloud with the RealSense camera and then use a RealSense-compatible open source tool called **CloudCompare** to compare a current pointcloud image with a previous image to see how the current image has changed over time.

https://www.danielgm.net/cc/

If the camera has to move along a rail line then another approach would be to record the depth data as a **bag** file (which is like a video recording containing depth data).  The bag file can then be played back and unlike a video, all of the depth data is preserved so that a RealSense program can use the bag as though it is a live depth stream.  A YouTube video in the link below where a camera is mounted to a high speed car on a highway provides a way to imagine how this might look.

https://www.youtube.com/watch?v=OwJmCyAn3JQ&t=23s

A RealSense project where the camera was mounted on a rail and moving took the approach of combining ROS with a digital **encoder** on the motor that moved the camera so that images could be captured at specific encoder positions at different positions along the rail route.

https://github.com/IntelRealSense/realsense-ros/issues/1581