realsenseai/librealsense #9816: rs-dnn does not work with L515

Uchihasasugey, 2021-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (realsense viewer v2.49.0) | 
| Operating System & Version |   {Win (10)  |                                       | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {object-tracking }                          | 

### Issue Description
pressed f5 to run the rs-dnn, it builds successfully but theres a bunch of errors

![Screenshot (
[image]
1)](https://user-images.githubusercontent.com/91737248/135622397-eef0b720-f181-42ee-ad2d-fb0661f9c6fc.p
[image]
ng)

MartyG-RealSense, 2021-10-01:
Hi @Uchihasasugey  Whilst the L515 does not support RGB resolutions 640x360 and 640x480 unless firmware 1.5.8.1 is installed in the camera, this is likely not the cause of this case's problem.  This is because although the rs-dnn program performs depth-color alignment, it does not specify a particular resolution / FPS configuration for depth and color.  The program should therefore use the default depth and color stream configurations for whichever RealSense depth camera model is attached.

Let's check the OpenCV wrapper and example building process first.  Did you build the librealsense SDK from source code on Windows and then select the DNN example from the 'CV' examples folder and press F5, as described in the instruction page below?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

Uchihasasugey, 2021-10-01:
[image]
I tried build the OpenCV and librealsense SDK like this, but I noticed when I search for "shared", it looks like there is something missing for OpenCV, even thought the configuring is done
[image]
For librealsense SDK, I actually have errors building,
[image]

[image]
CMake 3.21.3

MartyG-RealSense, 2021-10-01:
I see that when searching for 'shared', the **Advanced** option on your interface is unticked, whilst on the image on the instruction page Advanced is ticked.  Do you see the full list of settings when searching for 'shared' if a tick is placed in the Advanced box, please?

[image]

It may also be useful to refer to guides in the links below for installing OpenCV and then the RealSense SDK and OpenCV wrapper on Windows.

https://jessicayin.com/tutorials/#/opencv-34-installation/
https://jessicayin.com/realsense-opencv-installation

Uchihasasugey, 2021-10-02:
[image]
After following the guide, the build is successful, but I am not too sure where to find the missing dll?

MartyG-RealSense, 2021-10-02:
**opencv_dnn** is apparently a module of OpenCV relating to DNN functions, and '3415' likely refers to OpenCV version **3.4.15**.  Do you know whether 3.4.15 is the OpenCV version that you have installed?

Uchihasasugey, 2021-10-02:
I have installed OpenCV 3.4, directly from the website https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv, do I need to find opencv 3.4.15 then build it using CMake?

MartyG-RealSense, 2021-10-02:
The discussion in the link below about another RealSense user's problems with setting up rs-dnn may be helpful.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000714302--Problems-running-opencv-rs-dnn-realsense-camera-examples

Another link describes the particular LNK1104 error that you encountered with opencv_calib, though with an earlier version of OpenCV.

https://stackoverflow.com/questions/24297145/link-fatal-error-lnk1104-cannot-open-file-opencv-calib3d245d-lib-opencv-2/24297480

Uchihasasugey, 2021-10-03:
After deleting the previous OpenCV 3.4 and OpenCV 3.4.15 from the directory and redownloading,configuring and generating with OpenCV 3.4.15, then pressing ctrl+shift+B, I get errors
[image]
If I ignore the errors and proceed to generating the librealsense, then pressing F5 on rs-dnn, I get these errors
[image]
Do I go back to OpenCV 3.4?

MartyG-RealSense, 2021-10-03:
Are you able to uninstall the OpenCV versions that you have installed if you follow the procedure below?

1.  Type **programs** into the search box at the bottom of your Winows screen, next to the button in the corner where you shut down the PC.  This should allow you to find the **Add or remove programs** tool.

[image]

2.  Scroll down the list of programs to see if you can find entries for the installed OpenCV versions.  Then left-click on a program and click on the **Uninstall** button that appears.

[image]

Uchihasasugey, 2021-10-04:
I am unable to delete from using search, as I cannot find opencv in the lists of apps. Instead I had to delete manually from the directory containing it , C:/git/opencv, where I had copied the opencv zip file and unzipped there.
After deleting all versions of opencv, I downloaded the exe, then did the same steps in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv but the opencv library did not build successfully, with the same errors again 
[image]
Is there something wrong with my git or directory? The errors seem to be the same

MartyG-RealSense, 2021-10-04:
If you have already successfully installed OpenCV using the .exe installer, may I confirm that you are not installing OpenCV a second time afterwards by building it from source code with Git.  Instead, are you skipping to the **Building the OpenCV Samples VisualStudio project** section to install the librealsense OpenCV examples?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#building-the-opencv-samples-visualstudio-project

Uchihasasugey, 2021-10-04:
I thought I had to build it from source with Git as well. Now after skipping to building the Opencv samples, the configuring of librealsense was successful, but when I tried to add entry OpenCV_DIR, it says that I am missing a cmake file
[image]
I managed to find the .cmake previously when building Opencv 3.4  from source, however this time I cannot find it in Opencv 3.4.15

MartyG-RealSense, 2021-10-04:
It looks as though the guide is assuming that OpenCV was installed using Git earlier in the first half of the guide, hence the suggested path to OpenCV_DIR being C:/git in the second half of the guide.

 When installing OpenCV with the .exe file, I found that the path to the .cmake files will depend on the path that you specified to extract OpenCV to.  For example, the default download location for the installer file on Windows is the **Downloads** folder of your user directory.  So if you run the file from this location then it will suggest installing the **opencv** folder in the Downloads folder of your user directory by default.

[image]

The **OpenCVConfig.cmake** file should therefore be located in the **opencv/build** folder.

[image]

So my understanding of the instructions in the red error text is that you place the OpenCVConfig.cmake file in a directory of your choosing and then set OpenCV_DIR to that directory.

Uchihasasugey, 2021-10-05:
My librealsense configured and generated successfully, although I am unable to compile due to these errors
[image]
are these normal or did I do something wrong during the buildling?

MartyG-RealSense, 2021-10-05:
When I see an SDK library file name that ends in 'd' such as **realsense2d.lib** then it suggests to me that librealsense was built with CMake without CMAKE_BUILD_TYPE being set to **Release**.  Otherwise, librealsense can build in Debug mode, which can cause problems.

Is the **CMAKE_CONFIGURATION_TYPES** option settable to **Release** if you click the 'Configure' button on the CMake interface?

[image]

Uchihasasugey, 2021-10-11:
I got this same error after building with OpenCV 3.4.15. It is the error that occured when I built with Opencv 3.4

[image]

how should this error be fixed?