realsenseai/librealsense #9820: Match corresponding exported RGB and Depth images

blckmessiah, 2021-10-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   {Win10} |                                      
| Platform                                 | PC  |
| SDK Version                            |  { 2.45.0  }                         
| Language                            |  {Python }                          | 

### Issue Description
Hello, 

So i have extracted several RGB and Depth images from a .bag file that i have recorded using the D435i camera. The filenames of the RGB and Depth images don't seem to have any correlation. What i mean is that i can't see a way to determine which RGB image relates to each Depth image and vice versa. Is there maybe a way to do this? 

Thanks in advance

MartyG-RealSense, 2021-10-04:
Hi @blckmessiah  If it is possible for you to use the SDK's **rs-convert** bag file extract tool (if you have not done so already) then it can convert bag frames to depth and color images and use the frame timestamp in the filename.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

The Windows version of the SDK has a pre-built version of rs-convert that you can launch in the Windows **Command Prompt** interface.

https://support.intelrealsense.com/hc/en-us/community/posts/360051132573-Can-someone-help-me-on-how-to-get-rs-convert-exe-

blckmessiah, 2021-10-05:
Hello MartyG, 
So i have already extracted them using the rs_convert tool. The filename looks like this "name_Depth_1626539024069.83154296875000" for example. Which part of this is the timestamp? 

I would also like to ask something else. Is there a way to create a 4-dimensional array consisting of the 3 dimensions of the RGB images and 1 dimension that represents the distance value of each pixel? It would be very nice if you could give me some guidance on this!

MartyG-RealSense, 2021-10-05:
All of the number that comes after 'name_Depth_' would be the frame timestamp, the most important part of the value being the first half before the decimal point in the middle.  Their naming by numbers will likely mean that by default they will be listed in the folder that they were extracted to in the correct order from first to last.

I carefully researched your question about 4D arrays but could find no past examples of this in a RealSense project, unfortunately. 

Would storing the RGB and depth data in Python numpy arrays and aligning them suit your needs, as demonstrated in the link below?

https://stackoverflow.com/questions/67506025/intel-realsense-align-depth-and-color-in-two-numpy-arrays/69252861#69252861

MartyG-RealSense, 2021-10-06:
Thanks very much @blckmessiah for the update - I look forward to your next report.

MartyG-RealSense, 2021-10-16:
You are very welcome.  If you have finished with this case, please feel free to close it using the **Close Issue** button under the comment writing box.  Thanks again!

berylyellow, 2021-11-08:
@blckmessiah Hi~ Have you solved this problem? I could not find the correlation between Depth image and RGB image, it seems frames drop.  Could you please help me? Thanks.

blckmessiah, 2021-11-11:
Hello @berylyellow , i am sorry but i could not solve my problem. But to my knowledge there are some frames that are dropped during conversion i believe. Sorry i cannot be of any assistance.

woshicaobin, 2021-12-07:
> 嗨@blckmessiah 你有关于这个案例的更新，你可以提供吗？谢谢！

您好，请问您后面解决了深度图像数据的问题吗？

woshicaobin, 2021-12-07:
> 您好 MartyG， 所以我已经使用 rs_convert 工具提取了它们。例如，文件名类似于“name_Depth_1626539024069.83154296875000”。这是时间戳的哪一部分？
> 
> 我还想问点别的。有没有办法创建一个由RGB图像的3维和表示每个像素的距离值的1维组成的4维数组？如果你能在这方面给我一些指导，那就太好了！

您好，请问您解决了您深度图像数据的问题吗

MartyG-RealSense, 2021-12-07:
The above comment is a translation of the message earlier in this discussion in https://github.com/IntelRealSense/librealsense/issues/9820#issuecomment-934178997 with the added comment to @blckmessiah that says "Hello, have you solved the problem of your depth image data?"

blckmessiah, 2021-12-07:
@woshicaobin Hello, sorry but i have not. This approach kind of seemed like a dead end to me. 

@MartyG-RealSense Thanx for the translation!