realsenseai/librealsense #9822: L515 Depth Sensor Not working

YILWU-1, 2021-10-05:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1181820|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003ba980935|
|**Firmware Update Id**|0003ba980935|

Please provide a description of the problem

As soon as I connect the L515 and turn on the Depth Sensor in RealSence Viewer, i get the following one error
02:38:20 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:514 - Fatal error occurred (16)
[image]

MartyG-RealSense, 2021-10-05:
Hi @YILWU-1  This problem with **Fatal error occurred (16)** has been discussed previously in https://github.com/IntelRealSense/librealsense/issues/9329 and has been reported on Windows, MacOS and Ubuntu operating systems.  An official bug report about it has already been filed with Intel.  In the meantime, I will seek advice from Intel about your case.  Thanks very much for your patience.

YILWU-1, 2021-10-07:
Thank you very much. I've emailed you.

> Hi @YILWU-1 Intel have requested that you return your camera to the retailer that you purchased it from, using the **Return Material Authorization (RMA)** product returns procedure of that particular retailer.
> 
> Also, could you please email me the serial number of your L515 camera. You can find the serial number by clicking the **Info** option near the top of the RealSense Viewer's options side-panel. The email address to send the serial number to is:
> 
> [martyx.grover@intel.com](mailto:martyx.grover@intel.com)

MartyG-RealSense, 2021-10-07:
Thanks very much, I have received your serial number and passed it to Intel.

MartyG-RealSense, 2021-11-20:
Hi @YILWU-1   Do you have an update that you can provide about any progress in the process of returning your camera, please?  Thanks!

MartyG-RealSense, 2022-01-22:
Hi @YILWU-1   This case will be closed after a week from the time of writing this if we do not receive further comment.  Thanks!

closed 2022-01-28 by MartyG-RealSense