realsenseai/librealsense #9825: high latency on xaiver nx if align depth enabled

manuelvolk, 2021-10-05:
----------------------------------------------------------------------------------------------------

| Required Info                          |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D455 (FRAMOS D455e) | 
| Firmware Version                    | 05.12.13.50 | 
| Operating System & Version  |   Nvidia Jetpack R32.6.1 (based on Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra | 
| Platform                                 | Nvidia Jetson Xavier NX  |
| SDK Version                           |  2.45.10                          | 
| Language                               |  C++ & realsense2_camera (ROS) v2.3.0                          | 
| Segment			        |  Robot                          | 

### Issue Description

The Problem we encounter currently is already perfectly described in its symptoms by the following issue: [9519](https://github.com/IntelRealSense/librealsense/issues/9519)
(Why I open a new issue, because the issue not document any solution for the problem)

If we just launch rs_camera.launch the RGB images are received with no latency and the system
is only under a moderate / expected load.

If we launch the node with rs_aligned_depth.launch (so align_depth parameter is set to true) the rgb and
aligned depth stream show a significant latency and the system is under high load (as described
in issue  [9519](https://github.com/IntelRealSense/librealsense/issues/9519)).

Also as depcited in issue 9519 the effect is only present when the realsense-ros ros driver is
used. The realsense-viewer does not show the effect (is align_depth used here?).

To give you a little bit more context, here are the instructions which are executed for device bring-up:

```
#!/bin/bash

# install latest nvidia software updates
sudo apt-get update
sudo apt-get upgrade -y
sudo apt-get install -y nvidia-jetpack
 
# create symlink to fix the issue that
# nvidia-jetpack installs opencv 4 but 
# ros melodic expects opencv 3
sudo ln -s /usr/include/opencv4/ /usr/include/opencv
 
#install ros noetic
ROS_DISTRO=melodic

#install required packages
sudo apt-get install -y --no-install-recommends \
    git \
    cmake \
    build-essential \
    curl \
    wget \
    gnupg2 \
    lsb-release \
    ca-certificates \
 

sudo apt install -y libglfw3-dev libglew-dev libgtk-3-dev libssl-dev libusb-1.0-0-dev pkg-config  \
                 libgl1-mesa-dev libglu1-mesa-dev dkms libboost-dev \ 
                 libjsoncpp-dev libtclap-dev libeigen3-dev
 
sudo rm -rf /var/lib/apt/lists/*
 
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
 
sudo apt-get update
 
sudo apt-get install -y python3-catkin-tools
sudo apt-get install -y ros-${ROS_DISTRO}-desktop-full
 
echo "source /opt/ros/${ROS_DISTRO}/setup.bash" >> ~/.bashrc
source ~/.bashrc
 
# get driver for framos d455e camera
wget -N https://www.framos.com/framos3d/D400e/Software/Latest/FRAMOS_D400e_Software_Package_Linux64_ARM_latest.tar.gz
 
# unpack framos files
tar xpvf FRAMOS_D400e_Software_Package_Linux64_ARM_latest.tar.gz
 
# install FRAMOS camera suite and driver
sudo apt install -y ./FRAMOS_D400e_Software_Package/FRAMOS_CameraSuite*.deb
sudo apt install -y ./FRAMOS_D400e_Software_Package/FRAMOS-librealsense2*.deb
 
rm -rf FRAMOS_D400e_Software_Package*
 
cd ~/Desktop
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone https://github.com/pal-robotics/ddynamic_reconfigure.git
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros
git checkout 2.3.0
cd ../../
catkin build
echo "source ~/Desktop/catkin_ws/devel/setup.bash" >> ~/.bashrc
 
sudo apt-get autoremove -y
 
reboot 
```

I can also confirm, that this effect is still present after a clean installation
of Jetpack 4.6 and the execution of only the above listed steps. 

Are there any other differences to issue: [9519](https://github.com/IntelRealSense/librealsense/issues/9519)?
Yes, my guess is that FRAMOS not compile the librealsense with the cuda compile flag.

Any advise how to analyse or solve the problem is highly appreciated!

@ProtossDragoon as autor of issue: [9519](https://github.com/IntelRealSense/librealsense/issues/9519) could you perhaps provide additional information
on how you solve the problem?

manuelvolk, 2021-10-07:
Hi @MartyG-RealSense,

ok I have contacted FRAMOS now. In parallel I will try to investigate the problem with my D455 (Intel). Any advice on the best way to proceed?

ProtossDragoon, 2021-10-07:
Hi @manuelvolk,  I just reinstalled entire system several times with few changes (e.g. change build parameters or package/SDK installation order B->A instead of A->B ...) and it fixed the issue at a certain moment. 😭  
I tried to figure out the reason of the problem, but I couldn't. I'm very sorry I can't help you.