realsenseai/librealsense #9827: Why the playback of a bag are so different between the viewer and python opengl examle?

hwy1992129, 2021-10-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Operating System & Version |   Win 10   |                                      | 
| Platform                                 | PC |
| Language                            |  python                       | 

### Issue Description
I record a video as a bag via RealSense Viewer.
I modify the code and load it into the example pyglet_pointcloud_viewer_read.py. I don't think I modify anything about the filter. I got pointcloud as shown below.
[image]
It differs from what I see in RealSense Viewer.
[image]
When I use Open3D to create a point cloud from the bag, the point cloud is the same like the one in the first image.
[image]
How RealSense Viewer clean those noisy points? I have disabled the post-processing in RealSense Viewer. The point cloud doesn't have those noisy points.
I modify the configuration in the Viewer to record a video for better performance in the reconstruction, but the point cloud displayed in the viewer is incorrect as what it should be? Does the Viewer run an outlier removal?

hwy1992129, 2021-10-06:
@MartyG-RealSense 
> May I first confirm that the images provided above are shown in the correct order.
> 
1. The images are shown in the correct order.
> The first multi-colored image is stated to come from pyglet_pointcloud_viewer_read.py
> 
2. I modify pyglet_pointcloud_viewer_read.py, so that it can **load the *.bag file**.
> In regard to the pointcloud spray, could you test in the Viewer whether you achieve a cleaner image if you select the Default preset from the drop-down preset menu?
> 
3. There is no such preset after I load the bag file in Viewer. I am using L515.
[image]
All the results in my images are screen-captured after I load **the same bag file**. I only care about the pointcloud spray. I want to know why the pointcloud spray is gone in the Viewer.

I am sure when I was recording the bag. The pointcloud spray is not displayed in the Viewer, either. I modify the post-processing sharpening, pre-processing sharpening, noise filtering and other parameters for better data collection. I record the data as a bag when they look good. I also load the bag back to Viewer to recheck it, and it looks good. However, when the bag is loaded into the modified version of pyglet_pointcloud_viewer_read.py, and Open3D, I find the pointcloud spray....

MartyG-RealSense, 2021-10-06:
If you have modified the pyglet_pointcloud_viewer code to use a bag file as its data source instead of a live camera stream, I wonder if you have also set an option called **set_real_time** to False, as the default setting will be True.  The instruction should be placed directly beneath the **get_device().as_playback()** line of the script.  For example:

**playback = profile.get_device().as_playback()
playback.set_real_time(False)**

A Python bag playing script that demonstrates set_real_time = false is at https://github.com/IntelRealSense/librealsense/issues/6340

hwy1992129, 2021-10-06:
@MartyG-RealSense I tried it... It doesn't work......

hwy1992129, 2021-10-06:
@MartyG-RealSense The configuration parameters is shown below, when I doing the record....
[image]

MartyG-RealSense, 2021-10-06:
None of the settings applied in the RealSense Viewer will affect your Python application, as they are separate programs that do not interact with each other.  Therefore, equivalents of the Viewer settings would need to be deliberately programmed into the Python scripting.

hwy1992129, 2021-10-06:
@MartyG-RealSense OK, I see. The problem is caused by the setting in RealSense Viewer. I uncheck Use GLSL for Rendering and Use GLSL for Processing, and get the same result as the python scripts.
[image]
This setting make the pointcloud look nice, but it doesn't what the actual pointcloud is.

closed 2021-10-06 by hwy1992129