realsenseai/librealsense #9829: [realsense2_camera_node-1] [WARN]: Hardware Notification:Depth stream start failure,1.6335e+12,Error,Hardware Error

semi-kim, 2021-10-06:
i got error : [realsense2_camera_node-1] [WARN]: Device 912112070403 is connected using a 2.1 port. Reduced performance is expected.

my specific is 
UP Squared board (amd64)
ROS2 Galactic
Ubuntu 20.04.3

i followed all of instructions

when i typing ros2 launch realsense2_camera rs_launch.py initial_reset:=true

`
[INFO] [launch]: All log files can be found below /home/semi/.ros/log/2021-10-06-14-39-10-840438-semi-UPN-APL01-5069
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5071]
[realsense2_camera_node-1] [INFO]: RealSense ROS v3.2.2
[realsense2_camera_node-1] [INFO]: Built with LibRealSense v2.49.0
[realsense2_camera_node-1] [INFO]: Running with LibRealSense v2.49.0
[realsense2_camera_node-1] [INFO]: Device with serial number 912112070403 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3/1-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO]: Device with port number 1-3 was found.
[realsense2_camera_node-1] [INFO]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN]: Device 912112070403 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO]: Resetting device...
[realsense2_camera_node-1] [INFO]: Device with serial number 912112070403 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3/1-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO]: Device with port number 1-3 was found.
[realsense2_camera_node-1] [INFO]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN]: Device 912112070403 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO]: getParameters...
[realsense2_camera_node-1] [INFO]: setupDevice...
[realsense2_camera_node-1] [INFO]: JSON file is not provided
[realsense2_camera_node-1] [INFO]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO]: Device Serial No: 912112070403
[realsense2_camera_node-1] [INFO]: Device physical port: /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3/1-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO]: Device FW version: 05.12.15.50
[realsense2_camera_node-1] [INFO]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO]: Align Depth: Off
[realsense2_camera_node-1] [INFO]: Sync Mode: Off
[realsense2_camera_node-1] [INFO]: Device Sensors: 
[realsense2_camera_node-1] [INFO]: Stereo Module was found.
[realsense2_camera_node-1] [INFO]: RGB Camera was found.
[realsense2_camera_node-1] [INFO]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]: num_filters: 0
[realsense2_camera_node-1] [INFO]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN]: Could not set param: stereo_module.auto_exposure_limit with 0Range: [1, 165000]: parameter 'stereo_module.auto_exposure_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN]: Could not set param: stereo_module.auto_gain_limit with 0Range: [16, 248]: parameter 'stereo_module.auto_gain_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN]: Could not set param: rgb_camera.power_line_frequency with 3Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[realsense2_camera_node-1] [INFO]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[realsense2_camera_node-1] [INFO]: setupPublishers...
[realsense2_camera_node-1] [INFO]: setupStreams...
[realsense2_camera_node-1] [INFO]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN]: 
[realsense2_camera_node-1] [WARN]: Hardware Notification:Depth stream start failure,1.6335e+12,Error,Hardware Error
`

MartyG-RealSense, 2021-10-06:
Hi @semi-kim  May I first ask if you are able to test the camera in the **RealSense Viewer** program to see whether streaming can be started, please?  Thanks!

Also, can you confirm whether you are using a USB 2.1 connection, or whether the camera is plugged into a USB 3 port but is being mis-detected as 2.1?

semi-kim, 2021-10-07:
> There was another recent case at IntelRealSense/realsense-ros#2068 where a user of the ROS2 Galactic wrapper had the depth stream start failure error. Their problem was solved after following installation advice provided by Doronhi the RealSense ROS wrapper developer in IntelRealSense/realsense-ros#2068 (comment)

MY depth camera is D435
not D455

MartyG-RealSense, 2021-10-07:
The model number would not matter.  The problem in that case was in the installation.

semi-kim, 2021-10-08:
i follwed your comment that [ IntelRealSense/realsense-ros#2068 (comment) ](https://github.com/IntelRealSense/realsense-ros/issues/2068#issuecomment-918812871)

but realsense-ros2 didn't run

MartyG-RealSense, 2021-10-08:
The **roslaunch realsense2_camera rs_camera.launch** launch instruction quoted in the comment above is a ROS1 launch instruction for ROS Kinetic, Melodic or Noetic.  ROS2 - which includes Galactic - uses **ros2 launch realsense2_camera rs_launch.py**, as mentioned in your opening message.

Can you confirm what errors if any you receive if you now launch with rs_launch.py

MartyG-RealSense, 2021-10-12:
A particular ROS wrapper version should be matched as closely as possible with a specific librealsense SDK version listed in the release notes of that wrapper version.   Going through your case from the beginning, I note that you are using librealsense **2.49.0** with ROS2 wrapper **3.2.2**.  The 3.2.2 wrapper should be matched with librealsense **2.48.0**.

There is not a ROS wrapper version for SDK 2.49.0.  My understanding at the time of writing this is that the next supported librealsense SDK will be the upcoming **2.50.0** version, as described by Doronhi in https://github.com/IntelRealSense/realsense-ros/issues/2093#issuecomment-939776086

I would therefore suggest installing SDK 2.48.0 and then building ROS2 wrapper 3.2.2.

MartyG-RealSense, 2021-10-13:
The RealSense ROS2 wrapper does officially support Galactic in version 3.2.2, yes.

[image]

If you launch RViz2 with **ros2 run rviz2 rviz2** are you able to view topics successfully despite the control_transfer returned warnings?  If so then you can likely ignore the warning messages