realsenseai/librealsense #9846: Rosbag Camera Info viewing and Multicam

parthapn98, 2021-10-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.15.50) | 
| Operating System & Version | Linux (Ubuntu 16.0) | 
| Kernel Version (Linux Only)    |  05.12.15.50                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.49                      | 
| Language                            |  {python }                          | 
| Segment			|  {AR, MultiCam }                          | 

### Issue Description
1. How can i query the serial number from a bag file recording? {I can get the serial number from a device if its connected, i.e. rs.context.devices, but our setup requires us to id a camera after the recording was done to know the recording viewpoint.

2. Our additional cameras are still en route, so I havent tested this yet. But is it possible to record multiple cameras, either in a single or separate bag files, from the realsense viewer or in python script using enable_record_to_file()

Thanks!

MartyG-RealSense, 2021-10-19:
https://github.com/IntelRealSense/librealsense/issues/7932 provides an example of Python code for skipping the first 5 frames during playback in the script under the **Loading Code** heading.  You could certainly change '5' to '3'.

```
# Skip 5 first frames to give the Auto-Exposure time to adjust
for _ in range(5):
pipe.wait_for_frames()
for i in range(num_frames - 5):
print(i)
frameset = pipe.wait_for_frames()
```

I see no reason why the same code could not be used in the recording part of your application too to skip the first few frames when the pipeline is started.

MartyG-RealSense, 2021-10-19:
Great to hear that you developed a solution, @parthapn98 - thanks very much for the update!

parthapn98, 2021-11-02:
Hi @MartyG-RealSense,

are there any solved issues on bag file errors recorded with a python script?
Specifically regarding to timestamps of the frames.
The recording duration of the bag is fine, but then it jumps to a timestamp that very high "35xxxx.."
and the cause of the issue appears random
some bag files are fine and without issues and some do. We have been trying to debug this for a while with usb resets and trying to ensure we donot attempt to start streaming on an already streaming device. No luck yet.

Any suggestions?

MartyG-RealSense, 2021-11-02:
You should not be able to start a recording of a particular stream on a particular camera if recording of that stream has already started, as that camera stream will already be 'claimed' by the active recording.  You might receive an error that the camera is busy or it 'failed to set power state' if you try to start a new pipeline after one has already been started though.

Checking if a stream is already busy is not straightforward.  The error generated by a busy state is typically handled by catching the error in an exception check so that the program does not crash.  An example of exception catching in Python is highlighted in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L49

You could also have greater control over which camera is recording to which file by creating a separate pipeline and unique cfg configuration for each camera, manually providing the exact serial number for each camera instead of letting the program automatically detect the serial numbers and build a list of devices.  A Python example of this is in https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337

Have you also tried playing the bag file back in the RealSense Viewer and observing the timestamps to see whether the jump occurs in the Viewer playback?  If it does, this could indicate that there is a problem in the recording rather than in your Python code.

parthapn98, 2021-11-02:
>  The error generated by a busy state is typically handled by catching the error in an exception check so that the program does not crash.

Yes This is how we decide to either start or abort the recording, or just let the script crash on pipe.start

> You could also have greater control over which camera is recording to which file by creating a separate pipeline and unique cfg configuration for each camera, manually providing the exact serial [...]

We do a combination of these, by first creating a list of serial numbers and then create a list of pipes, profiles and configs each while using the serial number list

```python
for i in range(len(serials)):
        pipe[i] = rs.pipeline()
        conf[i] = rs.config()
        conf[i].enable_device(serials[i])

        if All:
            conf[i].enable_all_streams()
        else:

            if Depth:
                conf[i].enable_stream(rs.stream.depth, wd, hd, rs.format.z16, fps_depth)
            if IR1:
                conf[i].enable_stream(rs.stream.infrared, 1, wd, hd, rs.format.y8, fps_depth)
            if IR2:
                conf[i].enable_stream(rs.stream.infrared, 2, wd, hd, rs.format.y8, fps_depth)
            if RGB:
                conf[i].enable_stream(rs.stream.color, w, h, rs.format.rgb8, fps)
            if Motion:
                # conf[i].enable_stream(rs.motion_stream, -1)
                conf[i].enable_stream(rs.stream.accel)
                conf[i].enable_stream(rs.stream.gyro)

        # conf[i].enable_all_streams()
        conf[i].enable_record_to_file(out[i])
        # prof[i] = pipe[i].start(conf[i])
```

The problem is less frequent if we use option ALL, however its never completely gone.
Its much more frequent with the other settings

The remaining relevant part of the code is
```python
for i in range(len(serials)):
        prof[i] = pipe[i].start(conf[i])
        sensor[i] = prof[i].get_device().first_depth_sensor()
        sensor[i].set_option(rs.option.emitter_always_on, 1)
        record[i] = prof[i].get_device().as_recorder()
        rs.recorder.pause(record[i])
# delay warmuo
for i in range(len(serials)):
        rs.recorder.resume(record[i])
# delay recording duration

#--------------------- we tried this as optional for a fix, it did not resolve the issue
# for i in range(len(serials)):
#        rs.recorder.pause(record[i])  # pause before closing stream
##delay pause 
#--------------------

for i in range(len(serials)):
        pipe[i].stop()
        conf[i].disable_all_streams()
del pipe
    del prof
    del conf
    del record
    del sensor
```
The problem is faced even if we write out a separate short script to record from just one camera instead of auto detecting like this.

> Have you also tried playing the affected bag files back in the RealSense Viewer by drag-and-dropping the bag file into the Viewer's central panel and observing the timestamps to see whether the jump occurs in the Viewer playback? If it does, this could indicate that there is a problem in the recording rather than in your Python code.

We confirmed the timestamp issue with realsense viewer itself and not rqt or python bagreadeer
The recording for the assigned duration runs fine with proper fps display, but the end of file is marked at value "35xxxx.." seconds, and the timestamp at the end of the record duration suddenly jumps to this point. Indicating there are no frames between the end of recording of the bag and this specific time value.
the time value "35xxxx.." is actually specific and does not vary, Im writing from memory so don't remember what it was exactly

MartyG-RealSense, 2021-11-03:
I wonder whether the incorrect timestamp on the final frame is a consequence of how a bag file is 'indexed' by the SDK when a recording is ended by closing the pipeline.

Lookng at the code, the line **conf[i].disable_all_streams()** after the pipe stop instruction is likely to be unnecessary as all of the streams that are currently active on a pipeline should be automatically stopped anyway when the pipeline is closed.

I think it's a good idea that you implemented a pause instruction before the pipe stop.

parthapn98, 2021-11-03:
> I wonder whether the incorrect timestamp on the final frame is a consequence of how a bag file is 'indexed' by the SDK when a recording is ended by closing the pipeline.

Is there any alternative ways to end the recording or any recommendations on how to proceed here?
The scripts are designed for users to not require intricate knowhow of bag/ros and pyrealsense

MartyG-RealSense, 2021-11-03:
It looks as though out of the three bags playing simultaneously in the Viewer, one is playing normally and the other two have end times that are completely wrong and that is somehow affecting the visual updating of the images.  The timelines of the affected bags progress at a normal time rate that is the same as the working bag (even though the images are ot updating) and then both start and end times become wrong after the jump to the final frame.  

Do the bags play normally in the Viewer if played individually instead of three at the same time?  If so then this might suggest that the computer that the bags are being played on cannot cope with the processing demands of a simultaneous 3-bag playback.

When you say that the bags were recorded seconds apart, do you mean sequentially - that an active recording was stopped and then a new recording to a different file was rapidly begun?

The **rosbag record** bag recording function of ROS provides an automated means to do this called **--split**, where you can define a maximum recording duration or file size, after which the bag file recording is closed and a new file starts recording seamlessly from that point onwards.  So you could set the recording running and let rosbag record's split command take care of the start and stop process during recording.

[image]

parthapn98, 2021-11-03:
> (even though the images are ot updating)

The images are updating sorry there's not much distinguishing movement, I didnt realise it till after. But if you maximise and look closely the corresponding movement and update of frames are seen

> Do the bags play normally in the Viewer if played individually instead of three at the same time? If so then this might suggest that the computer that the bags are being played on cannot cope with the processing demands of a simultaneous 3-bag playback.

Yes they are identical when played back individually. I did em simultaneously to record them in one mp4. We have done record+playback on multiple tiers of devices to rule out hardware causes. And as mentioned even with just a single camera the issues are present

> When you say that the bags were recorded seconds apart, do you mean sequentially - that an active recording was stopped and then a new recording to a different file was rapidly begun?

I run my bagrecorder script N number of times, some outputs are with glitched timestamps, some are fine.
No rapid start and stop of file writing

MartyG-RealSense, 2021-11-03:
So the script's code seems to be okay if it can record correctly sometimes.  It would seem to be a question of stability then and reproducing recording results consistently.

It may be worth shutting down and rebooting the computer inbetween each individual recording test to eliminate the possibility that the script is malfunctioning when run again after the previous test.  This can occur with RealSense programs if the program was not properly closed down after the previous launch of the program.

parthapn98, 2021-11-05:
@MartyG-RealSense i ran some tests, the scripts ability to create a bagfile without the issue in reference isnt dependent on when its being run. i.e. not all first runs are usable.

On a different topic. are there any existing code for recording cameras in ros?

MartyG-RealSense, 2021-11-05:
Some ROS users have created Python projects to activate and control the **rosbag record** bag recording tool of ROS.  An example of a ROS node script in the Python language that was published in 2020 is in the link below.  

https://gist.github.com/vbschettino/13c1d0102fa184811b31397cb9a4e85d

parthapn98, 2021-11-05:
Thanks @MartyG-RealSense  Ill check it out. Before i leave for the weekend: is there a way to use rs-server for standard linux based computers not a raspi?

Ive got librealsense built, rs enumerate shows device status and rs-server shows server started.
Since its not a raspi, i donot set static ip. I can view the currently assigned ips of all devices connected to it, on my mobile, where the hotspot is hosted.
However when a second computer using realsense viewer on the same wifi enters the ip of the streaming computer, it gives the error
out of range value for argument "def"

If i try to add network device without running rs-server on host it gives this:
[getControls] error: Connection to server failed: Connection refused - -107

MartyG-RealSense, 2021-11-05:
rs-server should be able to be compiled on any Linux computer, not just a Raspberry Pi.  The open-source ethernet white-paper uses a Pi 4 board by default but states that the project can be adapted with minor modifications for other computing brands, citing **Up Board** as an example.

The design of the open-source networking system is that you can have multiple **remote** boards that have a camera attached to them, but there is a single **host** central computer to receive the feeds from the remote cameras.

The design of the librealsense SDK is that once an application has 'claimed' access to a particular stream on a particular computer then that stream cannot be accessed by another application (such as an instance of the Viewer on a second host computer).  Multiple applications can access different streams on the same camera, but two applications cannot access the same camera stream at the same time - the application that originally claimed the stream must disable that stream to release the claim on it and make it available to other applications to access.

The access rules are described in the SDK's **Multi-Streaming Model** described n the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

parthapn98, 2021-11-05:
> The design of the librealsense SDK is that once an application has 'claimed' access to a particular stream on a particular computer then that stream cannot be accessed by another application (such [...]

Yes I expected the same from experience with the RS camera behaviours. If you open up a particular stream, that cannot be accessed by anything else.

However, the original stream is not open anywhere, neither the receiver PC or the client where rs-server is started. 

* I start rs-server on PC1

* Open realsense-viewer on PC2 and try to connect network device, I get: out of range value for argument "def"

If i
* don't start rs-server on PC1
* Open realsense-viewer on PC2 and try to connect network device, I get: [getControls] error: Connection to server failed: Connection refused - -107

It is similar to the issue faced by this user. But for me the errors are seen on realsense-viewer not python with pyrealsense trying to access the streams

https://github.com/IntelRealSense/librealsense/issues/7459

MartyG-RealSense, 2021-11-06:
Can you confirm whether PC1 and PC2 are hosts or the remote board that the camera is attached to, please.  The white paper states that rs-server should be installed on Raspberry Pi, indicating that it is on the remote board and not the host computer that runs the RealSense Viewer.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-5-power-and-performance

parthapn98, 2021-11-06:
Hi Marty,

PC1 is a linux computer to which a d435 is connected. PC1 runs the rs server
PC2 is another linux computer on the same network and runs realsense-viewer

MartyG-RealSense, 2021-11-06:
Thanks very much for the clarification.

The white paper describes that rs-server on the remote computer connects to the first available camera [attached to the remote computer] and listens for incoming connection requests [presumably Network Camera connection requests made by the Viewer on the host computer].  

So when rs-server is not run and a Network Camera request is made by the Viewer on the host, it is logical that the **[getControls] error: Connection to server failed: Connection refused - -107** error would occur, because rs-server is not listening for incoming connection requests whilst it is inactive and so a bridge between remote and host cannot be established.  There is no interface active on the remote computer for the Viewer to talk to.

You also tried first running rs-server on the remote computer and then connecting to it from the Viewer on the host but received the error **out of range value for argument "def"**.

You mentioned earlier in the case that you did not configure at static IP because you are not using a Pi.  Whilst the requirement for a static IP does not seem to be exclusively for the Pi, the paper does suggest that the system has the flexibility to use other types of connection, such as wireless wi-fi / WLAN.  Can you confirm please whether the problem is that you can connect to the remote camera successfully with the original host computer (PC2) but you cannot connect with a second host computer that is also equipped with the Viewer?

parthapn98, 2021-11-12:
Hi Marty,

The rs-server was initially run with the following setup:
Wifi hostpot: 
* PC1 rs-server, (if realsense viewer is run locally streams can be accessed)
* PC2 realsense-viewer-network device: error out of range....
The ip addresses of each device were found by checking the live DHCP assignment from the hotspot base

Next Setup: Ethernet router+static ip
* PC1 rs-server, (if realsense viewer is run locally streams can be accessed)
* PC2 realsense-viewer-network device: error out of range....
The ip addresses of the corresponding pcs were verified by ping results

[image]

parthapn98, 2021-11-12:
Hi Marty,

Is there a way to do alignment of the depth frame to colour after receiving it via ROS?
EG, we run rs_camera.launch, receive it on a remote ros client. Now we want to align the rgb and depth stream.

I know we could do the alignment directly with rs_aligned_depht.launch but since the aligned depth image is resized to rgb stream, it will block up a significant bandwidth in our setup.

MartyG-RealSense, 2021-11-12:
Perhaps you could create a ROS adaptation of Intel's open-source ethernet system described in the link below, which streams RealSense data from a camera on a 'remote' board such as Raspberry Pi to a program such as the RealSense Viewer running on a central 'host' computer.  

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

This adaptation possibility is suggested by the paper itself.  When it mentions ROS, it has the number [10] beside it that refers to a RealSense ROS guide listed in the references at the bottom of the paper for using 3 cameras across 2 computers:

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

A RealSense ROS user in https://github.com/IntelRealSense/realsense-ros/issues/1808 has already performed a ROS adaptation.

parthapn98, 2021-11-13:
Hi Marty, we already have a working remote ros system with the realsense where we can access the stream from a client on the network.

My question is regarding is it possible to somehow use the realsense SDK to perform alignment of depth to RGB on the client side? Because the frames are no longer in the 'frameset' format to the receiver and instead published as images.

We do not publish the aligned depth stream from the hosts as these are resized to RGB resolution which is typically higher, and takes up more bandwidth.

The ros questions are separate in nature to the rs-server. We are exploring this since the rs-server is not working yet.

MartyG-RealSense, 2021-11-13:
A RealSense Python user in https://github.com/IntelRealSense/librealsense/issues/5296 took the approach of using **pickling** for remote alignment by transmitting the unaligned data.

parthapn98, 2021-11-19:
Hi Marty,

Is there a way to set emitter always on/ other emitter controls from the ros launch files?
(I wasnt able to find any parameter for these from the launch and node yaml files)

Whats the procedure for doing intercam sync between multiple cameras on ros? (Hardware sync cables involved)
(I only found one parameter 'enable_sync'(boolean) in the launch files
How should master/slave be designated?

MartyG-RealSense, 2021-11-19:
https://github.com/IntelRealSense/realsense-ros/issues/1472#issuecomment-719415753 suggests setting the **emitter_always_on** ROS parameter.

I do not have a ROS1 rosparam launch file code for emitter_always_on.  I believe that the code below may work though:

```
<rosparam>
/camera/stereo_module/emitter_always_on: true
</rosparam>
```

Alternatively, you could add the instruction **emitter_always_on:=true** to the roslaunch instruction.  For example:

**roslaunch realsense2_camera rs_camera.launch emitter_always_on:=true**

Regarding multiple camera sync, on the 400 Series cameras **Inter Cam Sync Mode** should be set to '1' for the Master camera and '2' for the Slave cameras.

**enable_sync** is not related to multiple camera hardware sync.  It instead gathers the closest frames of different sensors - infra red, color and depth - to be sent with the same timetag. 

You can define Inter Cam Sync Mode in ROS with the parameter **inter_cam_sync_mode** and information about setting it in ROS1 can be found in https://github.com/IntelRealSense/realsense-ros/issues/984#issuecomment-707729244

You do not necessarily need to use hardware sync in ROS for a multiple camera setup though.  Intel have published ROS guidance for 2 cameras on one PC and 3 cameras across 2 PCs.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

parthapn98, 2021-11-24:
Hi Marty,

Thanks for your help with the ros settings above.

We have come across an issue that crosses between the ros and sdk usage in python. Ill try to to it explain it concisely first

- we do roslaunch of any launch file(default or customised) that runs the realsense cameras and publishes topics
- we Ctrl-c the ros process
- the following code lines is executed in python next
```python
ctx = rs.context()
d = ctx.devices
```
> - the returned lists are then empty
> - the device can no longer be seen in realsense-viewer, (both cases of viewer being started before and after executing roslaunch)
> -  lsusb still displays the cameras
> - the devices are rediscovered if usb is unplugged and reinserted.

- however roslaunch of the launch files can still activate the cameras when they are not listed in rs.context()
> - we have not been able to replicate the usb unplug by software
> - usbreset of the device from cmd line returns 'Resetting  ... ok' but doesnt work
> - also tested python based usb reset from usb.core, no success.

I know other users with realsense have explored software based usb reset in the issues, have you come across anything that succeeded reliably?

MartyG-RealSense, 2021-11-24:
You could try a multiple camera version of the **hardware_reset()** instruction to reset only the cameras and not the entire USB port.  An example of such a script in Python is in https://github.com/IntelRealSense/librealsense/issues/5428#issuecomment-564482167

Note in that script, a list of all attached cameras is automatically built with **ctx.query_devices()**

parthapn98, 2021-11-24:
Hi marty, 

The ctx object itself is empty here

MartyG-RealSense, 2021-11-24:
Is the camera detectable if you first shut down the ROS launch with **rosnode kill --all** instead of using Ctrl-C?

parthapn98, 2021-11-24:
No its not detectable,

> top right is the terminal running roslaunch
> bottom right roscore

[image]

MartyG-RealSense, 2021-11-24:
As an alternative to closing down the ROS wrapper with Ctrl-C, would it be possible to run your Python code from within the active RealSense ROS wrapper as a Python node script, as discussed in https://github.com/IntelRealSense/librealsense/issues/6367

parthapn98, 2021-12-03:
Hi Marty,

Is there an option to enable the coloriser filter in the launch files for ros?

MartyG-RealSense, 2021-12-03:
Hi @parthapn98   You can enable the colorizer as a filter in ROS, similar to defining a pointcloud filter.

For example, if you only needed a colozier filter then you could define it in the roslaunch instruction like this:

**roslaunch realsense2_camera rs_camera.launch filters:=colorizer**

If you needed both a pointcloud and colorizer then you could define it like this, putting a comma with no spaces between the filter types:

**roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud**

If you prefer to set it in the launch file, I have not got a precise example for the colorizer.  However, I do have one for post-processing filters:

`<arg name="filters" default="disparity,spatial,temporal,decimation,pointcloud"/>`

So I would speculate (not certain) that as the colorizer is a 'filter' type, if it is possible to set it in the launch file then a launch file arg for it might look like this:

`<arg name="filters" default="colorizer"/>`

parthapn98, 2021-12-06:
Hi Marty,

We are pretty happy with the coloriser and its predefined compression for the compressed topic

1) Is there a way to set the filter options of the coloriser like
````c++
thr_filter.set_option(RS2_OPTION_MIN_DEPTH, min_depth);
thr_filter.set_option(RS2_OPTION_MAX_DEPTH, max_depth);
color_filter.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 0);
color_filter.set_option(RS2_OPTION_COLOR_SCHEME, 9.0f);		// Hue colorization
color_filter.set_option(RS2_OPTION_MAX_DEPTH, max_depth);
color_filter.set_option(RS2_OPTION_MIN_DEPTH, min_depth);
````

2) is the threshold filter available in the roslaunch CLI? filters:=threshold_filter didnot work
3) Is there a way to set the compression scheme and percentage for the /camera/depth/image_rect_raw/compressed topic when using the coloriser

MartyG-RealSense, 2021-12-06:
1.  I could not find specific information about setting colorizer filter options in the ROS wrapper, though I suspect that it would be approached in the same way as querying other filter types such as post-processing filters.  **Doronhi** the RealSense ROS wrapper developer provides guidance about this in https://github.com/IntelRealSense/realsense-ros/issues/742#issuecomment-720295533 

Using this guidance, you could potentially query what parameters are available for the colorizer filter in the ROS wrapper by using a **rosrun dynamic_reconfigure dynparam get** command.

https://github.com/IntelRealSense/realsense-ros/issues/2140#issuecomment-952850147 has a reference to setting histogram equalization using **/camera/colorizer/**

2.  The ROS wrapper does not support the threshold filter, though a maximum depth distance can be set with the **clip_distance** instruction.  If you require a minimum distance like the threshold filter provides, you could set a minimum distance in **rtabmap_ros** by using its **RangeMin** option, as described in the link below.

https://github.com/introlab/rtabmap_ros/issues/413

3.  I researched this question carefully but could not find information about it, unfortunately.

parthapn98, 2021-12-06:
The rqt recofigure had all the options for the coloriser including the jpeg/png compression quality settings
Thank you

We had some issues when we shifted the test to our embedded device(jetson nano) its cpus are not really powerfull enough to even apply the coloriser filter at more than 15fps, anything with filters use a cpu and the nano loses fps significantly.

Is there any implementation from a user who leveraged the gpu on a nano? or used a different CPU embedded device that could provide sufficient compute power?

MartyG-RealSense, 2021-12-06:
I'm pleased to hear that you were able to find the colorizer settings that you needed.  

If librealsense has CUDA support enabled to make use of an Nvidia GPU (like the one that Jetson boards have) then it can provide acceleration for **color conversion, depth-color alignment and pointclouds**.  If the type of processing that you are performing does not involve those specific types of operation then you may not benefit from CUDA support being enabled.  

Also, if the ROS wrapper was installed from packages with the wrapper's **Method 1** (librealsense and the wrapper installed together) then CUDA support would not be included in the librealsense build.  

Intel strongly recommends that RealSense users with a Nano enable the **barrel jack** connector for extra power using the information in the link below if your Nano model supports that.

https://www.jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

In regard to alternative embedded devices, you could consider a more powerful Jetson board model so that you can still take advantage of an Nvidia GPU, though this would have increased cost.  

You could also explore Intel's **NUC** computing range.  They are small enough to fit on a mobile robot, suited to industrial applications and are available in a wide range of configurations (from low end to high end) and price points, and in different form factors (board, partially completed kit and complete ready-to-run mini-PC).  NUCs also have a long history of being used with RealSense cameras dating back to the original generation of RealSense in 2014.

https://www.intel.com/content/www/us/en/products/details/nuc.html

There is also the hardware acceleration solution for vision computing called Intel Neural Compute Stick 2 (NCS2), which can be simply plugged into a USB port alongside the camera.  It is available from the official RealSense store.  I recommend googling for **realsense ros ncs2** to see whether an Intel NCS2 stick may be able to be leveraged in your ROS project.

https://store.intelrealsense.com/buy-intel-neural-compute-stick-2.html

parthapn98, 2021-12-08:
Hi Marty,

We reflashed our nano os and rebuilt everything with cuda support
-libuvc script with dbuild with cuda added
-ros using catkin build

We had some weird results that ill try to detail:

Prior Setup:
1) Subscribing to rgb compressed: Maxed out 1 cpu core 26FPS
2) Publishing depthimage with coloriser filter: maxed out 1 Cpu core: On Subscribe: 14 FPS

Current Setup:
1) Subscribing to rgb compressed: Doesnot Max out any cpu cores: 26FPS
2) Publishing depthimage with coloriser filter: maxes out 1 Cpu core: On Subscribe: 13-15 FPS

My expectations from going over the issues on similar topic in realsense ros, were compression would still be done on CPU and filters like coloriser, point cloud, etc would be on GPU.

Any suggestions on where i can look into to get to the bottom of this?

parthapn98, 2022-03-06:
Hi @MartyG-RealSense 

Is there some documentation for the working of depth-color alignment function rs_align?

I am looking into this since our current format of recording utilises realsense_ros, and we want to understand how far the performance hit would go if we used rs_aligned_depth.launch. (Also rs_aligned would resize the depth image to 1280x720 and BW is a small concern)

[I have taken a look at rs_align_example and rs_align_advanced example notebooks in librealsense, and [rs_align in](https://dev.intelrealsense.com/docs/rs-align) however they only show how to impement the functions and their outputs, similar to a python demo on the depth_color alignment that removes background beyond a certain distance.]

Is there any documentation on the internal workings of rs_align or is it possible to take a look at the code itself?

- Does it only crop and reproject the depth image?
- Or performs any other implementations?

MartyG-RealSense, 2022-03-06:
Hi @parthapn98  The full C++ source code for **rs-align** can be found at the link below, though the descriptive tutorial text is the same as the page that you linked to.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

As described in the opening paragraph of the tutorial, algnment will resize the depth field of view (FOV) to match the size of the color field of view.  As the color sensor FOV on D435 / D435i is smaller than the depth FOV, this can mean that the outer regions of the depth image will be excluded from the aligned image.  

Since your programming language is listed as Python at the top of this case, the SDK's Python example program **align_depth2color.py** may be a more useful reference.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

My understanding is that the **align_depth** function in the RealSense ROS wrapper makes use of the RealSense SDK's **align processing block** via the 'align_to' instruction like examples such as rs-align and align_depth2color.py do.  The align processing block will make automatic adjustments for differences in the depth and color resolution.

Depth to color alignment is a processing-intensive operation.  If you are able to use a computer with Nvidia graphics GPU hardware, such as an Nvidia Jetson board or a PC with Nvidia graphics, then you could make use of the RealSense SDK's ability to be built with **CUDA** support for acceleration of color conversion, alignment and pointclouds.  

The ROS wrapper is able to take advantage of CUDA acceleration if librealsense and the ROS wrapper are built separately instead of built together from packages with the wrapper's **Method 1** installation method (as the Method 1 packages do not provide CUDA support).

parthapn98, 2022-03-06:
###### Yes I have seen the files you refer to. These are example files that use the align function of the realsense sdk.
###### What I am looking for is the code inside the functions in the following lines.
###### Or a documentation describing how the alignment is performed.
[image]
------------
[image]
------------
Since i only have the python examples in my ide, navigating to the source of the function
[image]

leads to align.py containing
```python
import pybind11_builtins as __pybind11_builtins

from .filter import filter

class align(filter):
    """ Performs alignment between depth image and another image. """
    def process(self, frames): # real signature unknown; restored from __doc__
        """
        process(self: pyrealsense2.pyrealsense2.align, frames: pyrealsense2.pyrealsense2.composite_frame) -> pyrealsense2.pyrealsense2.composite_frame
        
        Run thealignment process on the given frames to get an aligned set of frames
        """
        pass

    def __init__(self, align_to): # real signature unknown; restored from __doc__
        """
        __init__(self: pyrealsense2.pyrealsense2.align, align_to: pyrealsense2.pyrealsense2.stream) -> None
        
        To perform alignment of a depth image to the other, set the align_to parameter with the other stream type.
        To perform alignment of a non depth image to a depth image, set the align_to parameter to RS2_STREAM_DEPTH.
        Camera calibration and frame's stream type are determined on the fly, according to the first valid frameset passed to process().
        """
        pass

```
###### This is likely because the realsensesdk wasnot built in python.
------------

> As described in the opening paragraph of the tutorial, algnment will resize the depth field of view (FOV) to match the size of the color field of view.

Does Align _resize_ only? Or does it perform reprojection from left stereo plane to rgb plane?

MartyG-RealSense, 2022-03-06:
My understanding is that the processing blocks are handled by the SDK file **rs-processing.hpp**:

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_processing.hpp

And the SDK file **align.cpp** includes rs_processing.hpp:

https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp

Aside from the rs-align tutorial information, the other main sources of offical C++documentation for the align process are here:

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1align.html

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#section-frame-alignment

The pyrealsense2 version of the align process documentation is here:

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.align.html

The link below may may a helpful reference in regard to alignment and **reprojection**.

https://github.com/IntelRealSense/librealsense/issues/5743

parthapn98, 2022-03-06:
Thank you Marty, the links have mostly answered my questions.

Ill go through the cpp file to understand the exact implementations in the future

Just glancing over the code i have some initial questions about how the extrinsics from left IR frame to rgb frame is calculated.
If you find anything on this please let me know. 

Thanks for the help again

MartyG-RealSense, 2022-03-06:
The transform between two stream types can be retrieved with the SDK instruction **get_extrinsics_to**

**C++**
https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1stream__profile.html#a0085b1f54ac6078c638ceb53ff89cc95

https://dev.intelrealsense.com/docs/api-how-to#section-get-and-apply-depth-to-color-extrinsics

https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/examples/sensor-control/api_how_to.h#L277

**Python**
https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.stream_profile.html#pyrealsense2.stream_profile.get_extrinsics_to

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888

parthapn98, 2022-03-08:
Hi Marty is there a way to ensure the published frames when using realsense_camera launch files, stay as close in sync as possible between all topics?

The difference is seen in the headers of the corresponding depth and image frames.

Issue is not present if we record with rs viewer or pyrealsense
[image]