realsenseai/librealsense #9864: [Error]:"No Streams are selected" Occurs in Remote connection of Python [L515]

Dongbox, 2021-10-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18.04) / Windows(10/11)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |  2.47  | 
| Language                            |  python   | 

### Issue Description
I have two devices here: 1.Windows Laptop and 2.NVIDIA Jetson TX2. And I want to set Jetson as an `rs-server`, connecting to my L515 camera, and in Windows Laptop I will use the python code from "https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py" to start a remote session.
After I deploy all services on two computers, which is the 2.47 version, the problems come in my python running.
[image]

First, I had checked it runs in Jetson TX2 normally, included running official intel viewer locally and remotely and python connection. It's all right.

- Running on Viewer locally in TX2
[image]
- Running Remote Session on Viewer locally in TX2
[image]
- Running Python Connection locally in TX2
[image]
Next, I started the `rs-server` in the Jetson TX2 and open the intel viewer in the windows Laptop to connect it and its still normal.
- Running Python Connection Remotely in Windows
[image]

The only problem occurs when I run python code in the terminal, I don't know why it says no streams here and I try to restart the `rs-server` or close the intel-viewer so as to keep only one device to receive the streams, but it all doest works.

Can you provide some tips here for me?
I had been switched the SDK version from 2.49 to 2.47 depending on one issue, but it seems not a valid method here.

Dongbox, 2021-10-18:
@MartyG-RealSense Thanks for your reply! 
But I always keep only one client to receive the streams from the server.  Like close the Viewer and restart the `rs-server` or even restart all computers such methods to confirm only one client to one server.

So it may not an appropriate solution to this issue here.

Dongbox, 2021-10-18:
> Does **net_viewer.py** print a 'Connected' confirmation message that a network connection to the camera has been successfully made?
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py#L24
> 
> I am not familiar with how the net_viewer.py program functions, but I cannot see in the code how it makes a connection to the camera like ssh'ing in the white paper does. I wonder if the program assumes that there is a network connection to the remote camera active already when the script is run.
[image]

It printed connected and camera information here which means it has successfully connect to the server. But error occurs when running `pipeline.start()`

```
dev = rsnet.net_device(ip)
print('Connected')
print('Using device 0,', dev.get_info(rs.camera_info.name), ' Serial number: ',
      dev.get_info(rs.camera_info.serial_number))
dev.add_to(ctx)
pipeline = rs.pipeline(ctx)
print('Start streaming, press ESC to quit...')
pipeline.start()
```

MartyG-RealSense, 2021-10-18:
I have studied the problem very carefully but cannot see an obvious reason for this problem.  As net-viewer.py was first introduced in librealsense **2.49.0**, I cannot rule out the possibility that the problem is related to something the program requires that is not included in the earlier 2.47.0 SDK that you are using.

MartyG-RealSense, 2021-10-19:
On September 17 2021 in https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-921425461 a RealSense team member advised that there were no plans to support L515 in rs-server.  This may be because the L515 model has been declared **end of life** by Intel, with the 400 Series stereo depth camera range being the main focus of the RealSense product range from this point onward.

closed 2021-10-19 by Dongbox