realsenseai/librealsense #9865: ModuleNotFoundError: No module named 'pyrealsense2'

didi1150, 2021-10-17:
| Camera Model                       | { D435 } | 
| Platform                                 | Raspberry Pi|
| SDK Version                            |  {2.0}                          
| Language                            |  {python }         

I am trying to make tensorflow lite work with the intel realsense cam d435 with raspberry pi. But I get the message             

e.py --modeldir=Sample_TFLite_Model/
Traceback (most recent call last):
  File "TFLite_detection_intel_realsense.py", line 15, in <module>
    import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'

I tried rebuilding the viewer, and setting the python path in the bashrc file, but it didn't work

didi1150, 2021-10-18:
How do I generate the cmakelists.txt in the build dir? 
I do mkdir build in the root directory, but I dont know how to add cmakelists,txt to the folder.

My output when I try to run the cmake command in the build folder:

pi@raspberrypi:~/build $ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
CMake Error: The source directory "/home/pi" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

MartyG-RealSense, 2021-10-19:
Could you go to the **librealsense > build > wrappers > python** folder on your Raspberry Pi and see whether there is a file there that starts with the name 'pyrealsense2' and ends in '.so' please like the file highlighted in the image below.  The moddle part of the filename will vary depending upon the Python version installed on your particular Pi.

[image]

didi1150, 2021-10-19:
@MartyG-RealSense could I add you somewhere else, eg discord because I dont want to wait a whole day for a response

MartyG-RealSense, 2021-10-19:
I apologize for the delay in responding further.  Support at locations outside of official RealSense forums cannot be provided.

The pyrealsense2.so script should not be renamed.  If you are using Python 3, you should use the one generated by the build process that has a complex long name, like the one highlighted in the image above.  This file should be copy and pasted unchanged next to your script, along with the librealsense2.so file.