realsenseai/librealsense #9903: formatting on this page mixes up instructions with code blocks

JHPHELAN, 2021-10-31:
The formatting on this page mixes up instructions with code blocks making it very confusing
JHP

MartyG-RealSense, 2021-10-31:
Hi @JHPHELAN   Could you provide a link to the page that you are referring to, please?  Thanks!

JHPHELAN, 2021-10-31:
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](url)

MartyG-RealSense, 2021-10-31:
The **Documentation** tag that I have placed on this case means that the issue with this documentation page has now been highlighted for review by the RealSense team at a future date.  Thaks very much for your report!

JHPHELAN, 2021-10-31:
Also on this page [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](url)
the line (which should be in 'code')
`sudo apt-get install librealsense2-utils sudo apt-get install librealsense2-dev
`
should be
`sudo apt-get install librealsense2-utils && sudo apt-get install librealsense2-dev
`
else it thinks the second command's words are packages which it can't find
JHP

MartyG-RealSense, 2022-03-19:
The broken formatting issues that you reported with the Jetson installation GitHub page have been corrected and **sudo apt-get install librealsense2-utils sudo apt-get install librealsense2-dev** is clearly formatted into two separate lines to make clear that they should be input individually.  So I will close this case.  Thanks again @JHPHELAN for your report!

closed 2022-03-19 by MartyG-RealSense