realsenseai/librealsense #9904: The Realsense camera D435i freezes in most of the SDK example scenes in Unity

Aylavanderwal, 2021-10-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  V2.49.0}                          | 
| Language                            |  unity }                          | 
| Segment			| AR }                          | 

### Issue Description
Hi all,

I am entirely new to this field so excuse me if I am asking silly things. However, the Realsense camera seems to freeze every time I play most of the SDK example scenes in Unity. TexturesDepthAndInfrared.unity seems to work just fine, but TexturesDepthAndColor.unity, SLAM.unity and ARbackground.unity load the camera for like 1/5th of a second and then the stream stops. This doesn't feel like it should be like this but I don't know how to fix it, anyone a clue what could be going on? In the Viewer.exe program, the camera seems to work just fine. 

Ultimately, I am looking for a way to open the pose estimator in Unity but I'll first need to get around this issue. 

Thanks so much if you know how to help out!

MartyG-RealSense, 2021-10-31:
Hi @Aylavanderwal   The RealSense Unity wrapper's SLAM example is for a RealSense camera model called the **T265 Tracking Camera** and do not work with the 400 Series camera models (which do not have the pose data feature built into them like the T265 does).

If you are using Unity 2020 or 2021, the Unity wrapper documentation page in the link below states that ARBackground is not available for use in Unity 2020.

https://dev.intelrealsense.com/docs/unity-wrapper#section-open-a-unity-scene

this documentation page also states that TexturesDepthAndColor requires a camera model with an RGB sensor.  Not all RealSense models have one, and most of the Depth Module Kit caseless circuit board versions of the cameras lack one.  Which RealSense camera model are you using, please?

Aylavanderwal, 2021-10-31:
Hi Marty Thanks for the quick response and thanks for clarrifying why the SLAM and ARBackground don't work! Sadly these are out of reach for my project then haha. Do you know other kits/libraries that can be used for body tracking purposes with the realsense?

In the side panel it says D435i, so it should have an RGB sensor. In the Viewer I can also toggle the RGB camera on and it works just fine, but when I open the TexturesDepthAndColor.unity it loads for a second and then I guess the stream freezes. But it doesn't give any Error output.

MartyG-RealSense, 2021-10-31:
There was a previous case in https://github.com/IntelRealSense/librealsense/issues/7743 of problems with the ARBackground and TexturesDepthAndColor example programs, so it may not be a problem with your particular computer, especially as the other Unity examples and the Viewer work fine.

If you would prefer a ready-made body tracking solution with Unity integration that is compatible with D435i then there are commercial software packages with free trial versions.

**Cubemos Skeleton Tracking SDK**
https://www.intelrealsense.com/skeleton-tracking/

**Nuitrack SDK**
https://nuitrack.com

An option for an open source body tracking solution with a Unity plugin is **OpenPose**.

https://github.com/CMU-Perceptual-Computing-Lab/openpose
https://www.intelrealsense.com/skeletal-tracking/

If you prefer to develop your own solution then Intel have published a YouTube seminar on training a body tracking system with RealSense.

https://youtube.com/watch?v=VSHDyUXSNqY

Aylavanderwal, 2021-11-01:
Hi MartyG-RealSense, 

case #7743 seems different as they have problems with a slow program, my streams don't start at all for these programs. 

But thanks a lot for the links to body tracking software, these will be very helpful!! 

Have a great day :)

MartyG-RealSense, 2021-11-13:
Case closed due to no further comments received.

closed 2021-11-13 by MartyG-RealSense

Aylavanderwal, 2021-11-13:
Hi MartyG-RealSense,

I discovered that the problem had to do with the incoming streams and their size. They were set on 30 fps but with an additional RGD stream, this caused the whole camera to freeze. So putting them on 15 fps fixed it for me!

MartyG-RealSense, 2021-11-14:
It's great to hear that you achieved a solution - thanks very much for the update!