realsenseai/librealsense #9905: Native backend librealsense FAILS on Jetson Nano

JHPHELAN, 2021-10-31:
With the ultimate goal of D455 / ROS2 / Ubuntu 20.04 / Raspberry Pi 4b I thought I'd see if the Jetson Nano installs had any better luck than I've had on the Pi.  The RSUSB backend approach hadn't work well so thought I'd try the native backend following:
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](url)
despite markdown formatting and command-line errors.  Says it's not been confirmed on the Nano but don't see why it shouldn't and why it hasn't been.  See attached log.  The indicated install script is nowhere to be found.  Have I missed something?  The supported L4T (32.4) version is slightly behind my current setup v32.6.
[2021.10.31 Native Backend FAIL.pdf](https://github.com/IntelRealSense/librealsense/files/7449236/2021.10.31.Native.Backend.FAIL.pdf)
JHP
@MartyG-RealSense

MartyG-RealSense, 2021-10-31:
Hi @JHPHELAN   Reading your log, it looks as though you installed librealsense successfully from packages with **sudo apt-key adv --keyserver** and then installed librealsense again from source code with the native backend (V4L) method.  You should choose one method or the other, as building from both packages and source code on the same computer can lead to error-causing conflicts in the installed files.

The separation between the package and source-code sections of the Jetson install guide is easier to see on the other version of the guide at the link below.

https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation#section-building-from-source-using-native-backend

The script command **./scripts/patch-realsense-ubuntu-L4T.sh** should be run from the root directory of the librealsense source code.  The root is the 'librealsense' folder with a list of files and folders in it.

[image]

JHPHELAN, 2021-10-31:
@MartyG-RealSense 
I believe it's referring to the directory on my Pi, not the directory in Github.  I tried just cc the one script and running it. Then it looks for another script that's not installed and crashes.  It seems I have to install the SDK to get the scripts onto the Pi but I can't run the script if I've already installed the SDK.  I'm going in circles!

MartyG-RealSense, 2021-10-31:
I note your mention of Pi in the comment above, but the Pi is not involved in this Jetson installation process.  Did you mean Nano, please?

It looks as though the instructions in the **Build and Patch Kernel Modules for Jetson L4T** section are assuming that you have obtained the librealsense source code already, either from the librealsense GitHub site or by downloading the source code as a zipped file from the 'Assets' file list on the SDK 'Releases' page.  

Once the librealsense source code is obtained, it asks you to navigate to the librealsense root folder with the list of folders and files shown in the image above.  So on a typical source code folder, the root may be at **librealsense > librealsense**, with the topmost folder being empty.  Then once in the root folder, input the command **/scripts/patch-realsense-ubuntu-L4T.sh** to patch the kernel.

At this point it is not asking you to build the librealsense SDK, just to patch the kernel.  The next section of the instructions after that - **Build librealsense2 SDK** - asks you to go back to the librealsense source code root directory and then build librealsense.

****************

If the above process does not work for you, there are a few other options that you could investigate.

-. Use the **libuvc backend** installation process.  This was the predecessor of the RSUSB backend method and has been known to work well on Raspberry Pi, though I do not recall an example of it being tried with Ubuntu 20.04

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

2.  Install on Nano from source code with the RSUSB method for Jetson at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

3.  If you want a native backend build, use the above procedure but disable RSUSB with **DFORCE_RSUSB_BACKEND=OFF** instead of setting it to ON so that it builds in V4L native mode.

JHPHELAN, 2021-10-31:
@MartyG-RealSense 
Yes, I meant Nano.
J

MartyG-RealSense, 2021-11-07:
Hi @JHPHELAN  Do you require further assistance with this case, please?  Thanks!

JHPHELAN, 2021-11-08:
[NANO RealSense SUCCESS.pdf](https://github.com/IntelRealSense/librealsense/files/7493579/NANO.RealSense.SUCCESS.pdf)
@MartyG-RealSense 
I believe I've (at least mostly) successfully installed the D455 depth camera with ROS2 under Ubuntu 18 on the Jetson NANO using a modified libuvc back-end install and the ROS2 wrapper from Source.  See attached log for details.  
Thanks for your support!
OK to close.
JHP

MartyG-RealSense, 2021-11-08:
You are very welcome, @JHPHELAN - it's great to hear that you were successful  :)  As you suggest, I will close the case as you are satisfied.  Thanks again for sharing your very detailed test reports!

closed 2021-11-08 by MartyG-RealSense