realsenseai/librealsense #9906: error

cute-me, 2021-11-01:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-11-01:
Hi @cute-me   Please first try installing the earlier **2.48.0** version of the RealSense SDK using the **Intel.RealSense.SDK-WIN10-2.48.0.3381.exe** automated Windows installer file in the 'Assets' file list on the page linked to below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

Does the error still occur on the 2.48.0 SDK version, please?

MartyG-RealSense, 2021-11-08:
Hi @cute-me  Do you require further assistance with this case, please?  Thanks!

ekkiiiii, 2021-11-08:
Hello, I had the same issue (either while trying to install new firmware to the D415 from the prompt inside the Viewer or while trying to add a source inside the viewer) and downgrading to 2.48.0. fixes it for me.

MartyG-RealSense, 2021-11-08:
Thanks very much @ekkiiiii for your advice to @cute-me about what worked for you in your particular case  :)

closed 2021-11-14 by MartyG-RealSense