realsenseai/librealsense #9908: can't connect d415 with realsense viewer in windows10

piaorou-tech, 2021-11-01:
Hello, I found two errors when opeing realsense viewer like the picture below. And it can't connect with d415, but I just open it with camera application.
[image]
The first on is :
18:19:50 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - _source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "??????"
And the second one is:
18:19:50 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:537 - acquire_power failed: _source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "??????"
Could someone help about it ? Thanks!

MartyG-RealSense, 2021-11-01:
Hi @piaorou-tech   The **IKSTopologyInfo** is a very rare error that was reported a few times in the early days of the RealSense 400 Series cameras (late 2017 - early 2018) but has not been seen since,  

Has the firmware driver of your D415 camera been updated since the camera was purchased.  If not, could you update the firmware to the latest version please to see whether it makes a difference to this error.

piaorou-tech, 2021-11-01:
> Hi @piaorou-tech **IKSTopologyInfo** is a very rare error that was reported a few times in the early days of the RealSense 400 Series cameras (late 2017 - early 2018) but has not been seen since as far as I am aware.
> 
> Has the firmware driver of your D415 camera been updated since the camera was purchased. If not, could you update the firmware to the latest version please to see whether it makes a difference to this error.

Hi,I have upgraded it on the other computer,and this is my new pc, I just move d415 on it.By the way, I install ubuntu on the same pc, and that version of realsense viewer can successfully connect d415.

MartyG-RealSense, 2021-11-01:
Thank you very much.  Next, please check whether the Windows privacy options for the camera have access to the camera hardware and access by camera applications both set to On.  You can do this using the instructions in https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

piaorou-tech, 2021-11-01:
> Thank you very much. Next, please check whether the Windows privacy options for the camera have access to the camera hardware and access by camera applications both set to On. You can do this using the instructions in [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)

yeah,I have checked it before

MartyG-RealSense, 2021-11-01:
Have you checked whether the **rs-capture** example program works in Windows using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 in order to determine whether it is a problem with the Viewer only?

piaorou-tech, 2021-11-01:
> Have you checked whether the **rs-capture** example program works in Windows using the instructions in [#8024 (comment)](https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656) in order to determine whether it is a problem with the Viewer only?

Hello,I just tried it. It was full of white like below. How did it happen?
[image]

MartyG-RealSense, 2021-11-01:
When you launch programs such as the RealSense Viewer and rs-capture, do you have any other programs already active?  If an already running program is accessing a particular camera stream then that stream will be inaccessible to programs that are subsequently launched.

The white screen of rs-capture could also be caused if a camera is not detected when the program launches.

piaorou-tech, 2021-11-01:
> When you launch programs such as the RealSense Viewer and rs-capture, do you have any other programs already active? If an already running program is accessing a particular camera stream then that [...]

Yes,I didn't activate any other programs.

MartyG-RealSense, 2021-11-01:
Next, we can check whether it may be a graphics-related problem by running the example program **rs-hello-realsense.exe** which is in the same folder as rs-capture.exe.  rs-hello-realsense generates a simple text readout of the depth at the center of the camera's view.  If this non-graphical example works, it would suggest that the graphics drivers on your new PC might need updating.

piaorou-tech, 2021-11-01:
Thanks for your help, I just reinstall the driver of two modules of camera about realsense. And now I can access rs-capture.exe. But the realsense viewer can only ouput depth stream. And it has something wrong with RGB stream like below.
[image]
It sas the device does not support adjusting the qualities of an incoming video signal,
such as brightness, contrast, hue, saturation, gamma, and sharpness.

piaorou-tech, 2021-11-01:
> Next, we can check whether it may be a graphics-related problem by running the example program **rs-hello-realsense.exe** which is in the same folder as rs-capture.exe. rs-hello-realsense generates [...]

It's Okay now, I found I didn't reinstall rgb modules. And now I suceed. Thank you very much!

MartyG-RealSense, 2021-11-01:
Great news, @piaorou-tech - thanks very much for the update  :)

closed 2021-11-07 by MartyG-RealSense