realsenseai/librealsense #9909: Low Image Quality from D435

codeandstuf, 2021-11-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I have created a script to take pictures or videos of cows when they walk through an area. 

#THIS IS MY SETUP
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 10)
pipeline_record_profile = pipeline.start()

#WHEN A COW IS DETECTED I CALL THIS
frames = pipeline.wait_for_frames()
saver = rs.save_single_frameset("FarmVisit3_" + datetime.now().strftime("_%m_%d_%Y_%H_%M_%S")+ "pic" + str(i))
saver.process(frame)

Many of the images I have taken have missing data chunks. Below is an example of how data chunks are missing. There are consistently black spots on the back left side of the cow. This leads me to believe the camera likely needs to be recalibrated, but I would like to know if my code is possibly the cause as well. I should note when I take videos of the cows instead of pictures there are very few videos that have missing data chunks.

Thanks in advance!

[image]

MartyG-RealSense, 2021-11-01:
Hi @codeandstuf  If your cows are a black and white breed such as Fresians then the camera will have difficulty capturing the black areas of the cow.  This is because it is a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces.  The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.

You could try filling in the black areas of the depth image by applying a post-processing filter with **hole-filling** properties in your Python script.  If the cow is in an indoor location such as a cow shed then you could also install a stronger external light source such as an overhead lamp, as doing so helps to bring out depth information from dark grey / black surfaces.

Having said that, I note that you mention that you mainly experience the black areas on pictures but not videos.  As you are saving a single frameset, this could suggest that you should skip several frames before taking a picture capture if you have auto-exposure enabled.  This is because it can take several frames for the auto-exposure to settle after the pipeline starts, which can result in problems in the depth image for those first handful of frames.  

Alternatively, if the lighting level in the camera's location is consistent all of the time that the camera is in use then you could disable auto-exposure and set a manual exposure value.  You will then not need to skip several frames and the first frame captured should be good.

MartyG-RealSense, 2021-11-08:
Hi @codeandstuf  Do you require further assistance with this case, please?  Thanks!

codeandstuf, 2021-11-09:
I do not. Thank you for your help!

closed 2021-11-09 by codeandstuf

MartyG-RealSense, 2021-11-09:
You are very welcone, @codeandstuf - thanks very much for the update!