realsenseai/librealsense #9913: Launch error

ejones03, 2021-11-02:
| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-11-02:
Hi @ejones03  Could you first try updating the graphics drivers on your Windows PC to see whether this resolves the error, please?

ejones03, 2021-11-03:
@MartyG-RealSense Yes, I updated the graphics drivers on my Windows PC. We are still receiving the same error message when trying to launch the program. Are there any other steps we can try?

MartyG-RealSense, 2021-11-03:
Thanks very much for updating your graphics drivers.  Next, can you test whether you can run a RealSense SDK example program called **rs-capture** using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

ejones03, 2021-11-04:
The RealSeanse Viewer will now open, but the application does not recognize the camera plugged in. We have installed the newest version of RealSense and updated the firmware on the camera (D435i). The camera works fine on another computer, so we think it may be a computer issue. After reading other comments, we checked the Device Manager, and the RGB and Depth is located under Universal Serial Bus devices instead of cameras. We uninstalled the RGB and depth, and also tried disabling the laptop's camera but neither of those worked. Any further suggestions?
[image]

MartyG-RealSense, 2021-11-04:
Next, please check that access to the camera hardware and access to camera applications are both set to On in the Windows privacy settings using the instructions in https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

ejones03, 2021-11-04:
Yes, we checked that and it is On in settings

MartyG-RealSense, 2021-11-04:
Thanks very much for checking.  When you uninstalled the RGB and Depth drivers in the Windows Device Manager, once both drivers were uninstalled did you unplug the USB connector, wait a couple of seconds and then plug the camera back in?  If it is working normally then both drivers should be reinstalled under the Cameras category.  Do they reappear under the Universal Serial Bus category on your particular computer?

It is worth noting that where the drivers are placed may depend on the particular Windows computer.  Whilst the two drivers are most commonly placed under Cameras, in rare cases they may appear under other categories such as Imaging Devices.  So it is not necessarily a sign of a driver problem.

ejones03, 2021-11-04:
Yes, when we uninstalled the drivers we waited a couple of seconds and plugged the camera back in. We tried it a couple times and they always reappeared under Universal Bus Drivers. They do not appear appear under Universal Bus Drivers on the other computer that is working, though, they are under Cameras.

MartyG-RealSense, 2021-11-04:
Are there any obvious differences between the two computers (different manufacturer, different Windows versions, etc).

ejones03, 2021-11-04:
They are the same manufacturer and make of laptop, however on the laptop that is not working it is a more recent Windows version. The one working is Windows 10 Pro Version 20H2, and the one not working is Windows 10 Pro Version 21H1

MartyG-RealSense, 2021-11-04:
Have you tried installing a newer SDK version than 2.47.0, such as 2.48.0 please?  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

[image]

ejones03, 2021-11-04:
Yes, we've tried version 2.47, 2.48, and 2.49. The applications will open, but still not recognize a camera.

MartyG-RealSense, 2021-11-04:
Has the camera ever worked correctly on this particular 20H2 laptop or is the problem one that has only manifested recently?

ejones03, 2021-11-04:
Yes, the camera was working on there for a long time. It just manifested a few days ago.

ejones03, 2021-11-04:
The update had been back from March, so it has been a while.

MartyG-RealSense, 2021-11-05:
Windows has a hidden **Hardware Troubleshoot** feature that can scan for problems in devices and try to automatically fix them if it does find an issue.  It can be activated by going to the Windows Command Prompt interface and inputting the command below.

**msdt.exe -id DeviceDiagnostic**

[image]

ejones03, 2021-11-08:
Thank you for that knowledge. I ran the Hardware Troubleshoot and it gave me the feedback "Restart your PC to finish installing drivers and updates". I restarted, but it did not resolve any issues regarding RealSense. I ran the Hardware Troubleshoot again and received the same feedback.

MartyG-RealSense, 2021-11-09:
Earlier in this discussion in https://github.com/IntelRealSense/librealsense/issues/9913#issuecomment-960158355 I requested that you try the **rs-capture** SDK example program.  Did you try it, and if so what was the result?

ejones03, 2021-11-09:
I tried it now - the window pops up for a second and then says it's not responding and it closes. It does not show a camera feed.

MartyG-RealSense, 2021-11-09:
It looks increasingly likely that it is an issue with this particular computer - either something in its hardware or in Windows - and not a problem with the camera hardware or its USB cable.

Next, could you detach the camera, open the Devce Manager and then go to the **View** menu option and select **Show hidden devices**.  This will show the drivers for RealSense camera models that have previously been installed but are not normally visible when the camera is not plugged in.

[image]

[image]

If there are any RealSense drivers listed in the Device Manager categories whilst the camera is unplugged, right-click on them and select **Uninstall device** like you did previously.  When all driver entries have been removed, insert the camera into the USB port.  Is the camera now detectable in the Viewer after doing so?

ejones03, 2021-11-09:
I tried this and when I opened the Intel RealSense Viewer, I am now back to getting this error "GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569"

MartyG-RealSense, 2021-11-10:
Does the standalone .exe version of the Viewer produce the same error?  This can be downloaded by clicking on the **Intel.RealSense.Viewer.exe** link in the Assets list.  The Viewer .exe should be launched from the folder location that it is downloaded to and not with the launch shortcut icon on the Windows desktop.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

ejones03, 2021-11-10:
Yes the standalone version gives the same error as well

MartyG-RealSense, 2021-11-10:
Do you receive the same **GL_INVALID_VALUE** error if you launch the **Depth Quality Tool** instead of the Viewer?

closed 2021-11-23 by MartyG-RealSense

MartyG-RealSense, 2022-02-07:
There was a case of the **GL_INVALID_VALUE** RealSense Viewer error at https://github.com/IntelRealSense/librealsense/issues/10187#issuecomment-1031762801 that was resolved by installing and running the old **2.21.0** version of the standalone Viewer and then going back to the modern Viewer and running it.