realsenseai/librealsense #9914: D435i RealSense Viewer and Python Depth Viewing Example not Consistent

LW-G38, 2021-11-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Linux (Raspbian GNU/Linux 10 buster)  | 
| Kernel Version (Linux Only)    |  5.10.60-v7l+                                       | 
| Platform                                 |Raspberry Pi 4|
| SDK Version                            |  2.49.0                         | 
| Language                            |  Python                     | 
| Segment			|  Robot                         | 

### Issue Description

I'm very green in Depth Sensing and just getting started to play with basic examples. I installed the whole package in Raspbian and use the viewer program and the OpenCV Python script to see the depth sensor stream separately. However, the stream from the OpenCV Python script could not recognize the objects that are close to the camera compared to the one generated by the viewer program (The distance is larger than 0.3m, the lower bound of the sensor.) I'm not sure which parameters I need to tune in .config. Can you please help me out? Thank you.

[image]

[image]

MartyG-RealSense, 2021-11-03:
Hi @LW-G38  If you are seeking to capture the maximum observable depth distance in the RealSense Viewer then you should ensure that a post-processing filter called **Threshold Filter** is disabled or set to a value of '10'.   By default this filter is set to '4' meters and restricts the maximum range of the rendered depth data to that distance if the filter is enabled and left on its default setting.

Scripts written in the SDK, such as in the pyrealsense2 wrapper for Python, do not contain post-processing by default, and such filters must be deliberately programmed into scripts.  So the output generated by a RealSense user's own applications will tend to look different to the output produced by the Viewer (which has a range of  post-processing filters enabled by default) before filters are programmed into an application.

[image]

On a colorized depth image, the color is typically shaded as blue for near-range surfaces and progressively shifts towards red (far distance) as distance increases.  So the Python image actually looks more accurate than the Viewer image where there is a range of different colours at near-range.

LW-G38, 2021-11-03:
Hi @MartyG-RealSense , thank you so much for your prompt response. I just have a couple of follow-up questions - I turned on the post-processing filters, and the colorized depth still looks different compared to the script - I assume there are other controls and visualization settings that affect the look, correct? I do agree that the Python script generated image shows more consistency at near-range, but as I do place my arm closer than my body to the camera, so I expect it shower cooler color than my body for instance.

[image]

MartyG-RealSense, 2021-11-03:
In addition to the post-processing defaults, the Viewer's **Depth Visualization** side-panel options apply a range of color-scheme and color **histogram equalization** settings by default too.  

[image]

These settings can be reproduced in a Python application using information in https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

LW-G38, 2021-11-03:
Thank you @MartyG-RealSense , I will start from that.

closed 2021-11-03 by LW-G38