realsenseai/librealsense #9915: Realsense Viewer works, examples and my own code don't. ("Frame didn't arrive within 15000")

DDreher, 2021-11-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.49.0 | 
| Language                            |  C++                          | 

Connected to a USB 3.2 slot, using the short data cable provided with the camera.

----------------------------------------------------------------------------------------------------

### Issue Description

Hey, whenever I call rs2::pipeline::wait_for_frames(), an exception is thrown:

> RealSense error calling rs2_pipeline_wait_for_frames(pipe:000001E4746D15D0):
>     Frame didn't arrive within 15000

This happens in all the example projects (e.g. rs-hello-realsense) as well as in my own code.
However, the Realsense-Viewer and the depth quality tool work just fine.
I found multiple issues like this, but none of the suggested fixes worked for me.

### Things I tried to fix the issue
- Ensured that the Realsense-Viewer (or any other application accessing the camera) is closed before running the examples / my own code.
- Tried different USB slots
- Checked the Windows camera settings and ensured that my programs are allowed to access the camera
- Ran the programs with administrator privileges
- Tried enabling streams explicitly
- Tried doing a hardware reset before creating the pipeline
- Tried calling try_wait_for_frames until I get a frame. Never returned true.

Is there any way I can further debug what exactly is going wrong here?

Thanks for your help.

MartyG-RealSense, 2021-11-03:
Hi @DDreher  May I first confirm whether you are using the pre-built Windows .exe versions of the example programs in the **Tools** folder of the RealSense SDK, please?

Also, can you confirm that you do not have the RealSense Viewer already open and streaming when launching the example programs, as this would block the example programs from being able to access the camera.

[image]

DDreher, 2021-11-03:
Hi @MartyG-RealSense,

thanks for the quick reply.

- I tried both the pre-built .exe versions and compiling them myself. No difference.
- The Realsense Viewer is definitely closed when launching the example programs.

I just tried an older version of the SDK (v2.42.0).
With this version the pre-built .exe samples work, but they fail with the exact same exception when I try to build them myself. No idea why that would be the case though...

MartyG-RealSense, 2021-11-03:
Thanks very much for the information.  Next, could you install SDK 2.48.0 using the automated installer file **Intel.RealSense.SDK-WIN10-2.48.0.3381.exe** and see whether the examples function in that SDK.  When the installer file is used to install the SDK on Windows, it overwrites the previous version installed with this method and so you do not need to be concerned about trying different versions on the same computer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

What method are you using to build the SDK example programs, please?  These examples have a different build method to applications that you have created yourself, and so they may fail in the way that you describe if they are built like a normal C++ program.  An Intel article in the link below describes how to build them.

https://www.intel.com/content/www/us/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

DDreher, 2021-11-03:
I installed SDK 2.48.0 with the automated installer as you described, then neither the examples nor the Realsense Viewer worked anymore, even after multiple hardware resets and restarts of the PC.
Uninstalled the SDK 2.48.0, re-installed SDK 2.42.0 and now both the viewer and the examples work again.

> What method are you using to build the SDK example programs, please? 

I built them in two ways:
- Via the VS solution contained in the samples directory, but with VS2019 instead of VS2017
- I also tried building them with VCPKG by enabling the "tools" feature.

Using VS2019 may be the issue here, but shouldn't the pre-built examples of the most recent SDKs work anyway?

MartyG-RealSense, 2021-11-03:
It's good to hear that a reinstall of 2.42.0 enabled both the Viewer and the examples to work.  

Virtual Studio 2019 should work fine with the RealSense SDK.

In this situation, where the pre-builts work but building the SDK examples doesn't, I would typically advise moving on to creating your own RealSense application, using the source code of the examples as a reference for writing your own script.  This is because due to the SDK examples having a special build process that is different to self-made applications, they are not necessarily a useful teaching aid for building RealSense projects.

DDreher, 2021-11-04:
I now also tried compiling the examples by cloning the repository and generating the solution with cmake as described in the support article you mentioned (https://www.intel.com/content/www/us/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html).    
Same issue, with lots of `WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'` when I start the pipeline.

> In this situation, where the pre-builts work but building the SDK examples from source code doesn't, I would typically advise moving on to creating your own RealSense application

The `Frame didn't arrive within 15000` also occurs in my own RealSense application, even when I try to implement a simple hello-realsense-like application.

MartyG-RealSense, 2021-11-04:
My research indicates that the **WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid** error is not specific to librealsense, but rather is generated by Visual Studio and can occur in non-RealSense projects such as ones using an ordinary webcam.  

Do you have any other cameras attached to your computer that could be conflicting with RealSense, please?  If you have a laptop, this could include a built in webcam on the screen.

DDreher, 2021-11-04:
No, I don't have any peripherals attached besides mouse and keyboard. It's a workstation, so no built in webcam either.

I guess the exceptions are caused by the way the projects are compiled, but I simply can't point my finger on the exact problem.

MartyG-RealSense, 2021-11-04:
Can you confirm whether the two RealSense drivers are shown in the **Cameras** category of the Windows Device Manager when the camera is plugged in?

[image]

DDreher, 2021-11-05:
Yes, both of the drivers show up with device status "working properly".

MartyG-RealSense, 2021-11-05:
I researched your case from the beginning in search of fresh insights.  Tanks very much for your patience.

When you install a new librealsense version on Windows (such as 2.49.0), are you linking your Visual Studio project to the files of that particular RealSenseSDK version (such as the prop files in the root folder of the currently installed SDK) and not linking the project to SDK files from a different version.

[image]

DDreher, 2021-11-08:
Hey, thanks for taking the time to look into this.

> When you install a new librealsense version on Windows (such as 2.49.0), are you linking your Visual Studio project to the files of that particular RealSenseSDK version (such as the prop files in [...]

Yes, the solution has the same configuration as specified in the .props files. I double-checked the resolved paths inside VS and they point to the SDK directories. There's also only ever one version of the SDK installed at the same time, so there should be no chance of linking to a different SDK.

MartyG-RealSense, 2021-11-08:
One RealSense user in https://github.com/IntelRealSense/librealsense/issues/4517 who experienced the **Frame didn't arrive withn 15000** error with rs-hello-realsense and other SDK examples resolved it by defining the depth stream specifically with a **config.enable_stream** instruction instead of allowing the camera to choose to apply its default configuration.

DDreher, 2021-11-09:
Thanks for the suggestion. I already tried this before opening this issue, sadly it didn't solve the problem.

> Things I tried to fix the issue
> * ...
> * Tried enabling streams explicitly

I did some further digging into the problem:
The exceptions are thrown whenever there's some interaction with the Windows Media Foundation using the stream indices (e.g. IMFSourceReader::GetNativeMediaType or IMFSourceReader::SetStreamSelection).

I guess this isn't expected behavior, right?

[image]

MartyG-RealSense, 2021-11-09:
Would it be an option for you to uninstall and then reinstall Visual Studio 2019 if you have not done so already in order to eliminate the possibility that there is a broken file or setting in your current VS19 installation?

DDreher, 2021-11-10:
I just tried with a fresh installation of VS17, no difference.

Lots of 

> Exception thrown at 0x7714B502 (KernelBase.dll) in rs-hello-realsense.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.

and then

> Exception thrown at 0x7714B502 in rs-hello-realsense.exe: Microsoft C++ exception: std::runtime_error at memory location 0x0113F828.
Exception thrown at 0x7714B502 in rs-hello-realsense.exe: Microsoft C++ exception: [rethrow] at memory location 0x00000000.
Exception thrown at 0x7714B502 in rs-hello-realsense.exe: Microsoft C++ exception: rs2::error at memory location 0x0113F968.

MartyG-RealSense, 2021-11-10:
Okay, thanks very much @DDreher for the update - I look forward to your test results from another PC.  Good luck!

DDreher, 2021-11-12:
Hey MartyG,

Yesterday I tried the camera on 3 different PCs, one running Linux, two running Win10. One of the Win10 PCs also had a totally fresh installation of Visual Studio.

On Linux everything ran fine and without issues.
On all of the Win10 PCs I encountered the issues described above. I think this is not a hardware issue, but a problem with Windows 10. Sadly, I need the camera to run on Windows 10 for my project, so I can't close this ticket.

Could you please tell me which exact version of Windows you are using to build the realsense-viewer.exe you ship with the installation package?    
Just for reference: The PCs I checked run on Windows 10 Pro, v21H1 (OS Build 19043.1348)

MartyG-RealSense, 2021-11-12:
The information about which specific Windows build was used to create realsense-viewer.exe is not available.  Having said that, the issue may not be related to a particular Windows build number.  I recently ran a test in https://github.com/IntelRealSense/librealsense/issues/9843 on two Windows PCs with 20H2 on, and the camera behaved normally on one whilst having missing infrared stream options in the Stereo Module section of the Viewer on the other.  

There was also a case a week ago in https://github.com/IntelRealSense/librealsense/issues/9913#issuecomment-961140573 where the camera would not work correctly on 20H2 machine but did on 21H1.

MartyG-RealSense, 2021-11-18:
Hi @DDreher  Do you have an update about this case that you can provide, please?  Thanks!

DDreher, 2021-11-18:
Hi, I made no further progress.
Just for the sake of completeness, I tested everything with another Realsense camera, but the mentioned issues remain.

MartyG-RealSense, 2021-11-18:
I reviewed this case from the beginning, and it seems that the **standalone** version of the RealSense Viewer has not been tested yet.  I do not mean the .exe in the SDK's 'Tools' folder but an .exe build of the Viewer called **Intel.RealSense.Viewer.exe** that has differences from realsense-viewer.exe.  This exe can be downloaded from the same 'Assets' list that the SDK installer file is in.  

[image]

Intel.RealSense.Viewer.exe should be launched from the Windows folder location that it was downloaded to and not with the launch shortcut icon on the Windows desktop.

DDreher, 2021-11-22:
Which exe I start doesn't seem to change anything.

MartyG-RealSense, 2021-11-22:
I reviewed your case again from the beginning to seek fresh insights.  It does not seem to be a problem with the camera, USB or the Windows build version, considering that the camera can work normally in the RealSense Viewer.  It has been demonstrated in other cases that the camera can work with Windows build 21H1.

My next suggestion would be to check the Windows **power plan** to see whether the camera might be being put to sleep by Windows power-saving settings.  You can check the currently set power plans by inputting **power plan** into the search box at the bottom of the Windows screen, next to the corner button where the PC is shut down.  On my own Windows PCs I set the power saving plans to 'Never' to minimize interruption.

[image]

On the power saving window, you can also left-click the **Change advanced power settings** link and then specifically prevent the USB port from being put to sleep in the **USB settings** category.

[image]

DDreher, 2021-11-23:
I just checked the setting: It was already disabled.

That being said, it's unlikely to be a power issue. The precompiled RealSense Viewer works, the one I built myself and my own programs using the Realsense SDK don't.

MartyG-RealSense, 2021-11-23:
The documentation for the **realsense-viewer** program states that when building it from source code, it should be built alongside the rest of the library.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer#implementation-notes

What this means is that instead of trying to compile the realsense-viewer program on its own, it should be built as part of a bundle of programs that include the example programs and also tools such as the Depth Quality Tool.  This is done by including the build terms **-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true** when building the RealSense SDK from source with CMake.  For example:

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

There is also the alternative method of building the examples using Visual Studio that you referenced earlier in this discussion.

https://www.intel.com/content/www/us/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

DDreher, 2021-11-24:
As mentioned, I already built the binaries correctly according to the documentation and in different ways, including generating a solution with cmake (with the correct flags) and using VCPKG.

I just tried again with a D455. The rs-motion example I compiled myself works just fine. All the examples which rely on the color and depth stream of the device don't.

I would really appreciate if you could ...
1. try to build the examples on Win 10 Pro, v21H1 (OS Build 19042.1348) and let me know if they work for you.
2. find out exact Win 10 version was used to build the binaries of v2.42 (because those actually worked reliably, newer ones don't.)
3. forward this issue to one of the responsible engineers who could investigate further.

MartyG-RealSense, 2021-11-24:
1.  I do not have the ability to perform a build of the examples from code on my computer, as I am not a developer.

2.  I will forward your question to Intel about the Windows 10 build used to compile the binaries of 2.42.0.

MartyG-RealSense, 2021-11-25:
Hi @DDreher  I received a response from Intel about your Windows version question.  The librealsense 'build farm' is based on Windows 10 RS5, and the binaries for v2.42 were compiled using Visual Studio 2015.  The compilations are done with default CMake optimization flags and RelWithDebugInfo build configuration.

DDreher, 2021-11-29:
Hi @MartyG-RealSense,

Thanks again for asking Intel.
I discussed this with my colleagues and we decided to move forward using a different product.

Thank you for your help and support.

Best

Daniel

closed 2021-11-29 by DDreher

MartyG-RealSense, 2021-11-29:
Thanks very much @DDreher for the update.  I hope that we will see you again in the future.  Best wishes for your project!