realsenseai/librealsense #9916: librealsense fails to claim USB interface on MacOS Monterey

dominiccarrano, 2021-11-03:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 |
| Firmware Version                   | N/A |
| Operating System & Version |  MacOS Monterey (12.0.1) |
| SDK Version                            |  2.49                          |
| Language                                |  python  |
### Issue Description
After upgrading my machine from macOS Catalina (10.15.6) to Monterey (12.0.1), all RealSense applications I have and pyrealsense2 code I’ve written -- which all previously ran without issue -- are broken. They all give me the same error message: **failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS**. I’ve verified both physically and via System Preferences that the D415 is connected to the machine with a USB 3 connection using the same cable that the D415 was shipped with.

When I run rs-enumerate-devices, I get:
```
> rs-enumerate-devices
Could not create device - failed to set power state . Check SDK logs for details
 02/11 16:49:13,578 ERROR [0x70000cff3000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
```

When I run realsense-viewer, I get:
```
> realsense-viewer
 02/11 16:52:08,669 ERROR [0x70000c115000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
 02/11 16:52:08,669 ERROR [0x70000bf8c000] (sensor.cpp:537) acquire_power failed: failed to set power state
 02/11 16:52:08,673 WARNING [0x70000bf8c000] (rs.cpp:308) null pointer passed for argument “device”
 02/11 16:52:08,673 WARNING [0x70000bf8c000] (rs.cpp:2621) Couldn’t refresh devices - failed to set power state
```

I wondered if something was wrong with the device firmware and tried updating it, which gives me:
```
> rs-fw-update -l
connected devices:
 03/11 14:56:46,495 ERROR [0x70000b0b1000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
RealSense error calling rs2_create_device(info_list:0x600003554e70, index:0):
    failed to set power state
 03/11 14:56:46,497 ERROR [0x119b34600] (sensor.cpp:537) acquire_power failed: failed to set power state
```

I uninstalled and then rebuilt librealsense from scratch for Monterey using the newest version of Xcode that’s Monterey compatible, but these issues still persist even after that. The [librealsense homebrew page](https://formulae.brew.sh/formula/librealsense) indicates that Monterey is supported. I’d greatly appreciate any guidance on fixing the **failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS** problem for Monterey, thank you!

MartyG-RealSense, 2021-11-03:
Hi @dominiccarrano   May I first confirm please whether you are using an Intel-based Mac computer or a newer Apple Silicon M1 model.

If you do have an Intel based Mac, have you attempted to build librealsense from a binary with brew using **brew install librealsense**

If you have chosen not to install librealsense with a brew binary because you require pyrealsense2 to be installed too, a RealSense Mac user recently shared a method in https://github.com/IntelRealSense/librealsense/issues/9687 that they developed to install a pre-built pyrealsense2 on Mac using pip **install**.

dominiccarrano, 2021-11-03:
Hi @MartyG-RealSense, thanks for the quick response! I'm on an Intel-based Mac. 

I've separately tried:
- **brew install librealsense**
- The steps described in [this link](https://github.com/GitwellAnyohub/Real_Sensible_Moseq/blob/master/bash_and_install/install_realsense_sdk_mac_os.sh)
- The method you've linked (although the author says it's only supported on Catalina and Big Sur, not Monterey)

and all produce this same RS2_USB_STATUS_ACCESS error.

MartyG-RealSense, 2021-11-04:
As you were earlier able to successfully use the camera on the same Mac with Catalina, that would make it unlikely that hardware problems with the camera or with the Mac computer are the cause of this problem.  It could be an OS software issue with Monterey.

Is your camera plugged into a hub?  There have been reports of USB detection issues recently from Mac users who have upgraded to Monterey, particularly with hubs.  An example discussion from the past week is in the link below.

https://www.macrumors.com/2021/10/29/monterey-usb-hub-issues-reported/

dominiccarrano, 2021-11-04:
Thank you for the link, I wasn't aware of this. The camera was plugged into a hub, which had no issues on Catalina. But even after plugging it into the machine with a direct USB3 connection, I get the same error. 

My understanding from perusing previous issues on this GitHub is that RS2_USB_STATUS_ACCESS typically means that libusb had trouble reading from a USB due to something permissions related. Is there any guidance from Intel on troubleshooting this for Mac (Monterey in particular)? I haven't been able to find anything online about how to debug this that wasn't specific to Linux.

MartyG-RealSense, 2021-11-05:
I researched the Monterey USB problem further.  In the majority of cases, waiting for Apple to release a fix in a patch seemed to be the most common advice.

There was one case where a Mac user restored access to their webcam in Monterey by resetting the Mac permissions with the command **tccutil reset All** in the terminal.

jeredhdean, 2021-11-08:
I have the same error as @dominiccarrano and have walked through the same troubleshooting with the exact same results. Also an Intel Mac with Monterey that used to be a happy RealSense user with Big Sur. 

First tried: tccutil reset Camera
Then tried: tccutil reset All

Still getting the RS2_USB_STATUS_ACCESS errors. If anyone finds a fix I'd love to know about it.

MartyG-RealSense, 2021-11-08:
@jeredhdean   Thanks very much for your report!

@dominiccarrano  Have you found any solutions since your last report, please?

dominiccarrano, 2021-11-09:
I got the same result as @jeredhdean: still have RS2_USB_STATUS_ACCESS after each of those tccutil reset commands.

MartyG-RealSense, 2021-11-09:
Reports about the Monterey USB issues strongly indicate that Thunderbolt ports are not affected.  Thunderbolt ports accept USB-C connections.  So conceivably, a camera might work if attached to a Thunderbolt port via a USB-A to USB-C adapter or a USB cable with USB-C micro sized connectors on both ends of the cable.

Point number 3 of a guide to the Monterey USB issue in the link below suggests trying resetting the System Management Controller (SMC).

https://cleanmymac.com/blog/usb-devices-disconnecting-monterey

dominiccarrano, 2021-11-10:
Thanks for the additional info on this, @MartyG-RealSense. Unfortunately, using a Thunderbolt connection instead didn't fix the issue, nor did an SMC reset. I should mention that this is on a late 2014 Mac mini with no Thunderbolt 3 (i.e., USB-C) ports, so I was using Thunderbolt 2 during that attempt.

However, when I booted up the Mac in Safe Mode the camera did work. In Safe Mode, `rs-enumerate-devices` and `rs-fw-update -l` behaved just as expected without any RS2_USB_STATUS_ACCESS error, and I was able to use pyrealsense2 to stream RGB-D frames within a Python shell. I couldn't use `realsense-viewer` in Safe Mode due to some component of glfw3 being blocked or failing. After rebooting the Mac again normally (i.e., not in Safe Mode), I got the same set of errors as before. I then tried all the suggestions under "If the issue doesn't continue in safe mode" (https://support.apple.com/en-us/HT201262), but still no luck.

MartyG-RealSense, 2021-11-11:
Thanks very much @dominiccarrano for the information about being able to access the camera in Safe Mode!  It seems likely though that a resolution for this issue will have to come from Apple's side via an update for Monterey.

sam598, 2021-11-14:
I am getting the same error in MacOS Monterey, but I don't think it is the same as the Monterey USB hub issue.

This code:
```
import pyrealsense2.pyrealsense2 as rs

ctx = rs.context()
print(len(ctx.devices))
```

Will print "1" when a RealSense camera is connected, so the python wrapper can tell that a device is connected.

But if I try to read and print the camera's serial number:
```
for i in range(len(ctx.devices)):
    sn = ctx.devices[i].get_info(rs.camera_info.serial_number)
    print(sn)
```
it will fail with `RuntimeError: failed to set power state`

None of the built example or realsense-viewer will be able to access the device properly.

Under system report MacOS can clearly see that a realsense device is connected under Hardware > USB
```
Intel(R) RealSense(TM) Depth Camera 455 :

  Product ID:	0x0b5c
  Vendor ID:	0x8086  (Intel Corporation)
  Version:	50.d0
  Serial Number:	110223061847
  Speed:	Up to 5 Gb/s
  Manufacturer:	Intel(R) RealSense(TM) Depth Camera 455 
  Location ID:	0x01210000 / 3
  Current Available (mA):	900
  Current Required (mA):	720
  Extra Operating Current (mA):	0
```

The device is also visible under Hardware > Camera
```
Intel(R) RealSense(TM) Depth Camera 455 :

  Model ID:	UVC Camera VendorID_32902 ProductID_2908
  Unique ID:	0x121000080860b5c
```

Interestingly if I open Apple's Photobooth App `Intel(R) RealSense(TM) Depth Camera 455` is available as an option along with `FaceTime HD Camera`. When selected it will show the RGB-Infrared stream and the IR projector is on at full power.

To me it seems that MacOS is able to power and access the RealSense camera without issue. There is something that Apple has changed that is causing the RealSense SDK to not work properly. This will likely require a change to the SDK that someone from Intel will need to investigate.

MartyG-RealSense, 2021-11-15:
Thanks very much everyone for your reports.  I will highlight the issue to Intel.

Nikhil-Kasukurthi, 2021-11-15:
I'm facing the same issue as well, with L515.

aagaard, 2021-11-15:
I also have the same issue with D415 and D435

sam598, 2021-11-15:
Confirmed the same behavior on both a late 2019 Intel MacBook Pro as well as a late 2021 M1 Pro MacBook Pro.

MacOS Monterey Version 12.0.1
RealSense SDK Version 2.50.0 and FW 5.13.0.50

MartyG-RealSense, 2021-11-16:
I discussed this case with Intel.  We will wait for Apple to release a fix for Monterey first.  Then we can see if RealSense related problems are still present after that.

sam598, 2021-11-16:
How does that make sense? If the issue was USB then you would not be able to access the camera at all. It appears the issue is with the SDK.

MartyG-RealSense, 2021-11-16:
Nevertheless, the above position of waiting for an Apple fix first is the one that has been chosen to be taken.

dominiccarrano, 2021-11-16:
Ran a few more experiments (all with the system booted up normally, i.e. not in Safe Mode): 
- I can use the D415 in Photo Booth to take photos and videos, so macOS itself doesn't seem to have an issue with accessing the camera.
- From digging through the librealsense source code, I found that `RS2_USB_STATUS_ACCESS` is an alias for `LIBUSB_ERROR_ACCESS`, which typically indicates a permissions error. I gave `rs-enumerate-devices` full permissions by going into System Preferences > Security & Privacy > Full Disk Access and adding the application, and then re-ran it. I still got the same bug, but the error message was slightly more verbose:

```
> rs-enumerate-devices # without full disk access
Could not create device - failed to set power state . Check SDK logs for details
 16/11 10:03:06,047 ERROR [0x70000c0c7000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS

> rs-enumerate-devices # with full disk access
 16/11 10:01:42,062 ERROR [0x70000f7f3000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
Could not create device - failed to set power state . Check SDK logs for details
 16/11 10:01:42,063 ERROR [0x10edef600] (sensor.cpp:537) acquire_power failed: failed to set power state
```

Vangos, 2021-11-17:
I have been facing the same issue since I updated to macOS Monterey. 

The camera appears in the USB list and I can properly use it **as a webcam**.

> @MartyG-RealSense May I first confirm please whether you are using an Intel-based Mac computer or a newer Apple Silicon M1 model.

To avoid spending time on testing different architectures, the chipset is irrelevant to this issue. I have compiled RealSense for Intel (x86_64), Apple Silicon (arm64), and Universal (x86_64 + arm64) and I've used it without any problems in Big Sur. 

Here's the test: I created a universal RealSense build on my M1 machine (Monterey). Obviously, it did not work on Monterey. However, when I used that exact same build on an Intel machine (Big Sur), it worked fine.

Same for x86 and ARM builds: they work on Big Sur, but not on Monterey.

Also, I recompiled libusb (latest version) and linked that to RealSense. No luck.

**It seems this issue is 100% related to Monterey and power delivery over USB.**

> @sam598 If the issue was USB then you would not be able to access the camera at all. It appears the issue is with the SDK.

The RealSense SDK uses the sensor in **full power**. I guess that's why it requires a dedicated USB-3/C. All other apps (like FaceTime or Photobooth) use RealSense **as a webcam**, limiting it to 30 FPS.

There are numerous user reports [here](https://www.macrumors.com/2021/10/29/monterey-usb-hub-issues-reported/). So, for now, I tend to agree with @MartyG-RealSense that it's something Apple should work on. However, I'm keeping the thread alive to ensure Intel engineers will take care of that for version 2.50.

MartyG-RealSense, 2021-11-26:
As a side-note on the subject of Monterey, it is worth highlighting that the simple installation procedure **brew install librealsense** now supports Monterey.

https://formulae.brew.sh/formula/librealsense

rowungiles, 2021-11-29:
On macOS Big Sur 11.6.1, I am also seeing this issue using an M1 Mac mini.

This install of Big Sur is absoluely fresh, I have only installed Xcode and the brew installs outlined on the Intel Dev site. The install was a downgrade from Monterrey specifically to get librealsense working.

I have the following output from 
Starting Xcode
Running `realsense-viewer`
Initial fail and then
Unplugging and re-plugging the camera in several times:

```
2021-11-29 12:52:33.855654+0000 realsense-viewer[18210:105475] Metal API Validation Enabled
2021-11-29 12:52:33.880607+0000 realsense-viewer[18210:105475] fopen failed for data file: errno = 2 (No such file or directory)
2021-11-29 12:52:33.880635+0000 realsense-viewer[18210:105475] Errors found! Invalidating cache...
2021-11-29 12:52:33.927411+0000 realsense-viewer[18210:105475] +[MTLIOAccelDevice registerDevices]: Zero Metal services found
 29/11 12:52:34,132 ERROR [0x17062f000] (handle-libusb.h:61) failed to set kernel driver auto detach: 0, error: RS2_USB_STATUS_NOT_SUPPORTED
 29/11 12:52:34,133 ERROR [0x17048b000] (sensor.cpp:572) acquire_power failed: failed to set power state
 29/11 12:52:34,134 WARNING [0x17048b000] (rs.cpp:310) null pointer passed for argument "device"
 29/11 12:52:34,134 WARNING [0x17048b000] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
 29/11 12:53:38,472 ERROR [0x1706bb000] (handle-libusb.h:61) failed to set kernel driver auto detach: 0, error: RS2_USB_STATUS_NOT_SUPPORTED
 29/11 12:53:38,472 ERROR [0x101fefd40] (sensor.cpp:572) acquire_power failed: failed to set power state
 29/11 12:53:38,473 WARNING [0x101fefd40] (rs.cpp:310) null pointer passed for argument "device"
 29/11 12:53:38,473 WARNING [0x101fefd40] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
 29/11 12:53:54,563 ERROR [0x1706bb000] (handle-libusb.h:61) failed to set kernel driver auto detach: 0, error: RS2_USB_STATUS_NOT_SUPPORTED
 29/11 12:53:54,563 ERROR [0x101fefd40] (sensor.cpp:572) acquire_power failed: failed to set power state
 29/11 12:53:54,564 WARNING [0x101fefd40] (rs.cpp:310) null pointer passed for argument "device"
 29/11 12:53:54,564 WARNING [0x101fefd40] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
 29/11 12:54:45,116 ERROR [0x1706bb000] (handle-libusb.h:61) failed to set kernel driver auto detach: 0, error: RS2_USB_STATUS_NOT_SUPPORTED
 29/11 12:54:45,116 ERROR [0x101fefd40] (sensor.cpp:572) acquire_power failed: failed to set power state
 29/11 12:54:45,118 WARNING [0x101fefd40] (rs.cpp:310) null pointer passed for argument "device"
 29/11 12:54:45,118 WARNING [0x101fefd40] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
```

I built the latest using master which I downloaded as a zip today.

I followed the build instructions from the [Intel Dev site](https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk). Except for using the cmake command from [#9999](https://github.com/IntelRealSense/librealsense/issues/9999#issuecomment-980108444)

Vangos, 2021-12-01:
> As a side-note on the subject of Monterey, it is worth highlighting that the simple installation procedure **brew install librealsense** now supports Monterey.

Note: Supports Monterey on Intel Macs only. M1 is still not supported by the standard installation procedure.

rowungiles, 2021-12-06:
We've just tried the SDK on an Intel mac with Monterrey and can confirm `RS2_USB_STATUS_NOT_SUPPORTED`.

Joe-Palmer, 2021-12-06:
I have just tried installing with `brew install librealsense` on this Intel Mac:
[image]
However, I'm getting the same USB not supported error:
```
joes-imac:~ joe$ realsense-viewer 
 06/12 10:42:22,506 ERROR [0x700009caf000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/12 10:42:22,507 ERROR [0x700009b26000] (sensor.cpp:537) acquire_power failed: failed to set power state
 06/12 10:42:22,508 WARNING [0x700009b26000] (rs.cpp:308) null pointer passed for argument "device"
 06/12 10:42:22,508 WARNING [0x700009b26000] (rs.cpp:2621) Couldn't refresh devices - failed to set power state
 ```
> > As a side-note on the subject of Monterey, it is worth highlighting that the simple installation procedure **brew install librealsense** now supports Monterey.
> 
> Note: Supports Monterey on Intel Macs only. M1 is still not supported by the standard installation procedure.

@Vangos how did you get it to work on an Intel Mac?

Vangos, 2021-12-06:
> @Vangos how did you get it to work on an Intel Mac?

Just to clarify: in my setup, it never worked on Monterey ([check my previous comment](https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-971468019)). @MartyG-RealSense mentioned:

> As a side-note on the subject of Monterey, it is worth highlighting that the simple installation procedure brew install librealsense now supports Monterey.

I have no reason to doubt that, but I have downgraded my Intel Mac to Big Sur.

All my M1 Macs are running Monterey, and neither the official `brew install` procedure nor a custom compilation is working.

MartyG-RealSense, 2021-12-06:
Thanks everyone for your ongoing reports.  I am keeping Intel updated on this issue.

oori, 2021-12-06:
The internet is smeared with Monterey & USB power issues. This seems to be a known issue since half a year, as it existed in Monterey beta. Unfortunately, it persists even in the current 12.1 beta :(
Seems like downgrade to BigSur is the only viable option in the near future.

oori, 2021-12-15:
Unfortunately, issue still there after MacOS 12.1 update.   No known workaround.

MartyG-RealSense, 2021-12-15:
Intel's stance about this issue is that it should be addressed by Apple rather than RealSense.

ecmnet, 2021-12-16:
D455 for me not working,  but T265 working fine on an Intel Monterey.

Meow25Meow, 2021-12-21:
I have a similar problem on my MacBook Pro (M1). macOS Monterey 12.1, when I run RealSense D435 on an application called Processing with the library "Intel RealSense for Processing 2.4.0 ", it stops working with the following error message

`RealSenseException: failed to set power state`

Although the development environment and the model used are a little different from yours, I believe that the problem was caused by the same reason you are facing.
Incidentally, the above problem did not occur when running the older macOS or Ubuntu, so it is most likely a problem specific to macOS Monterey.

Japanese
私のMacBook Pro(M1)でも、似たような問題が発生しています。macOS Monterey 12.1 で、RealSense D435をProcessingというアプリケーション上で、ライブラリ"Intel RealSense for Processing 2.4.0"を導入した状態で動作させると、以下のようなエラーメッセージとともに動作が停止します。

`RealSenseException: failed to set power state`

皆様とは開発環境や使用機種は少々異なりますが、皆様が直面している問題と同様の原因で動作しなかったものと思われます。
ちなみに、古いmacOSやUbuntu上で同機種を動作させると、上記の問題が発生しなかったため、やはりmacOS Monterey特有の問題である可能性が高いです。

timwedde, 2021-12-21:
Just started encountering this same issue on Big Sur (Intel Mac) with 2.50.0.
Downgrading to 2.49.0 made it work again, so this is definitely not just an Apple issue.

sam598, 2021-12-29:
@MartyG-RealSense Given all of the new information have your colleagues at Intel had a chance to investigate yet?

MartyG-RealSense, 2021-12-29:
@sam598   This issue is not being investigated by Intel, as the official position remains that it is a problem that should be addressed by Apple and not by Intel.

sam598, 2021-12-29:
@MartyG-RealSense I have doubts that Apple is going to contribute code to Intel's SDK.

MartyG-RealSense, 2021-12-29:
@sam598  I have forwarded your ongoing concerns about this issue to Intel.

sam598, 2021-12-30:
Thank you @MartyG-RealSense. Did you also forward the recent feedback from @timwedde @Meow25Meow @ecmnet and @oori? With you being the only way to communicate with your coworkers at Intel it is hard to know what information gets through.

MartyG-RealSense, 2021-12-30:
@sam598  I forward the issue link, so that everyone's comments are read.

laughingrice, 2022-01-03:
Could this be related to the change in behavior of opencv?

From my recent tests, opencv used to be able to open a camera directly and find the correct backend, now the backend needs to be chosen explicitly. At least when using in python:

import cv2
cam = cv2.VideoCapture(2, cv2.CAP_AVFOUNDATION)
e, f = cam.read()

as well as via photobooth, I can get an image from the realsense (D435), if it is connected when I fire them up I get the IR stream, if I disconnect and reconnect when the software is already running it goes to the RGB stream.

Trying to run opencv without choosing the backend returns 
(-215:Assertion failed) !info.backendFactory.empty()

trying to run librealsense directly also results with the USB error.

MartyG-RealSense, 2022-01-03:
Thanks very much @laughingrice for adding your experiences to the information shared by Monterey users in this discussion

MartyG-RealSense, 2022-01-03:
Hi everyone, after discussing the latest reports with Intel, the decision was made to file an official Intel bug report to investigate the issue further.  I have added an **Enhancement** tag to this case to indicate that it should be kept open until the investigation process is completed.

cansik, 2022-01-12:
As mentioned in https://github.com/IntelRealSense/librealsense/issues/10035#issuecomment-1010324381:

I have a implemented a [CI pipeline](https://github.com/cansik/pyrealsense2-macosx) to build the librealsense2 python library for MacOS (macOS Catalina (`10.15`), macOS Big Sur (`11.0`)). Version `2.49.0` worked as expected, but since version `2.50.0` the users and I have the same problem as mentioned in this issue. It seems it is not just a `Monterey` problem, but general with `2.50.0`.

- Reading the current device-count works (tested on `BigSur`)
- Starting a pipeline leads to a `No device connected` exception (tested on `BigSur`)

Also a nightly build (https://github.com/IntelRealSense/librealsense/commit/c94410a420b74e5fb6a414bd12215c05ddd82b69 using the head commit in the librealsense2 lib) did not help fixing this issue. Because of that I have removed the pre-built wheel packages from the repository (https://github.com/cansik/pyrealsense2-macosx/issues/2).

gr-markin, 2022-01-13:
I have the same behavior as @timwedde on Big Sur (Intel) after upgrading to 2.50.0 with L515. The previous version 2.49.0 worked without issues.

jackjansen, 2022-01-14:
Same here: 2.49 worked 2.50 fails. On an M1 mac running Monterrey (under Rosetta2) and an Intel Mac running Big Sur. 

And I'm installing librealsense via `brew`, and it refuses the build 2.49 even after you jump through the numerous hoops required to build older versions of packages that aren't specifically configured for it.

I've had to move to Windows for the time being, until this is fixed:-(

hajdebliem, 2022-02-01:
i managed to build 2.50 with an m1 mac under Monterey 12.2.

I can run the realsense-viewer from the terminal with sudo, once i add the app manually to the accessibility tab in system preferences/security.

however it is unstable and i get the following errors:
[image]

MartyG-RealSense, 2022-02-01:
Thanks so much @hajdebliem for sharing the progress that you made with using librealsense with MacOS Monterey  :)

Firmware 5.12.12.100 is the recommended firmware for librealsense **2.44.0** so it may be worth updating your firmware to **5.13.0.50**, the version recommended for 2.50.0.

hajdebliem, 2022-02-01:
I moved to a different machine to update the firmware. 
Back to my m1 Mac-> using the realsense-viewer ist still a mess ->
it randomly disconnects and freezes/crashes… 

however; all of my processing sketches do work once again!!

cansik, 2022-02-01:
@hajdebliem If you are using [realsense-processing](https://github.com/cansik/realsense-processing) as processing library: This one still uses librealsense2 version `2.44.0`, so the firmware version `FW: 05.12.12.100` should be fine.

MartyG-RealSense, 2022-02-01:
Hi @hajdebliem and @cansik  Thanks again for your help!

hajdebliem, 2022-02-01:
thanx @cansik  – for some reason processing4  (did not test 3) threw the same error in its console as the real sense-viewer. Something along the lines of "could not get power state"  and would immediately crash. So i am pretty happy so far as after the firmware upgrade and SKD update the sketches run without any issues so far…

gycn, 2022-02-13:
I downloaded release 2.49.0 and built it using XCode on a M1 Mac running Monterey (12.0), but I still get the error `failed to claim usb interface`. Is this because I'm using M1?

sam598, 2022-02-22:
> i managed to build 2.50 with an m1 mac under Monterey 12.2.
> 
> I can run the realsense-viewer from the terminal with sudo, once i add the app manually to the accessibility tab in system preferences/security.
> 
> however it is unstable and i get the following errors: [image]
> 
> RS: D435 FW: 05.12.12.100

This seems to confirm my fears and explains a lot of the behavior that has been seen so far. Apple has continued to add additional security features around camera access and privacy on both iOS and MacOS. My guess is that the Realsense SDK itself needs to request camera permissions before starting the device as an incoming video stream.

This would explain why Realsense cameras can be accessed as a webcam through first party apps.
This would explain why the issue occurs on the newer OS with new security permissions, and not just newer hardware.

@MartyG-RealSense any update on this? These new features in MacOS are not going to go away, and it seems clear that this is an issue that Intel needs to provide a solution for. If they do not have the developers or resources to continue to support MacOS, then they need to officially delist it as a supported platform.

MartyG-RealSense, 2022-02-22:
@sam598  No, there is no update.  I keep Intel advised of new cases that occur and are reported on this discussion.

MartyG-RealSense, 2022-02-23:
@sam598  I consulted with Intel further about the Monterey issue and was advised that the Supported Platforms information for MacOS only affirms official support up to the **High Sierra** OS.  The subsequent releases after High Sierra were Mojave, Catalina, Big Sur and Monterey.  Whilst Catalina and Big Sur work with the RealSense SDK, that compatibility is not officially guaranteed.

[image]

sam598, 2022-02-23:
@MartyG-RealSense that's even more frustrating as it is not possible to get MacOS devices with OSX High Sierra anymore.

If Intel no longer has the resources to investigate the privacy issue it would be better to clarify that Intel does not support MacOS at all.

It would also be good to know if this has officially been abandoned so the community can begin trying to work on a solution, as opposed to waiting on Intel. Especially since this issue has been around since at least November of last year.

sam598, 2022-02-23:
> Hi everyone, after discussing the latest reports with Intel, the decision was made to file an official Intel bug report to investigate the issue further. I have added an **Enhancement** tag to this case to indicate that it should be kept open until the investigation process is completed.

@MartyG-RealSense since the investigation is over is there any insight that was found that could help the community solve this issue?

MartyG-RealSense, 2022-02-23:
@sam598  There is no further information to add.  At present the best theories in this discussion are your own one about camera permissions and the progress made by @hajdebliem earlier at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1026756545

MartyG-RealSense, 2022-02-28:
Hi @Vangos  Thanks so much for your testing and information sharing!  I will highlight your findings to Intel.

[77 more replies]

closed 2026-10-29 by MartyG-RealSense