realsenseai/librealsense #9917: I want to know the power of realsense d435

Sweet-orange-gethub, 2021-11-04:
Hello, I want to know the power of realsense d435.thank you

MartyG-RealSense, 2021-11-04:
Hi @Sweet-orange-gethub   Could you provide more information please.  For example, do you mean that you want to know how much electrical power a RealSense D435 camera uses?  Thanks!

Sweet-orange-gethub, 2021-11-04:
Yes, it's power consumption

MartyG-RealSense, 2021-11-04:
In general, I would recommend budgeting 2 watts of power for each camera that you are using.  

In practice though, power usage is a much more complex subject.  The data sheet document for the 400 Series cameras provides very detailed information about this.  

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

Below is a table extracted from the data sheet about power usae of the camera's Vision Processor D4 hardware component.

[image]

Page 82 onwards of the data sheet breaks down in tables the power used by various components on the Vision Processor D4 and the camera Depth Module when in 'max operating mode' (described by the data sheet as "maximum resolution of the depth and RGB imagers at 30 FPS").

closed 2021-11-18 by MartyG-RealSense