realsenseai/librealsense #9918: Qt creator embeded with Intel RealSense SDK cannot recognized the file

Hupengjie-coder, 2021-11-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am tring to use Qt creator with Intel RealSense SDK 2.0 to create a demo in order to case the iamge in the UI interface, however, When I copy the code from the other Question's answer, I find that there is something wrong in my pro file, and here is the configuration of my project, and I use them in the windows systems, and I can't open the UI even if the project can be qmake. so I would like to hear your suggestion and critize, Looking forward to your reply! Thanks!
` CONFIG += c++11
    
    # You can make your code fail to compile if it uses deprecated APIs.
    # In order to do so, uncomment the following line.
    #DEFINES += QT_DISABLE_DEPRECATED_BEFORE=0x060000    # disables all the APIs deprecated before Qt 6.0.0
    
    SOURCES += \
        CAMERA.cpp \
        main.cpp \
        mainwindow.cpp
    
    HEADERS += \
        CAMERA.h \
        mainwindow.h
    
    FORMS += \
        mainwindow.ui
    
    # Default rules for deployment.
    qnx: target.path = /tmp/$${TARGET}/bin
    else: unix:!android: target.path = /opt/$${TARGET}/bin
    !isEmpty(target.path): INSTALLS += target
    
    #RealSense SDK
    INCLUDEPATH += "C:/Program Files (x86)/Intel RealSense SDK 2.0/include/"
    LIBS += "C:/Program Files (x86)/Intel RealSense SDK 2.0/bin/x86/realsense2.dll" #-realsense2 #-llibpxcmd #-llibpxcutilsmd  << Here is the error
    
    # opencv
    #INCLUDEPATH += E:\opencv3\include

    #LIBS += E:\opencv3\x64\mingw\bin\libopencv_*.dll# opencv 动态链接`
Here is the code links: [code links](https://github.com/Hupengjie-coder/Qt-with-Intel-RealSense/blob/main/demo1.rar)

MartyG-RealSense, 2021-11-04:
Hi @Hupengjie-coder   On the line where you note that you are experiencing an error, you set the path to realsense2.dll to use the version of this dll in the **x86** folder (i.e the 32-bit version of the file).   Could you therefore try changing bin/x86 to **bin/x64** to use the 64-bit version of the dll, please and see whether this corrects your problem?  

[image]

Hupengjie-coder, 2021-11-05:
Sorry, when converting it to x64, the project cannot be compiled totally, only with the x86 can the project run, but the UI interface cannot be opened, you can try it in your Qt app.

MartyG-RealSense, 2021-11-05:
I am not able to install and run Qt Creator on my computer to try to replicate your results with a test app, unfortunately.

Can you confirm please if you are attempting to embed a RealSense camera image inside a QT Creator app's window, please?  And you are able to build the project successfully but the image is not displaying?

Hupengjie-coder, 2021-11-08:
That's right, I can't imshow the image but the project can be built successfully, I tried serial ways to overcome it , however, they were not play in work.

MartyG-RealSense, 2021-11-08:
There are very few references unfortunately about using Qt Creator with RealSense and imshow except for a script provided at https://github.com/IntelRealSense/librealsense/issues/2923#issuecomment-449569073

closed 2021-11-20 by MartyG-RealSense