realsenseai/librealsense #9921: "cannot create std::vector larger than max_size()" while call pipline.start()

HEIDIES, 2021-11-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   Linux (Ubuntu 20.04.2)  | 
| Kernel Version (Linux Only)    |  5.11.0                                        | 
| Platform                                 | PC (Intel NUC)  |
| SDK Version                            |  2.34.0                           | 
| Language                            |  C/C++                          | 
| Segment			|  others                          | 

### Issue Description
As described in title, a rs2::error "cannot create std::vector larger than max_size()" caught while calling pipline.start()

MartyG-RealSense, 2021-11-04:
Hi @HEIDIES  Your SDK and firmware combination is not ideal.  SDK **2.34.0** had an issue with continuously generating timing errors for some RealSense users and so I recommend not using that version if possible.  

Also, if firmware 5.12.13.50 is used then the minimum SDK version used with the camera should be **2.39.0** because of internal changes in the firmware that were introduced after firmware 5.12.7.100.  If you need to use SDK 2.34.0 then firmware 5.12.7.100 would be a better choice.

[image]

In addition, kernel 5.11 is not yet officially supported by the RealSense SDK and so there may be risk of instability unless your librealsense was built with the **RSUSB backend** method that is not dependent on Linux versions or kernel versions.

However, I note that your code had run well for you on your current RealSense configuration until several days ago.

If you have not changed anything on your configuration recently (such as installing librealsense again), could you try using an Ubuntu system monitoring tool such as **htop** to monitor your computer's available memory capacity when your program is run, please?  This will help to eliminate whether a memory leak could be causing your computer to have insufficient memory to run the program correctly.

HEIDIES, 2021-11-05:
Hi @MartyG-RealSense, thanks for your advice, I remake and reinstalled the library, then the error miraculously disappeared. Everything runs well now.

closed 2021-11-05 by HEIDIES

MartyG-RealSense, 2021-11-05:
Great news, @HEIDIES - thanks very much for the update!