realsenseai/librealsense #9923: Getting depth data of all frame

Karin-Sugi, 2021-11-05:
Hi, i'm using D435i with matlab. I can get depth data at a point (center of the frame, like example code) 

Now,  i'd like to get depth data of all frame, not one point. I tried it with for loop, and got error saying
```
Error: realsense.depth_frame/get_distance (line 16)
out of range value for argument "y"
Error: a_trial_02 (line 20)
parfor i = 1:height
```
I think it means the depth frame is not so big though i set 
```
width = depth.get_width();
height = depth.get_height();
```
I mean
```
parfor i=1:width
    for j=1:height
        depth_data_set(i,j)=depth.get_distance(i,j);
```
I made code like this.
Why i,j don't suit depth.get_distance(x,y) command? How can i fix it?

Thanks

MartyG-RealSense, 2021-11-05:
Hi @Karin-Sugi  Another RealSense MATLAB user in https://github.com/IntelRealSense/librealsense/issues/8241#issuecomment-769326593 wrote a script that used **depth.get_distance(x,y)** in a loop and believed that this was inappropriate, so provided a script in https://github.com/IntelRealSense/librealsense/issues/8241#issuecomment-769735273 that used a different approach to achieve the goal.

Karin-Sugi, 2021-11-05:
Hi, @MartyG-RealSense 
Thanks for your answer! You mean i should not use **depth.get_distance(x,y)**

MartyG-RealSense, 2021-11-05:
You can use **depth.get_distance(x,y)** in RealSense MATLAB wrapper applications.  How it is implemented in the script will of course make a difference to whether it works or not.

A RealSense MATLAB user in https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-925553900 who was using get_distance recently shared their successful solution.

https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-925553900

```
% get the distance
dis_matrix = zeros(1080,1920);
for y = 0:(depthHeight-1)
for x = 0:(depthWidth-1)
depth_dis = depth_frame.get_distance(x,y);
dis_matrix(y+1,x+1) = depth_dis;
end
end
```

Karin-Sugi, 2021-11-06:
@MartyG-RealSense Hi, I could!! Great thanks for your help! Now, I close here.

closed 2021-11-06 by Karin-Sugi

MartyG-RealSense, 2021-11-06:
Great to hear that you were successful, @Karin-Sugi - thanks very much for the update!