realsenseai/librealsense #9924: Set Camera Settings using pyrealsense2 like realsense-viewer and load_json for multiple devices Problem

HasanFarag, 2021-11-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |  Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.11.0-38-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.49.0                    | 
| Language                            |  python                       | 
| Segment			|  Robot                     | 

### Issue Description
Firstly, We use advanced mode in pyrealsense2 to load exported settings json from the viewer and compare the depth images from the viewer and pyrealsense2, we find out that they are not the same. ( We check the cpp files of sdk and found that librealsense does not set all settings exposed in the viewer). We tried even to use visual preset in pyrealsense2 and postprocessing but still depth images from the viewer are not the same from pyrealsense2. The viewer depth images are much better in quality. How to use the settings from the viewer properly in pyrealsense2 to get the same depth quality (at least visually for us)?

Secondly, We tried to use advanced mode in pyrealsense2 to load exported settings json for different cameras, but it only sets the settings for first device properly and ignore the rest of devices. we use advanced mode in pyrealsense2. How to use advanced mode in pyrealsense2 for multi-camera?

MartyG-RealSense, 2021-11-05:
Hi @HasanFarag  The RealSense Viewer applies a range of post-processing and colorization settings by default.  When creating your own application, such as a Python script, these settings are not included by default and must be deliberately programmed into the application.

If you have already replicated the post-processing filters that are enabled by default in the Viewer, you could check that you have fully replicated the Viewer's default **color scheme** and **histogram equalization** settings.  https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671 and https://github.com/IntelRealSense/librealsense/issues/7089#issuecomment-672105974 provide Python scripting resources for configuring the color scheme and histogram on / off status.

A RealSense team member in https://github.com/IntelRealSense/librealsense/issues/6075#issuecomment-599953561 describes how enabling histogram equalization provides a "nice rainbow of colors", so if histogram is disabled by default in your Python application then this may account for some of the difference between your Viewer and Python images.

In regard to applying settings for individual cameras, you could set up a separate pipeline for each camera with that camera's unique serial number, and configure an individual set of cfg instructions for each pipeline.  An example of a Python script for doing this can be found in https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337

MartyG-RealSense, 2021-11-12:
Hi @HasanFarag  Do you require further assistance with this case, please?  Thanks!

HasanFarag, 2021-11-12:
Sorry, I do not have time to test it. I will test it soon , meanwhile, i will close the issue till further investigation. Thanks for rapid response.

closed 2021-11-12 by HasanFarag

MartyG-RealSense, 2021-11-12:
Thanks very much @HasanFarag for the update!  Please do feel free to re-open this issue or create a new one when you are ready to resume.  Good luck!